nkolban/esp32-snippets #189: Log level makes difference

chegewara, 2017-11-11:
Im trying to run SampleServer with multiple characterisyics with the same UUID. I have 2 characteristics, each characteristic has 3 different descriptors. More or less it works until i switch log level. With log level verbose works, with log level debug or lower it crash on first ESP_GATTS_ADD_CHAR_EVT event with error in getService()->getHandle().

nkolban, 2017-11-12:
Would you happen to have a small sample that illustrates the problem.  I believe you have supplied enough information for someone else to code it up ... but it sounds like you may have a test app to hand?

chegewara, 2017-11-18:
Just wanted to post example with code to recreate this issue but seems its fixed already.

closed 2017-11-18 by chegewara