nkolban/esp32-snippets #190: BLE Scan not ending after defined duration

nkolban, 2017-11-12:
When we call the BLEScan::start() method we specify an interval (in seconds) after which the scanning should end.  What we are finding is that  it simply isn't ending.  It keeps on going.

nkolban, 2017-11-13:
Following a great hint from Mr @chegewara , it has been found that the BLE scan will not end **only** if there is an advertising device out there that is sending some strange information.  In my world, I have an Android phone on which I run nRFConnect.  When I set it to advertise, the ESP32 BLE scanning doesn't end.  When I disable the nRFConnect advertising from the phone, BLE scanning stops again.  This is a pretty big clue.  I examined the records we get from scan results and see the following when we receive an incoming scan request from the phone:

```
D (944) BLEUtils: Received a GAP event: ESP_GAP_BLE_SCAN_RESULT_EVT
D (950) BLEUtils: Unknown esp_ble_evt_type_t: 18 (0x12)
D (950) BLEUtils: search_evt: ESP_GAP_SEARCH_INQ_RES_EVT, bda: 7c:9c:6d:d6:60:ee, dev_type: ESP_BT_DEVICE_TYPE_BLE, ble_addr_type: BLE_ADDR_TYPE_RANDOM, ble_evt_type: *** Unknown ***, rssi: -84, ble_adv: ??, flag: 26, num_resps: 1, adv_data_len: 24, scan_rsp_len: 24
D (979) BLEAdvertisedDevice: - setRSSI(): rssi: -84
D (984) BLEAdvertisedDevice: Type: 0x01 (ESP_BLE_AD_TYPE_FLAG), length: 1, data: 1a
D (992) BLEAdvertisedDevice: Type: 0x09 (ESP_BLE_AD_TYPE_NAME_CMPL), length: 16, data: 53414d53554e472d534d2d4739353055
D (1003) BLEAdvertisedDevice: - setName(): name: SAMSUNG-SM-G950U
D (1009) BLEAdvertisedDevice: Type: 0x0a (ESP_BLE_AD_TYPE_TX_PWR), length: 1, data: f9
D (1017) BLEAdvertisedDevice: - txPower: -7
D (1021) BLEAdvertisedDevice: Type: 0xff (ESP_BLE_AD_MANUFACTURER_SPECIFIC_TYPE), length: 4, data: 11991234
D (1031) BLEAdvertisedDevice: - manufacturer data: 11991234
D (1037) BLEAdvertisedDevice: Type: 0x07 (ESP_BLE_AD_TYPE_128SRV_CMPL), length: 16, data: 592fa49fde4e2baa264250b992d4ba91
D (1048) BLEAdvertisedDevice: - addServiceUUID(): serviceUUID: 91bad492-b950-4226-aa2b-4ede9fa42f59
D (1057) SampleScan: Advertised Device: Name: SAMSUNG-SM-G950U, Address: 7c:9c:6d:d6:60:ee, manufacturer data: 11991234, serviceUUID: 91bad492-b950-4226-aa2b-4ede9fa42f59, txPower: -7
```

and here is another device that we can examine for comparison:

```
D (686) BLEUtils: Received a GAP event: ESP_GAP_BLE_SCAN_RESULT_EVT
D (686) BLEUtils: search_evt: ESP_GAP_SEARCH_INQ_RES_EVT, bda: 96:57:a5:0b:a6:a4, dev_type: ESP_BT_DEVICE_TYPE_BLE, ble_addr_type: BLE_ADDR_TYPE_PUBLIC, ble_evt_type: ESP_BLE_EVT_CONN_ADV, rssi: -66, ble_adv: ??, flag: 6, num_resps: 1, adv_data_len: 17, scan_rsp_len: 24
D (706) BLEAdvertisedDevice: - setRSSI(): rssi: -66
D (711) BLEAdvertisedDevice: Type: 0x01 (ESP_BLE_AD_TYPE_FLAG), length: 1, data: 06
D (718) BLEAdvertisedDevice: Type: 0xff (ESP_BLE_AD_MANUFACTURER_SPECIFIC_TYPE), length: 5, data: 650001c905
D (728) BLEAdvertisedDevice: - manufacturer data: 650001c905
D (734) BLEAdvertisedDevice: Type: 0x0a (ESP_BLE_AD_TYPE_TX_PWR), length: 1, data: 06
D (742) BLEAdvertisedDevice: - txPower: 6
D (746) BLEAdvertisedDevice: Type: 0x03 (ESP_BLE_AD_TYPE_16SRV_CMPL), length: 2, data: 78fe
D (754) BLEAdvertisedDevice: - addServiceUUID(): serviceUUID: 0000fe78-0000-1000-8000-00805f9b34fb
D (763) BLEAdvertisedDevice: Type: 0x09 (ESP_BLE_AD_TYPE_NAME_CMPL), length: 22, data: 454e56592050686f746f203738303020736572696573
D (775) BLEAdvertisedDevice: - setName(): name: ENVY Photo 7800 series
D (782) SampleScan: Advertised Device: Name: ENVY Photo 7800 series, Address: 96:57:a5:0b:a6:a4, manufacturer data: 650001c905, serviceUUID: 0000fe78-0000-1000-8000-00805f9b34fb, txPower: 6
```

Two immediate items appear different.  The first is the `esp_ble_evt_type_t` is unknown and the second is that the `ESP_BLE_AD_TYPE_FLAG` is different.

nkolban, 2017-11-13:
We added some new function to dump the BLE Advertised Data flags (adFlags) and find for the failing device we get:

```
flag: 26 ([LE General Discoverable Mode] [Simultaneous LE and BR/EDR to Same Device Capable (Controller)] [Simultaneous LE and BR/EDR to Same Device Capable (Host)] )
```

while for the devices that don't cause problems we get:

```
flag: 6 ([LE General Discoverable Mode] [BR/EDR Not Supported] ),
```

chegewara, 2017-11-15:
Well, i have case that does not match my earlier experience. On one devKitC i have installed BLEServer with customized code, the other is running esp32_explore. Explorer is running scan and sometimes scan stop after 1 sec (its time i have set) and sometimes it just run until stop (teens seconds). My esp32 are only devices in range that are using BLE, so any other device cant be the reason of this behavior.

TianHao-Espressif, 2017-11-15:
@nkolban , do you call `esp_ble_resolve_adv_data`  when receive advertising? And could you provide me the advertising data sent from your Samsung phone?

nkolban, 2017-11-18:
I rebuilt from the latest ESP-IDF and a replaced sdkconfig file (from the latest template) and am now no longer able to recreate the issue.  I'm going to work on the assumption that *something got fixed* and hope that will be the end of the story.

closed 2017-11-18 by nkolban