nkolban/esp32-snippets #194: HttpServer not reachable after not connected for a while

Gfast2, 2017-11-13:
Hi Kolban,

Thanks for your books & repositories! They are awesome! I've learned tons of stuff from them! 👍 
Right now I investigate all my full time job on the development based on ESP32.

Today, I spent really many hours to try out your `HttpServer` implementated in C++. I can successfully fire it up and test out some basic "Ask" -> "Response" demo. But no matter how I isolate this simple demo webserver. ( I mean I comment out all other unrelated tasks, only init the `ETH` interface firstly and then if I got IP Addresss - `case SYSTEM_EVENT_ETH_GOT_IP:`, I start the webserverTask, which contain the basic usage of this `HttpServer`). It won't react to my http request from Google Chrome after a While, that I stop refreshing the webpage.
The first unsuccessful http request do let the free heap size shrinkt, but my handlerFunction for that same request won't get triggered at all.

I do know the problem can be everywhere from the network-switcher to Network-cable or even my very cheap _LAN8720 ETH Board_. But perhaps you have a very "sharp noise", that can tell the problem in the first second! 🙈 

_P.s: Because the webserver is such a small one, I don't bother to create a **Gist** for it. Here is the Webserver_

```
#include "esp_log.h"
#include "esp_system.h" // esp_get_free_heap_size()
#include "esp_eth.h"
#include <string>

#include "webserver.h"
#include "HttpServer.h"

#include "sdkconfig.h"

#define TAG "WEBSERVER"

void getHandlerFunc(HttpRequest* pRequest, HttpResponse* pResponse){
	ESP_LOGW(TAG, "In hello world handler");

	std::string temp = pRequest->getPath();
	char * buf = (char *)(temp.c_str());

	ESP_LOGW(TAG, "%s", buf);
	pResponse->setStatus(HttpResponse::HTTP_STATUS_OK, "OK");
	pResponse->addHeader(HttpRequest::HTTP_HEADER_CONTENT_TYPE, "text/plain");

	const char home[] = "/";
	const char favi[] = "/favico.ico";

	if(strcmp(buf, home) == 0) {
		pResponse->sendData("Hello back");
	} else if(strcmp(buf, favi) == 0) {
		pResponse->sendData("data:image/x-icon;base64,AAABAAEAEBAQAAEABAAoAQAAFgAAACgAAAAQAAAAIAAAAAEABAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAZjMAAHpHFABMTEwAgICAAMDAwAD///8AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAFVVVVVVVVVVAAAAAAAAAAMABTAAZmJWYwBGMABmYhVjBGZVVREREVNGZmZmERERI0ZmZmYREREjBGZVVREREVMARjAAZmIVYwAFMABmYlZjAAAAAAAAAANVVVVVVVVVVQAAAAAAAAAAAAAAAAAAAAD//wAA//8AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAP//AAD//wAA");
	}

	pResponse->close();
}

void webserver(void *data) {

	ESP_LOGW(TAG, "Web server get inited.");

	HttpServer * httpServer = new HttpServer();
	httpServer->addPathHandler(HttpRequest::HTTP_METHOD_GET, "/", getHandlerFunc);
	httpServer->addPathHandler(HttpRequest::HTTP_METHOD_GET, "/favicon.ico", getHandlerFunc);
	httpServer->start(80);

	while (1) {

		uint32_t freeHeap = esp_get_free_heap_size();
		ESP_LOGI(TAG, "free heap: %d", freeHeap);
		vTaskDelay(500/portTICK_RATE_MS);

	}
}
```

Gfast2, 2017-11-13:
BTW:
If I do really quick page refreshing (in Chrome with shortcut _**F5**_), then the webserver get crashed.
_The following are the error log and the `gdb` Log_
```
I (35575) WEBSERVER: free heap: 157928
W (35624) WEBSERVER: In hello world handler
W (35624) WEBSERVER: /
assertion "next > (intptr_t)block" failed: file "/home/gfast2/esp/esp-idf/components/heap/./multi_heap.c", line 116, function: get_next_block
abort() was called at PC 0x400e6b1b on core 1
0x400e6b1b: __assert_func at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdlib/../../../.././newlib/libc/stdlib/assert.c:63 (discriminator 8)

Backtrace: 0x40086de0:0x3ffd3e10 0x40086eab:0x3ffd3e30 0x400e6b1b:0x3ffd3e50 0x4008650f:0x3ffd3e80 0x4008680b:0x3ffd3ea0 0x400824f6:0x3ffd3ec0 0x400828e9:0x3ffd3ee0 0x4000bec7:0x3ffd3f00 0x400f806d:0x3ffd3f20 0x4011b1d9:0x3ffd3f40 0x40119a3d:0x3ffd3f80
0x40086de0: invoke_abort at /home/gfast2/esp/esp-idf/components/esp32/./panic.c:553

0x40086eab: abort at /home/gfast2/esp/esp-idf/components/esp32/./panic.c:553

0x400e6b1b: __assert_func at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdlib/../../../.././newlib/libc/stdlib/assert.c:63 (discriminator 8)

0x4008650f: get_next_block at /home/gfast2/esp/esp-idf/components/heap/./multi_heap.c:339
 (inlined by) assert_valid_block at /home/gfast2/esp/esp-idf/components/heap/./multi_heap.c:150

0x4008680b: multi_heap_free_impl at /home/gfast2/esp/esp-idf/components/heap/./multi_heap.c:339

0x400824f6: heap_caps_free at /home/gfast2/esp/esp-idf/components/heap/./heap_caps.c:136

0x400828e9: _free_r at /home/gfast2/esp/esp-idf/components/newlib/./syscalls.c:42

0x400f806d: operator delete(void*) at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/del_op.cc:46

0x4011b1d9: __gnu_cxx::new_allocator<char>::deallocate(char*, unsigned int) at /home/gfast2/workspace/ioc-110-ethernet/components/cpp_utils/./FreeRTOS.cpp:151
 (inlined by) std::allocator_traits<std::allocator<char> >::deallocate(std::allocator<char>&, char*, unsigned int) at /home/gfast2/esp/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/5.2.0/bits/alloc_traits.h:386
 (inlined by) std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_destroy(unsigned int) at /home/gfast2/esp/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/5.2.0/bits/basic_string.h:185
 (inlined by) std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_dispose() at /home/gfast2/esp/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/5.2.0/bits/basic_string.h:180
 (inlined by) std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::~basic_string() at /home/gfast2/esp/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/5.2.0/bits/basic_string.h:544
 (inlined by) FreeRTOS::Semaphore::give() at /home/gfast2/workspace/ioc-110-ethernet/components/cpp_utils/./FreeRTOS.cpp:140

0x40119a3d: SockServ::acceptTask(void*) at /home/gfast2/workspace/ioc-110-ethernet/components/cpp_utils/./SockServ.cpp:134

Entering gdb stub now.
$T0b#e6GNU gdb (crosstool-NG crosstool-ng-1.22.0-73-ge28a011) 7.10
Copyright (C) 2015 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "--host=x86_64-build_pc-linux-gnu --target=xtensa-esp32-elf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /home/gfast2/workspace/ioc-110-ethernet/build/ethernet_demo.elf...done.
Remote debugging using /dev/ttyUSB0
0x40086de0 in invoke_abort () at /home/gfast2/esp/esp-idf/components/esp32/./panic.c:139
139	        *((int*) 0) = 0;
(gdb) l
134	#endif
135	    while(1) {
136	        if (esp_cpu_in_ocd_debug_mode()) {
137	            __asm__ ("break 0,0");
138	        }
139	        *((int*) 0) = 0;
140	    }
141	}
142	
143	void abort()
(gdb) 

```

---
Because the still missing Hardware / Know-how on debugging these problems, would someone give me small tipps about how to debug these issues? I do appriciate that! 🙇

nkolban, 2017-11-14:
The "cpp_utils" code (of which the HttpServer is a component) is full of diagnostic tracing.  To enable this, compile the code after having changed the "make menuconfig" entry in "Component config > Log output > Default log verbosity" to "Verbose".  Recompile all the code and re-run.  We should now see a lot of good diagnostics written to the console.  Try your tests again and capture the console output.  Once done, paste the results to a pastebin and post a link here.

Gfast2, 2017-11-14:
Hi  @nkolban ,

Thanks for your advices. Yesterday night I figure out may be I have some hardware problem, too. Many software Problem I found is hight suspected related to this hardware problem.

[This](https://github.com/espressif/esp-idf/issues/1237#issuecomment-344089470) is the related problem.

If I fix this today, I will try the webserver. And if I still get stucked. I will do as you said. If not, I will close this issue later.

Thanks again. 🏖

Gfast2, 2017-11-14:
Hi,

I figured out, it is only a harware problem caused by ETH's quartz pin not very properly pull-up (supposing ly till now). But not the problem of @nkolban 's library.

So I close this _issue_

nkolban, 2017-11-14:
I'm afraid you are very far ahead of me.  I have NEVER tried to play with ESP32 and Ethernet.    That whole area is a mystery to me and ... as such ... is spooky.

closed 2017-11-14 by nkolban

Gfast2, 2017-11-14:
Hi @nkolban , It's such a neat feeling that you are always there! ⛱ 

Right now I'm fighting with the Ethernet+SD card jumbo solution. Because I still did not get the chance to get a [***Olemix ESP32-Gateway***](https://www.olimex.com/Products/IoT/ESP32-GATEWAY/open-source-hardware), I did a DIY one. It gives me really meserable problems sometimes. 
[***This one***](https://github.com/espressif/esp-idf/issues/1237#issuecomment-344241246) is what I addressed today. Since I fixed the hardware problem today, I did go many steps on the software side. So thanks again.