nkolban/esp32-snippets #195: BLE beacon advertising service UUID

menesesleonardo, 2017-11-13:
I'm following this example for nrf52 boards for the development of a platform for indoor location using Arduino.

[https://github.com/adafruit/Adafruit_nRF52_Arduino/tree/master/libraries/Bluefruit52Lib/examples/Projects/rssi_proximity](url)

There, as you can see, all the peripheral devices advertise a 128-bit service UUID with flag 0x07 from bluetooth GAP specificcations.
[image]

[image]

[https://www.bluetooth.com/specifications/assigned-numbers/generic-access-profile](url)

Then, a central device filters peripherals and sorts those who have that specific UUID according to their rssi.

I wanted to make my ESP32 A PERIPHERAL, BUT When I created a BLE server with an ESP32 board, uploading this example from ESP32 Arduino BLE library, 

[https://github.com/nkolban/ESP32_BLE_Arduino/tree/master/examples/BLE_server](url)

The advertised UUID doesn't have this flag and looks like an unknown service, here are the screenshots
[image]

[image]

So the Central device doesn't recognize it at all.

I appreciate any help. Thanks

chegewara, 2017-11-13:
BLE_server is general purpose example. From BLE specification you can learn that advertising packet has to be 31 bytes or less in total. What you can see on first screenshot you provided is:
- type 0x01 - flags, length 2 bytes,
- type 0x0A - TX power, length 2 bytes,
- type 0x07 - advertised 128 bit UUID, length 17 bytes, 1 byte + 16 bytes UUID,
- type 0x09 - name, length 12 bytes.

On screenshot from BLE_server you have more types of advertised data, in such there is no space for UUID. What you can do is:
- shorten device name or even leave it empty string: BLEDevice::init("")
- type 0x12, i have no idea where is it from. Its slave connection interval range,
- you need to add `pAdvertising->addServiceUUID(UUID);` where UUID is the uuid you want to be advertised and central will looking for, it may be `UUID =  BLEUUID(pService->getUUID())`.

If you will have more questions or issue you are welcome to ask.

menesesleonardo, 2017-11-13:
Hello @chegewara , thanks for your help, unfortunately no luck.

Here is the link to the code
https://pastebin.com/u8AKcuqP

[image]

As i showed before, i'd like to remove all that advertised data and leave only what is needed for the application. 
* 0x01
* 0x0A
* 0x07
* 0x09

chegewara, 2017-11-13:
Try like that:
```
  BLEAdvertising *pAdvertising = pServer->getAdvertising();
  pAdvertising->addServiceUUID(SERVICE_UUID);
 pAdvertising->start();
```

One more question. You provided link to ESP_32BLE_Arduino, its the repo where you from have this example. Is this repo where you from have BLE library? Because it is very outdated, and in this case my advise is to read this https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/ArduinoBLE.md, especially Replace paragraph.

menesesleonardo, 2017-11-13:
@chegewara it worked!!!! thank you!

[image]

I don't know why it publishes all the information twice, but the central can find it anyway.

Also, about the library, I followed windows instructions from the core page just one week ago,
https://github.com/espressif/arduino-esp32

Do you suggest I still need to delete and change all that?

I think this issue can be closed. I'll do it with your answer.

Thanks

chegewara, 2017-11-13:
It is advertised twice because its standard advertising and response advertising. It is not required but it is implemented in this library. Also Standard advertising and response advertising can differ, but again, in this library both have the same value.

It is not required to change library, but if you are using latest esp-idf then its good to change library with this repo. It is still developed and all the time new features are implemented, also some bugs are fixed.

closed 2017-11-13 by menesesleonardo

Phando, 2018-03-21:
Updating the library and modifying the advertising worked for me, thanks!!!!

planevina, 2022-11-11:
It's worked for me too. Thanks