nkolban/esp32-snippets #197: BLE server tx power

menesesleonardo, 2017-11-13:
Hello, I'm following this example of a BLE server

https://github.com/nkolban/ESP32_BLE_Arduino/tree/master/examples/BLE_server

however, the default tx power of this server is -21dBm.

I've seen that this value is standardized to -21(ultra low power) -15(low power), -7 and -4 (medium) and 0,1 and 4 (high power). Is it possible to adjust this value?

chegewara, 2017-11-14:
I think this is question to espressif tem, because i dont see any option to setup tx power in esp-idf stack.

nkolban, 2017-11-14:
While I believe you are after the function on Arduino BLE, we have implemented a solution for ESP-IDF.  What this means is that the next time we cut an Arduino build (or if you build an Arduino BLE library yourself) it should be present.  We have implemented:

```
BLEDevice::setPower(powerLevel)
```

This sets the "default" power level of the ESP32.  The available choices are as documented here:

http://esp-idf.readthedocs.io/en/latest/api-reference/bluetooth/controller_vhci.html?highlight=esp_ble_tx_power_set#_CPPv220esp_ble_tx_power_set20esp_ble_power_type_t17esp_power_level_t

d3cline, 2017-11-17:
I tried building from source and adding this, but I can't seem to figure it out, what am I doing wrong, 

The file states, 

`
/**
 * @brief Set the transmission power.
 * The power level can be one of:
 * * ESP_PWR_LVL_N14
 * * ESP_PWR_LVL_N11
 * * ESP_PWR_LVL_N8
 * * ESP_PWR_LVL_N5
 * * ESP_PWR_LVL_N2
 * * ESP_PWR_LVL_P1
 * * ESP_PWR_LVL_P4
 * * ESP_PWR_LVL_P7
 * @param [in] powerLevel.
 */
`

So I add, 

`
  BLEDevice::setPower(ESP_PWR_LVL_P1);
`
which renders, 

```
In file included from /home/d3cline/devl/Arduinos/Luci-ESP32/Luci-ESP32.ino/Luci-ESP32.ino.ino:23:0:
/home/d3cline/Arduino/libraries/ESP32_BLE/src/BLEDevice.h: In function 'void setup()':
/home/d3cline/Arduino/libraries/ESP32_BLE/src/BLEDevice.h:56:14: error: 'static void BLEDevice::setPower(esp_power_level_t)' is private
  static void setPower(esp_power_level_t powerLevel);
              ^
Luci-ESP32.ino:101: error: within this context
   BLEDevice::setPower(ESP_PWR_LVL_P1);
                                     ^
exit status 1
within this context
```

chegewara, 2017-11-17:
You do nothing wrong. For some reason, @nkolban, this is private function https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/BLEDevice.h#L56

d3cline, 2017-11-17:
Since it was a private function I decided to just edit line 272 of BLEDevice.cpp,

`	esp_err_t errRc = ::esp_ble_tx_power_set(ESP_BLE_PWR_TYPE_DEFAULT, ESP_PWR_LVL_P1);`

This is of course very hacky, but it builds. Will report my findings if it fixes radio issues.

d3cline, 2017-11-17:
Adjusting these values does seem to change dBm using nfc toolbox to verify, and it seems to reduce the bug I was having, fyi.

chegewara, 2017-11-17:
I did quick fix and PR is ready to merge. Thanks for finding it.

closed 2017-11-17 by nkolban

chegewara, 2018-07-29:
`N` stands for negative, `P` stands for positive.
`ESP_PWR_LVL_N14` is negative 14 (-14dBm)
`ESP_PWR_LVL_P7` is positive 7 (+7dBm)

Andreafoxalarm, 2019-02-15:
Good morning
I'm using the ESP32 WROOM module with AT commands.
I'm using the module as a BLE Server, but I've seen that the maximum distance I reach is 10m.
I saw that it is possible to calibrate the power of the module via the PHY_INIT_DATA.BIN file but I was not able to bring the distance of the bluetooth above 10m.
can you help me?
thanks 
Best regards

chegewara, 2019-02-15:
hi @Andreafoxalarm 
i think you will get best answer when you ask on esp-idf issue tracker.

adityathakekar, 2020-02-20:
How to reduce the range of esp32 wroom ble beacon?I tried all the above power levels.

chegewara, 2020-02-22:
Try:
`esp_ble_tx_power_set(ESP_BLE_PWR_TYPE_ADV, ESP_PWR_LVL_P1);`

adityathakekar, 2020-02-22:
@chegewara thanks . That works. Can we tune it to reduce even further below N14?

chegewara, 2020-02-23:
https://docs.espressif.com/projects/esp-idf/en/v3.3.1/api-reference/bluetooth/controller_vhci.html#_CPPv415ESP_PWR_LVL_N14

adityathakekar, 2020-02-24:
yes read those documents. That's the lowest I feel. Need to adjust the reading at the scanning app.

TheUruguayo, 2020-10-16:
Hi everyone, 
maybe someone can explain this to me: I've tried all of the configurations below, but... if there is a difference, is tiny. Even if I set them  from N14 to P9. Just in case, I'm pasting (in arduino ide) each of the lines, one at a time.
esp_ble_tx_power_set(ESP_BLE_PWR_TYPE_CONN_HDL0, ESP_PWR_LVL_P7);
esp_ble_tx_power_set(ESP_BLE_PWR_TYPE_CONN_HDL0, ESP_PWR_LVL_N14);
esp_ble_tx_power_set(ESP_BLE_PWR_TYPE_CONN_HDL0, ESP_PWR_LVL_P9);
esp_ble_tx_power_set(ESP_BLE_PWR_TYPE_CONN_HDL0, ESP_PWR_LVL_N0);

chegewara, 2020-10-16:
https://docs.espressif.com/projects/esp-idf/en/v3.3.1/api-reference/bluetooth/controller_vhci.html#_CPPv420esp_ble_power_type_t

TheUruguayo, 2020-10-16:
> https://docs.espressif.com/projects/esp-idf/en/v3.3.1/api-reference/bluetooth/controller_vhci.html#_CPPv420esp_ble_power_type_t

Thank you, that's where I checked before posting. I'm missing something?. Did I miss something there? thats the only line that i added to de code lets say this one: esp_ble_tx_power_set(ESP_BLE_PWR_TYPE_CONN_HDL0, ESP_PWR_LVL_P9 ); but theres not much difference with the Rssi lvl that i'm measuring before and after that.

chegewara, 2020-10-16:
I think that descriptions are very clearly in that docs. Which part you dont understand?

TheUruguayo, 2020-10-16:
> I think that descriptions are very clearly in that docs. Which part you dont understand?

Thank you for replaying. What I am seeing when I change the values ,is very little difference it the RSSI thati measuring, so maybe I am doing something wrong. Is it ok to just paste this esp_ble_tx_power_set(ESP_BLE_PWR_TYPE_CONN_HDL0, ESP_PWR_LVL_P9 ); in the void setup{}.
Im trying to go deeper with the esp32 but Im a engineer student, so in short a newbie.

chegewara, 2020-10-16:
> Is it ok to just paste this esp_ble_tx_power_set(ESP_BLE_PWR_TYPE_CONN_HDL0, ESP_PWR_LVL_P9 ); in the void setup{}.

No, it is not ok. Again, please read carefully docs i posted earlier. Here is very clear statement:
> BLE tx power type ESP_BLE_PWR_TYPE_CONN_HDL0-8: for each connection, and only be set after connection completed. when disconnect, the correspond TX power is not effected.

**set after connection completed.**

You have to be connected to peer device, then TX power for that particular connection will be affected.

I am guessing you need this:
https://github.com/nkolban/esp32-snippets/issues/197#issuecomment-589961720
> ESP_BLE_PWR_TYPE_ADV : for advertising/scan response. ESP_BLE_PWR_TYPE_SCAN : for scan

TheUruguayo, 2020-10-16:
tha

> > Is it ok to just paste this esp_ble_tx_power_set(ESP_BLE_PWR_TYPE_CONN_HDL0, ESP_PWR_LVL_P9 ); in the void setup{}.
> 
> No, it is not ok. Again, please read carefully docs i posted earlier. Here is very clear statement:
> 
> > BLE tx power type ESP_BLE_PWR_TYPE_CONN_HDL0-8: for each connection, and only be set after connection completed. when disconnect, the correspond TX power is not effected.
> 
> **set after connection completed.**
> 
> You have to be connected to peer device, then TX power for that particular connection will be affected.
> 
> I am guessing you need this:
> [#197 (comment)](https://github.com/nkolban/esp32-snippets/issues/197#issuecomment-589961720)
> 
> > ESP_BLE_PWR_TYPE_ADV : for advertising/scan response. ESP_BLE_PWR_TYPE_SCAN : for scan

Thank you  for taking the time to answer much appreciated!!. It's seem that I do not fully understand what's going on yet, but this helped me. I will dig deeper.

commit dcdd635e (2017-11-17): Merge pull request #208 from chegewara/master

Quick fix #197