nkolban/esp32-snippets #198: ESP32 BLEclient.ino issue

Goutham8128, 2017-11-14:
Hi

I am using a sensor dongle which has sensor information in it. It is acting as a BLE server. I am using ESP32 acting as a BLE client module.I uploaded the BLE client code into the esp32 in Arduino IDE but after its execution it is just displaying the BLE server name and its service UUID. It is not able to read the characteristics of the BLE server. I want the BLE client(ESP32) to read the characteristics of the sensor dongle.
I have specified the Service UUID and specific characteristic UUID in the BLEclient.ino. But it is just reading the services but not able to display the sensor information inside the module(BLE server).

chegewara, 2017-11-14:
From this line https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/tests/BLETests/Arduino/BLE_client/BLE_client.ino#L78, when serviceUUID equals service UUID advertised by your dongle, esp32 should have connect to your server dongle. There may be one issue, if dongle is advertising more than one service because at the moment getServiceUUID() returns only one advertised UUID and its the first found. You may try to change this line to connect if dongle name is recognized.

Just to be sure. You are using library from this repo and not from official esp32/arduino?

Goutham8128, 2017-11-14:
Hi

As you said the line 78 is same in my code and Dongle is advertising one service and exhibits 3 to 4 characteristics. I have mentioned the specific characteristic UUID in the code. though it is not connecting.
And i cloned the libraries from esp32/arduino only. Shouldn't i do this?

chegewara, 2017-11-14:
Can you paste some logs from serial output?

Goutham8128, 2017-11-14:
[image]

chegewara, 2017-11-14:
Ok, now i assuming that you changed serviceUUID in BLE_client code to match that one from dongle, which is 0x2ea7. Can you try to use whole advertised serviceUUID as a string? its 00002ae7-.....

Goutham8128, 2017-11-14:
no I used the whole UUID
[image]

chegewara, 2017-11-14:
On the first screenshot serviceUUID looks different. Compare it and change in your code please.

Goutham8128, 2017-11-14:
The service UUID which is shown in the console window is not correct.Actually the service UUID which i mentioned in the code is correct, this I have verified with the Sensor dongle documentation.But when i upload the code to ESP-32 the console window is showing the different service UUID.How is it possible?
And when i run the BLEscan.ino code of arduino/esp32 library i have got the same output i.e the same service UUID which i have got in the output console of the BLEclient.ino.

chegewara, 2017-11-14:
This mean we have issue with library. Requires investigations. Be patience please and thank you for all provided info.

Goutham8128, 2017-11-14:
Sure and thanks for your guidance

Goutham8128, 2017-11-15:
Hi

[image]

i am using the arduino/esp32 library .I followed the procedure mentioned in Neil kolban's book for esp32. what do u mean by using this repo? Both libraries aren't same?

chegewara, 2017-11-15:
No they are not the same. Arduino library last time was updated about month ago and there has been made a lot changes and bug fixes since then. You may want to read this https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/ArduinoBLE.md

Goutham8128, 2017-11-15:
I have done this steps.The BLEclient.ino is compiled with no errors but not displaying the correct service UUID and not getting connected to the characteristics.So,What might be the problem for that?

chegewara, 2017-11-15:
This may be some issue with library, i will have to investigate this. I dont have this dongle,  but i will try to use second esp32 to emulate it.

Goutham8128, 2017-11-15:
OK. thank you.

issue with the BLEclient.h?

chegewara, 2017-11-15:
More like with BLEUUID, just guessing. Or RemoteService. There is few possibilities. Or maybe just BLEClient.ino

Goutham8128, 2017-11-15:
ok

chegewara, 2017-11-15:
Like i said, i dont have that dongle, but i did quick test with BLE_client connected to BLE_server and it works.  Here you can find listings ive been using to test https://github.com/chegewara/esp32-tests-library/tree/master/arduino/.

In your case you only need to change those lines to match your needs https://github.com/chegewara/esp32-tests-library/blob/master/arduino/BLE_client_uuid_advertising/BLE_client_uuid_advertising.ino#L12-L14

Goutham8128, 2017-11-16:
thank you. I will get back to you with the result.

Goutham8128, 2017-11-16:
Hi

I changed those two lines and added #define SERVICE_UUID and #define CHARACTERISTIC_UUID to my code but there is no change.It is displaying another service UUID and not getting connected to the characteristics.

Is this code only making two esp32's act as client and server or can i use other module as Server to send data to esp32?

chegewara, 2017-11-16:
Oh no, for sure you can use any module, any BLE peripheral to connect with esp32. Just there is some glitch i cant debug without device. Maybe @nkolban will have one and can help with this issue. Anyway, i will try something else later today.

Goutham8128, 2017-11-16:
ok.
Try to help me with this.Thank you.

Goutham8128, 2017-11-17:
[image]

Hi
In Neil Kolban's book ,it is mentioned that the service UUID format is in the form of which i highlighted in the image. Only the first 2 bytes are needed to identify the type of service. So it is matched with the output in the serial monitor console i.e 0002ea7. So I think in the code the characteristic function is not getting executed it is getting stopped at the services loop.
So should I do any changes in the code to get connected to the characteristics of the BLE server?

chegewara, 2017-11-17:
Well known services are 16 bit, just like in your case https://github.com/nkolban/esp32-snippets/issues/198#issuecomment-344175281
BLE_client is nice example code. You have this line https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/tests/BLETests/Arduino/BLE_client/BLE_client.ino#L81, and if its not printed in serial monitor then you need to move few lines back and check why. With help comes this line: https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/tests/BLETests/Arduino/BLE_client/BLE_client.ino#L75
Like i said few days ago there is a difference between service UUID in your code, you said its service uuid from manual, and service UUID printed in serial monitor recognised by BLE scan.

Try to change this line https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/tests/BLETests/Arduino/BLE_client/BLE_client.ino#L78

with this code:
`if (advertisedDevice.haveServiceUUID() && advertisedDevice.getServiceUUID().equals(BLEUUID((uint16_t)0x2ea7))) {`

Goutham8128, 2017-11-17:
Well, this is the output.Sensor is connected to the server but not getting connected to the characteristics.

[image]

Goutham8128, 2017-11-17:
I think just the Serial.print is getting executed not the main loop function.
because it is getting error as 0x003e which means connection is not established

chegewara, 2017-11-17:
Now you have to enumerate characteristics to see what UUID they have or using ie nRF on android device just check what characteristics UUID are. This is example how i am doing this. Of course you can always install this app https://github.com/nkolban/ESP32_Explorer and use its functionality to enumerate characteristics.

chegewara, 2017-11-17:
I missed one thing from this screenshot. There is no output from loop:
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/tests/BLETests/Arduino/BLE_client/BLE_client.ino#L115
or this 
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/tests/BLETests/Arduino/BLE_client/BLE_client.ino#L118

Goutham8128, 2017-11-17:
We have 10 characteristics on this service 0x2ea7.Each characteristic is having individual 128bit UUID's  which we have verified with the nRFconnect. One of the characteristic UUID is  2ea78970-7d44-44bb-b097-26183f402401

Goutham8128, 2017-11-17:
And in the output window nothing is missed.We have not got output of either of that two lines 115 and 118. The line connected to server indicates the line 34 and 38 are printed but the main loop is not getting executed.If it does then the line 115 would get executed. I think the main loop is not getting executed.

chegewara, 2017-11-17:
The main loop is executed, just code cant pass thru this line https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/tests/BLETests/Arduino/BLE_client/BLE_client.ino#L41

We can join iirs if you wish, will be faster to talk

Goutham8128, 2017-11-17:
yeah
Sure i will get back to you by 3pm.I came for lunch.Sorry for the delay.

chegewara, 2017-11-17:
Well, its 9 am. I have to get some sleep now.

Goutham8128, 2017-11-17:
ok

what is your time zone?

chegewara, 2017-11-17:
Whenever you are ready. Im online for next few hours

nkolban, 2017-11-18:
I'm checking in for duty as well.  Its thanks giving holidays for the next 9 days ... so ping on IRC in channel `#ESP32` if we can offer assistance.

Goutham8128, 2017-11-18:
[image]

nkolban, 2017-11-18:
Are you available for a link screen share/skype?  neil.kolban is my skype

nkolban, 2017-11-18:
I chatted extensively with the raiser.  We saw that the ESP32 is indeed receiving an indication that the UUID is 16bit however when looking at the advertisement through nRFConnect and through Linux, we see that it appears to be advertising a 128bit UUID.  The discrepancy is odd to say the least.  Our next plan is to have the user install the latest ESP-IDF and the latest version of the BLE libraries.  Then, with debugging on, run it again.  Assuming that the problem is still present, we will add some code to the library to dump (in hex) the advertising message and see what it looks like from a raw data perspective.  From there we should be able to manually walk through it and see what we find.

chegewara, 2017-11-19:
https://github.com/nkolban/esp32-snippets/issues/172#issuecomment-345485748

Goutham8128, 2017-11-20:
Hi
Thanks for the assistance.Sorry for the delay.I was engaged with some other work. I Checked that reference link which you have sent.
I will try and get back to you.

But I am able to get connected to the server only for the first time when you told me to change the code in line no 41 i.e assigning long UUID and in line no 78 short UUID. When i have done that changes it got connected but when i executed it again it is again the same case which i have told you earlier.

Goutham8128, 2017-11-20:
@nkolban  yeah sure we will connect on IRC when you get free time. I will plan accordingly.

chegewara, 2017-11-20:
@Goutham8128 No worries, there is no rush.

Goutham8128, 2017-11-20:
I am able to get connected to the server only for the first time when I changed the line 41(assigning short UUID) and line 78 (assigning Long UUID) as you told. When I did this server is connected only for the first time.the screenshot I have send above.Help me out with this.

Goutham8128, 2017-11-21:
Hi

Did u check out with this issue?

chegewara, 2017-11-21:
@Goutham8128 Can we join iirc?

Goutham8128, 2017-11-21:
yeah in 5 mins

Goutham8128, 2017-11-21:
[image]

Goutham8128, 2017-11-21:
[image]

Goutham8128, 2017-11-21:
[image]

Goutham8128, 2017-11-23:
Any updates ? @nkolban @chegewara

chegewara, 2017-11-23:
Ok, another idea. When you connect with esp32 to dongle and then disconnect, you cant connect again. But can you connect with nrfconnect? With different configurations:
1. connect with esp32, disconnect esp32 and try to connect android,
2. connect with nrfconnect hen disconnect and again try to connect with nrfconnect,
3. connect with nrfconnect, disconnect and try to connect with esp32.

More info and logs will give more chance to track issue.

Goutham8128, 2017-11-27:
I had done all the three cases.In 
Case1: After connecting with esp32 the dongle is still advertising .This I verified with nrfconnect as it is appearing in scan
Case2: Once connected with nrfconnect and after disconnecting we are succesfully getting reconnected to nrfconnect.
Case3:After connecting and disconnecting with nrfconnect , I tried to connect with esp32 but there is no change in the output as it is displaying "connected to server" and the UUID below in the output log.

chegewara, 2017-11-27:
1. yes, it is advertising, but did you try to connect with nrfconnect when esp32 is still connected?

3. as we know, esp32 can connect at least one time with no issue, the issue is when we try to connect second time. im trying to understand if this is the case when you reset all devices, then connect with nrfconnect and only nrfconnect, and then when nrfconnect is still connected, can you connect with esp32 that first time when it should connect with no issue

nkolban, 2017-11-27:
I ordered a dongle/device and am awaiting delivery for testing.

Goutham8128, 2017-11-27:
okay! When might be the delivery?

nkolban, 2017-11-27:
I believe I ordered on Tuesday ... this is Sunday ... assuming its coming from a domestic supplier and they have stock, I'd assume I should have it by this time next week.  I haven't seen any emails relating to shipment.

Goutham8128, 2017-11-27:
okay fine!

nkolban, 2017-11-27:
well shoot ... was going through issues and re-read this one.  I thought it "odd" that I hadn't received a confirmation receipt ... have gone back through my credit card history and there is no indication that I purchased the device.   I remember a conversation on IRC (that is now gone) where we speced out the identity of the device ... but I don't have that anymore.  Can you post back the identity of the device and I'll order another?

nkolban, 2017-11-27:
Okay ... found it:

https://www.mouser.com/ProductDetail/Dialog-Semiconductor/DA14583IOTSENSDNGL/?qs=sGAEpiMZZMvzWM6X48YRJD1x4Z6Y1PCOCKLqamhNUmQ%3d

I have now ordered and THIS time I got a positive confirmation by email ... so its on its way.  Lets imagine 10 days from today.

[21 more replies]

closed 2018-11-15 by chegewara