nkolban/esp32-snippets #201: What data format expect?

SuzukiYushin, 2017-11-15:
I'm developing a iPhone app by Unity.
And I using this asset.

https://github.com/jerrytron/neverending-story

It sends a byte array.
My device(Lolin32) seems to be receive some data.
But it shows up different number.

Do you have any Idea for solution?
Thanks

`#include <BLEDevice.h>
#include <BLEUtils.h>
#include <BLEServer.h>
#include <BLE2902.h>

#define SERVICE_UUID        "4fafc201-1fb5-459e-8fcc-c5c9c331914b"
#define CHARACTERISTIC_UUID "beb5483e-36e1-4688-b7f5-ea07361b26a8"

BLECharacteristic *pCharacteristic;

bool deviceConnected = false;

class MyServerCallbacks: public BLEServerCallbacks {
    void onConnect(BLEServer* pServer) {
      deviceConnected = true;
    };

    void onDisconnect(BLEServer* pServer) {
      deviceConnected = false;
    }
};

class MyCallbacks: public BLECharacteristicCallbacks {
    void onWrite(BLECharacteristic *pCharacteristic) {
      
      char sNum[20] = "55";
      std::string value = pCharacteristic->getValue();

      if (value.length() > 0) {
        Serial.println("*********");
        Serial.print("New value: ");
        for (int i = 0; i < value.length(); i++)
          Serial.print(value[i]);

        Serial.println();
        Serial.println("*********");
      }
    }
};

void BLEInit() {
  BLEDevice::init("wind9");
  BLEServer *pServer = BLEDevice::createServer();
  pServer->setCallbacks(new MyServerCallbacks());
  BLEService *pService = pServer->createService(SERVICE_UUID);
  pCharacteristic = pService->createCharacteristic(
                      CHARACTERISTIC_UUID,
                      BLECharacteristic::PROPERTY_READ   |
                      BLECharacteristic::PROPERTY_WRITE  |
                      BLECharacteristic::PROPERTY_NOTIFY |
                      BLECharacteristic::PROPERTY_INDICATE
                    );

    pCharacteristic->addDescriptor(new BLE2902());

  pCharacteristic->setCallbacks(new MyCallbacks());

  pCharacteristic->setValue("Hello World");
  pService->start();

  pServer->getAdvertising()->start();
}

void BLEUpdate(float windDegree) {
    if (deviceConnected) {
      char sNum[20] = "55";
      dtostrf(windDegree, 5, 2, sNum);
      pCharacteristic->setValue(sNum);
      pCharacteristic->notify();
      //Serial.println(sNum);
    }
    delay(200);
}
`

nkolban, 2017-11-15:
Howdy.  I understand you are developing an iPhone app using the Unity framework.  You say ... "it sends a byte array" ... should I assume that it is sending it over BLE?  You say that your ESP32 receives some data.  You then say that it "shows up different number".  Does that mean the length of the data received is not what you expected or that the content of the data is not what you expected?

SuzukiYushin, 2017-11-15:
①I'm sending it over BLE.
②The content of the data is not what I expected.

When I send
        Value = (byte)201 or (byte)0x00C9
Result is like this symbol  → �

When I send
        Value = (byte)55
Result is  → 7

Here is my sending part of Unity(C#).

`
	void SendByte (byte value)
	{
		byte[] data = new byte[] { value };

		BluetoothLEHardwareInterface.WriteCharacteristic (_connectedID, _serviceUUID, _writeCharacteristicUUID, data, data.Length, true, (characteristicUUID) => {
			BluetoothLEHardwareInterface.Log ("Write Succeeded");
		});
	}
`

nkolban, 2017-11-18:
Howdy ... sorry for the delay ... but I think I see the problem.

The data you are receiving in an array of bytes ... a byte is 8 bits of binary data ... decimal 0 to decimal 255.  When you code:

```
Serial.print(value[i]);
```
we need to look at the data type of value and we will find it is an array of bytes that are treated as characters.  If the byte value is decimal 55 then it will be printed as ASCII character '7' (see the following table  http://www.asciitable.com/)

If what you want to do is print the decimal value of the byte, then try the following:

```
Serial.print((int)value[i]);
```

This should cast/convert the byte/character value to an integer which is then printed in decimal (by default).

SuzukiYushin, 2017-11-20:
Thanks for the assistance!
It works well perfectly.

closed 2017-11-20 by nkolban