nkolban/esp32-snippets #205: Analog read + BLE

SuzukiYushin, 2017-11-15:
Sorry another question.
AnalogRead value does not be change(like 4095) when I use BLE same time.

Do you have any idea?

`
/*
    Video: https://www.youtube.com/watch?v=oCMOYS71NIU
    Based on Neil Kolban example for IDF: https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/tests/BLE%20Tests/SampleNotify.cpp
    Ported to Arduino ESP32 by Evandro Copercini

   Create a BLE server that, once we receive a connection, will send periodic notifications.
   The service advertises itself as: 4fafc201-1fb5-459e-8fcc-c5c9c331914b
   And has a characteristic of: beb5483e-36e1-4688-b7f5-ea07361b26a8

   The design of creating the BLE server is:
   1. Create a BLE Server
   2. Create a BLE Service
   3. Create a BLE Characteristic on the Service
   4. Create a BLE Descriptor on the characteristic
   5. Start the service.
   6. Start advertising.

   A connect hander associated with the server starts a background task that performs notification
   every couple of seconds.
*/
#include <BLEDevice.h>
#include <BLEServer.h>
#include <BLEUtils.h>
#include <BLE2902.h>

BLECharacteristic *pCharacteristic;
bool deviceConnected = false;
uint8_t value = 0;

// See the following for generating UUIDs:
// https://www.uuidgenerator.net/

#define SERVICE_UUID        "4fafc201-1fb5-459e-8fcc-c5c9c331914b"
#define CHARACTERISTIC_UUID "beb5483e-36e1-4688-b7f5-ea07361b26a8"

class MyServerCallbacks: public BLEServerCallbacks {
    void onConnect(BLEServer* pServer) {
      deviceConnected = true;
    };

    void onDisconnect(BLEServer* pServer) {
      deviceConnected = false;
    }
};

void setup() {
  Serial.begin(115200);

  // Create the BLE Device
  BLEDevice::init("MyESP32");

  // Create the BLE Server
  BLEServer *pServer = BLEDevice::createServer();
  pServer->setCallbacks(new MyServerCallbacks());

  // Create the BLE Service
  BLEService *pService = pServer->createService(SERVICE_UUID);

  // Create a BLE Characteristic
  pCharacteristic = pService->createCharacteristic(
                      CHARACTERISTIC_UUID,
                      BLECharacteristic::PROPERTY_READ   |
                      BLECharacteristic::PROPERTY_WRITE  |
                      BLECharacteristic::PROPERTY_NOTIFY |
                      BLECharacteristic::PROPERTY_INDICATE
                    );

  // https://www.bluetooth.com/specifications/gatt/viewer?attributeXmlFile=org.bluetooth.descriptor.gatt.client_characteristic_configuration.xml
  // Create a BLE Descriptor
  pCharacteristic->addDescriptor(new BLE2902());

  // Start the service
  pService->start();

  // Start advertising
  pServer->getAdvertising()->start();
  Serial.println("Waiting a client connection to notify...");
}

int sensorP = 0;

void loop() {

  if (deviceConnected) {
    Serial.printf("*** NOTIFY: %d ***\n", value);
    pCharacteristic->setValue(&value, 1);
    pCharacteristic->notify();
    //pCharacteristic->indicate();
    value++;
  }
  delay(200);
  
      sensorP = analogRead(15);
    Serial.println(sensorP);
}
`

nkolban, 2017-11-15:
What we will need to do is examine how the Arduino API called analogRead() works.  Presumably it uses either ESP-IDF APIs or reading from lower level registers.  Once we see how analogRead() works, we will then search the ESP32 forum and issues and see if there are known puzzles in this area.  These BLE C++ classes use the exposed ESP-IDF APIs.  What that means to us is that there isn't anything in the code of this repository that either permits nor denies Analog usage.  It may turn out to be a limitation of the ESP32 and BLE drivers.

chegewara, 2017-11-15:
We have to remember it is not Arduino UNO nor any arduino device. We need to configure analog pins before we can ead from it. Most likely its just a attenuation, but i may be wrong. 

https://www.youtube.com/watch?v=CbacRysML5c

SuzukiYushin, 2017-11-17:
Thank you for you're advice.
I'll try to use ESP-IDF.

chegewara, 2017-11-17:
You can do it with arduino, its not a problem.

https://www.youtube.com/watch?v=RlKMJknsNpo

chegewara, 2017-11-19:
Ok then. This is working code, but im suggesting to play with it for a while.
It is setup to read from pin 36. Also this is good start to play with analog inputs http://esp-idf.readthedocs.io/en/latest/api-reference/peripherals/adc.html. Any other questions are welcome.

```
#include <BLEDevice.h>
#include <BLEServer.h>
#include <BLEUtils.h>
#include <BLE2902.h>
#include <driver/adc.h>

BLECharacteristic *pCharacteristic;
bool deviceConnected = false;
uint8_t value = 0;

#define SERVICE_UUID        "4fafc201-1fb5-459e-8fcc-c5c9c331914b"
#define CHARACTERISTIC_UUID "beb5483e-36e1-4688-b7f5-ea07361b26a8"

class MyServerCallbacks: public BLEServerCallbacks {
    void onConnect(BLEServer* pServer) {
      deviceConnected = true;
    };

    void onDisconnect(BLEServer* pServer) {
      deviceConnected = false;
    }
};

void setup() {
  Serial.begin(115200);

    adc1_config_width(ADC_WIDTH_BIT_12);
    adc1_config_channel_atten(ADC1_CHANNEL_0,ADC_ATTEN_DB_11);
 
  // Create the BLE Device
  BLEDevice::init("MyESP32");

  // Create the BLE Server
  BLEServer *pServer = BLEDevice::createServer();
  pServer->setCallbacks(new MyServerCallbacks());

  // Create the BLE Service
  BLEService *pService = pServer->createService(SERVICE_UUID);

  // Create a BLE Characteristic
  pCharacteristic = pService->createCharacteristic(
                      CHARACTERISTIC_UUID,
                      BLECharacteristic::PROPERTY_READ   |
                      BLECharacteristic::PROPERTY_WRITE  |
                      BLECharacteristic::PROPERTY_NOTIFY |
                      BLECharacteristic::PROPERTY_INDICATE
                    );

  pCharacteristic->addDescriptor(new BLE2902());

  // Start the service
  pService->start();

  // Start advertising
  pServer->getAdvertising()->start();
  Serial.println("Waiting a client connection to notify...");
}

int sensorP = 0;

void loop() {

  if (deviceConnected) {
    Serial.printf("*** NOTIFY: %d ***\n", value);
    pCharacteristic->setValue(&value, 1);
    pCharacteristic->notify();
    //pCharacteristic->indicate();
    value++;
  }
  delay(200);
    int val = adc1_get_raw(ADC1_CHANNEL_0);
 
      sensorP = analogRead(36);
    Serial.println(sensorP);
    Serial.println(val);
}
```

SuzukiYushin, 2017-11-20:
Thank you for you're advice!
I'm using this module.So I can get values from "SVP" as 36pin.

https://ja.aliexpress.com/store/product/WEMOS-Project-LOLIN32-ESP32-wifi-bluetooth-basis-ESP-32-learning-kit/2787042_32813879774.html

chegewara, 2017-11-20:
You can use any ADC1 pin. I did use of pin 36 cause i just copy/paste from esp-idf docs. Here you can see what pins have ADC capability. Any pin ADC1_0 to ADC1_7. Pins ADC2 has some cons, but you can also use some of them.
https://raw.githubusercontent.com/gojimmypi/ESP32/master/images/myESP32%20DevKitC%20pinout.png

esridhar126, 2018-02-12:
Hi everyone, I am working on the BLE 4.0, HM-10(CC2540/CC2541) module for my product updation from Bluetooth 2.0. Main controller i am using is Arduino Uno/ Mega. For 2.0 I easily got Firmata from Arduino examples, but BLE4.0 is more advanced one having many specs to interface, i cant get a Firmata from anywhere. In arduino also its available for minimum modules that doesnot  include Arduino mega/Uno.
Is anyone from here suggest me or help how to interface my Microcontroller with API/ APk apps with or without using Standard Firmata BLE. (If it is with standard Firmata it is more helpful.)
My main motive is to control my product with IOS.