nkolban/esp32-snippets #206: Current best practice to update?

d3cline, 2017-11-16:
What is the current best practice to update or keep updated. I apologize since there is so much noise around this issue I wanted to ask and a ticket seemed like the sanest way. 

I am having an issue which I believe should be solved by tx power and I read https://github.com/nkolban/esp32-snippets/issues/197

which I believe is the path to a fix. But I am unsure what is the current best practice to keep up to date or be sure I am on the same version as that fix. I will backup my arduino environment before I make any changes to be able to roll back .

chegewara, 2017-11-16:
I believe this is best option for you and most arduino ide users https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/ArduinoBLE.md#replacing-the-version-that-comes-with-arduino-esp32

You can track this repo for any changes, and whenever you see that update is fixing any bugs or add feature that suits you you should update to latest version. Its just my opinion.

nkolban, 2017-11-18:
Mr @chegewara Is bang on.  There is no hard science as to when we refresh the Arduino libraries as distributed with the Arduino ESP32 release.  My opinion would be to wait until the official 3.0 release of the ESP-IDF.

d3cline, 2017-11-18:
I would wait if I could but I am developing a product which I need to be ready enough to demo to people, which this library does great. I am happy with dealing with a rough road as long as I can ask questions and provide feedback. I was able to get it to update using the build instructions.

closed 2017-11-18 by d3cline

nkolban, 2017-11-18:
Thanks for the kind words ... but do realize, that this is an open source, community project.  Use at your own risk.  If something goes foobar, we'll try and assist as best we can ... but there are no SLAs or assurances ... we all chip in as best we can.