nkolban/esp32-snippets #207: Crashing when using FreeRTOS task in cpp_utils

rweller59, 2017-11-16:
I've got the following code in main. Nothing else running. The task logs every 5 secs as you would expect. However, if I uncomment either of the lines below the startTask function, the program will crash.
In the commService::init() function, I commented everything out so its just returning. In the std::strcpy line, the variable is defined in a namspace in another file.

All of the code seems to run fine until I started playing with the freetros task.

void task_main(void *pvParameter) {
	while(1) {
		ESP_LOGI(TAG, "Hello from main task");
		FreeRTOS::sleep(5000);
	}
	FreeRTOS::deleteTask ();
}
void app_main(void)
{
	ESP_LOGI(TAG, "*********************************************** STARTING APP_MAIN");
	FreeRTOS::startTask(&task_main, "task_main", NULL, 512);
	//commService::init();
	//std::strcpy(config::network.wifi[0].ssid, "xxx");
	while(1){
		vTaskDelay(10 / portTICK_PERIOD_MS);
	}

chegewara, 2017-11-16:
Im not sure, but i think you need to change 

> void task_main(void pvParameter)
into 
`void task_main(void* pvParameter)`

The parameter is type void* but you have regular void

rweller59, 2017-11-16:
thanks. yep, caught that and I did get further into the code then when I first posted before it crashes. Trying to figure out where its crashing now. Funny, I've got all the code running but as soon as I add this task in, things start crashing. I'll post again as soon as I figure out where its crashing now. I'm using c++. Is there something I should be aware of using the freetros tasks?

chegewara, 2017-11-16:
Yes, there is one thing and impossible to debug. Task has to be in endless loop or have deleteTask(NULL) before exit. But i see its not your case. 
If you have more tasks its good to have something to turn task into blocked state at least for a moment, like delay or waiting on queue. Be aware about task stack size, minimum is 768 if i remember, but 1024 or even 2048 is good start.

rweller59, 2017-11-16:
Got your last message while typing. Changing to 2048 fixed the issue so far - I had figured since this loop was doing nothing I would be ok at 512 but guess not.
I started out trying to get this to work inside a class instance then switched to main just to get a simple task working. I'm going back to the class so will leave this open for a bit.

rweller59, 2017-11-16:
Ok, so I'm back to trying to create a task in class. Basically the same code as I had in main.
When I define function as:  static void serviceWIFI::task_mqttLoop(void *pvParameter) {}, I get a compile message that says: cannot declare member function 'static void

When I define the function without static, I get a compile error when declaring the task:  ISO C++ forbids taking the address of an unqualified or parenthesized non-static member function to form a pointer to member function - when I define as FreeRTOS::startTask(&task_mqttLoop, "task_mqttloop", NULL, 2048);

chegewara, 2017-11-16:
Been there many times before i understood. Yes, if you want to use class member function in startTask then that function must be static. But you cant define function as static, you can only declare it, usualy  in header file. Then definition is without static word.

rweller59, 2017-11-16:
so...
int header:  static void task_mqttLoop(void *pvParameter);
int cpp: void serviceWIFI::mqttLoop(void) {}
And that works? It does compile but I do not understand it.

chegewara, 2017-11-16:
In header file:
```
class serviceWIFI{
//code
    static void mqttLoop(void* param);
//code
}

```
In cpp file:
```
void serviceWIFI::mqttLoop(void* param){
//code
}
```
serviceWIFI is your class if i understand correct, and mqttLoop is member function.

rweller59, 2017-11-17:
yeh, that's what I did, and it does now compile. I've just always made consistent definitions in header and cpp so, I learn something new every day

rweller59, 2017-12-09:
I've been ignoring this related issue but now I've got to clean it up. As stated above, I'm using a static class member as my freetros task. This task is a monitor for the class, i.e., checking status of various class elements. In order to monitor the class I need access to the class instance object.

How can I pass the class object or an object function pointer to the static freetros task?

nkolban, 2017-12-09:
Let us assume you have a function called "myTaskEntryPoint" this is the entry point into your FreeRTOS task.  The function could be declared as:
```
void myTaskEntryPoint(void *arg) {
  // Do some task work ...
}
```
or it could be a static member of a class:
```
class MyClass {
public:
   static void  myTaskEntryPoint(void *arg) {
     // Do some task work ...
   }
}
```

Pause here and validate that you understand these notions and, if not, post back with what isn't clear.

Now let us assume that you want to be able to access a C++ class object within the task body.  The challenge here is "How do I address it" ...?  Imagine we have the following:

```
MyClass myClassInstance;
// Do something here
xTaskCreate(myTaskEntryPoint, ...);
```

How then does:

```
void myTaskEntryPoint(void *arg) {
  // Do some task work ...
  // Access the methods and properties of the instance in myClassInstance
}
```

In this story there are only two possibilities:

1. Your class instance is made a global variable.  If you define the class instance as either a global variable or create a pointer to the class instance as a global variable, then your task runner can access the class instance.

2. You pass in a pointer to the class instance.  Of the two, I like this one better ... let us now look at that.

Your code now becomes:

```
MyClass myClassInstance;
// Do something here
xTaskCreate(myTaskEntryPoint, ..., &myClassInstance, ...);
```

and your task handler now becomes:

```
void myTaskEntryPoint(void *arg) {
  MyClass *pMyClassInstance = (MyClass *)arg;
  // Do some task work ...
  // Access the methods and properties of the instance in myClassInstance
  pMyClassInstance->myMethod();
  int value = pMyClassInstance->myValue;
}
```

While we'll always try and assist as best we can, this question (as I understand it) has nothing to do with the code found in this repository and everything to do with C++ programming practices in general and programming in C++ in an ESP32 environment in particular.  The best place to post these questions is on the ESP32 forum found here:

https://esp32.com/index.php

rweller59, 2017-12-09:
ok. I got it. I like the second option as well.

Sorry about posting here rather than the espidf forum (or maybe this is just c++ stuff and not related to either). I'm trying to utilize as much of the cpp_utils library as possible and get a bit lost occasionally as to where the root of my issues/questions lie.

closed 2017-12-09 by rweller59

nkolban, 2017-12-09:
No problems ... this one I would consider to be a cpp_utils question as it explicitly asks "How do I use the Task" class.

rweller59, 2017-12-09:
thanks for the explanation. I can now call monitoring tasks, passing the class instance so that the task has access to class instance functions