nkolban/esp32-snippets #211: Descriptor callback class

chegewara, 2017-11-19:
We need BLEDescriptorCallback class with at least onWrite() method. Now we start to send notifications and indications when client connects to server, but this is not good approach. We also cant have an option to stop sending notifications when client is still connected but dont need/want to receive any notifications/indications. 
With descriptor callback we will be able to move start/stop notifications from server callback and to have the proper control on it.

nkolban, 2017-11-19:
If I understand correctly, we are looking for a BLEDescriptorCallbacks class that is similar in nature to the BLECharacteristicCallbacks class.  Sounds good.

nkolban, 2017-11-19:
The implementation of this function has been completed.

It follows the identical model for callbacks for characteristics.  Here is a sample test that I used to validate that it works:

https://pastebin.com/TyDqLGC3

chegewara, 2017-11-19:
Good work.

PS I lost half hour trying to implement it un arduino ble_uart example, just i forgot to export it to zip and re-import in arduino.

closed 2017-11-19 by chegewara

nkolban, 2017-11-19:
Oh man ... I'm sorry you lost work/time.  That sucks.  Use it as a learning experience maybe?

chegewara, 2017-11-19:
Its just my fault, i was preparing to implement those changes in example, i did merge changes and just forgot to export library to zip

pana216, 2019-06-20:
Hello there,
I have two questions. First of all I tried to modify the given code in the pastebin link mention from Mr  @nkolban earlier at this post. I changed following code: 
(working with ESP32-idf and eclipse on a ESP32DevKitCv4)

`void onRead(BLEDescriptor* pDescriptor) {
// Do something before the read completes. 
	  pDescriptor->setValue("hello");
  	ESP_LOGD(LOG_TAG, "Descriptor onRead called");
  }`

What I expected to see is that the value of the description changes when clicking the "arrow down button" on the nRF Connect app, but it doesn't (see screenshot below): 
[image]

This is what my Terminal (ESP_LOG shows):

D (394641) BLEDevice: gattServerEventHandler [esp_gatt_if: 3] ... ESP_GATTS_READ_EVT
V (394641) BLEUtils: GATT ServerEvent: ESP_GATTS_READ_EVT
V (394641) BLEUtils: [conn_id: 0, trans_id: 13, bda: 79:45:6e:01:c2:b7, handle: 0x2b, is_long: 0, need_rsp:0]
D (394651) BLEServer: >> handleGATTServerEvent: ESP_GATTS_READ_EVT
D (394661) BLECharacteristic: >> handleGATTServerEvent: ESP_GATTS_READ_EVT
D (394671) SampleServer_Callbacks: Descriptor onRead called
D (394671) BLECharacteristic: << handleGATTServerEvent
D (394681) BLEServer: << handleGATTServerEvent

Second: What does the BLE2902.h do? At the beginning I just thought about creating a BLEDescriptor object, creating a Callback on that and change the value onRead. How can I deal with the BLE2902 but setting my own UUID like I can do it on the BLEDescriptor class?

Thank you in Advance!
Cheers 
Pana

chegewara, 2019-06-20:
Hi,
descriptor 2902 is special descriptor that allows ble client to tell if client want to receive notifications or indications or not. Only allowed values are 0x0, 0x0001 and 0x0002.

pana216, 2019-06-20:
Okay got it thank you very much! So I'll forget about the BLE2902 class and continue my work with the BLEDescriptor class. 
I wrote this code based on the BLEDescriptor but still the value of the Descriptor won't change. Did I miss anything?
[https://pastebin.com/0QHmgNP7](url)

chegewara, 2019-06-20:
https://www.bluetooth.com/specifications/gatt/descriptors/

pana216, 2019-06-20:
wow... sorry for disturbing you - I misunderstood the definition of the descriptor