nkolban/esp32-snippets #213: WebServer.cpp compile errors

yaoyogi, 2017-11-20:
Thank you for publishing your code.  After a pull I got a couple of compile errors:

After a quick look at last checkin for chunking a couple of compile errors resulted:

1. duplicate definitions of WebServer::HttpResponse::sendData(...)

2. WebServer::HttpResponse::sendChunkHead()  {
     ....
    mg_send_head(m_nc, m_status, -1, m_headers.c_str());

m_headers is a std::map but I don't see c_str() defined for it

nkolban, 2017-11-20:
Thank you sir ... recreated ... studying how we got here and fixing now.

nkolban, 2017-11-20:
Hmmm ... since this is an open source project and anyone can make changes ... which is good ... we find someone did .. and didn't get it quite right.  For example:

https://github.com/nkolban/esp32-snippets/commit/7c136cbfd5d21e36f0fd96395cb68c01a96d1552

chegewara, 2017-11-20:
Nice loop from "type casting"

nkolban, 2017-11-20:
Code changes made.  The two changes were the deletion of the duplicate function ... which was identical in both cases.  And the second was an attempt to send headers by asking a "map" to turn itself into a string.  That wouldn't work.  Since we build a string version of the headers elsewhere, I factored that out into its own function called buildHeaders() which returns a std::string and then we ask that for its NULL terminated string representation.  All now compiles cleanly.

nkolban, 2017-11-20:
When you get a chance, please test again.  If there are problems related to this fix, post them here.  If there are problems from other parts, create a new issue.  If you are delighted and all works as you desire, close this request when ready.

yaoyogi, 2017-11-21:
Fixes address compilation issue.  Thank you.

closed 2017-11-21 by yaoyogi

jonface, 2017-11-23:
I realise this is closed, but a clean checkout of the esp-idf (today), I'm getting,

```
In file included from C:/Users/jmasters/Downloads/esp32_win32_msys2_environment_and_toolchain-20170918/msys32/home/jmasters/esp/gatt_client/main/gattc_demo.c:4:0:
C:/Users/jmasters/Downloads/esp32_win32_msys2_environment_and_toolchain-20170918/msys32/home/jmasters/esp/gatt_client/components/cpp_utils/BLEUtils.h:15:18: fatal error: string: No such file or directory
compilation terminated.
make[1]: *** [/home/jmasters/esp/esp-idf/make/component_wrapper.mk:243: gattc_demo.o] Error 1
make: *** [C:/Users/jmasters/Downloads/esp32_win32_msys2_environment_and_toolchain-20170918/msys32/home/jmasters/esp/esp-idf/make/project.mk:448: component-main-build] Error 2

```

I've hijacked the gatt_client example in case you were wondering, I've looked in menuconfig, do I need to enable the STL somehow?

Thanks