nkolban/esp32-snippets #217: WebServer.cpp panics consistently

yaoyogi, 2017-11-21:
The attached example file is a minimal webserver handling a simple GET request.  I use a Java HTTP client making HTTP requests to the server.  The single Java test client is making calls every 200ms to 2000ms randomly.  The stacktraces are seen consistently and after after a short time (10 to 100 requests).   

Environment:
Mac OS X 10.12.2 (Sierra)
 Latest IDF pull (about 2 weeks back)
 Latest pull of the Kolban cpp_utils (as of this writing)
The ESP32 is in STA mode and connected to my local WiFi network. 

[ex_kolban_webserver_bug_stack_traces.txt](https://github.com/nkolban/esp32-snippets/files/1490550/ex_kolban_webserver_bug_stack_traces.txt)

[ex_kolban_webserver_bug.cpp.txt](https://github.com/nkolban/esp32-snippets/files/1490545/ex_kolban_webserver_bug.cpp.txt)

The stacktraces seem to be related to alloc/dealloc related to std::string use.  I did a cursory look at the code and didn't see anything immediately obvious.  However, my last few months with ESP32 IDF and C++ have turned up some interesting issues with std::xxx (mostly although not limited to std::string) types.  Namely, I see  what appear to be random alloc/dealloc problems like the one I am seeing here.  They're usually around cases where a class instance members holding onto stack allocated std::string's.  

I am generally aware of the various std::string assigning, passing, copying, value/ref etc... semantics, but empirically, I keep seeing problems with this idiom, in this environment.  When I remove such cases from my code (or others) or go to std::string* those ghost issues disappear.   I don't like code voodoo much, but this is what I am seeing.  I am guessing it is potentially environment/compiler issue on how it handles RVO or such (pure speculation and I am not a C++ compiler guy).

Just for grins, I made some changes to FreeRTOS .h/.cpp - Namely I left the std::string usage in the API's intact but internally don't hold onto to them in any members; I just hold onto their c_str() results.  Note this removes any of the copy/ref/value semantics.   I selected the FreeRTOS class as it was in the chain of the first stacktrace, and it has the std::string pattern I mentioned.  Those updated files are attached.   The test is now in the 10K successful request range and still running...

[FreeRTOS.cpp.txt](https://github.com/nkolban/esp32-snippets/files/1490678/FreeRTOS.cpp.txt)

[FreeRTOS.h.txt](https://github.com/nkolban/esp32-snippets/files/1490675/FreeRTOS.h.txt)

My compiler environment:
> c++ --version
Apple LLVM version 8.0.0 (clang-800.0.42.1)
Target: x86_64-apple-darwin16.3.0
Thread model: posix
InstalledDir: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin

> g++ --version
Configured with: --prefix=/Applications/Xcode.app/Contents/Developer/usr --with-gxx-include-dir=/usr/include/c++/4.2.1
Apple LLVM version 8.0.0 (clang-800.0.42.1)
Target: x86_64-apple-darwin16.3.0
Thread model: posix
InstalledDir: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin

chegewara, 2017-11-21:
That stacktrace is saying you have problem with stack size and there is heap corruption. I dont know why its problem with stack in HttpServerTask, the stack should be high enough, but try to change stack size in this line:
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/HttpServer.cpp#L87

yaoyogi, 2017-11-22:
Thank you for the quick response and recommendation.

After changing the stack size 32*1024, it is failing again in the previously mentioned ways.  It's just taking generally more requests before it panics, up to 180.  But I also still see the exact same failures/stacktraces with as few as 4 requests and usually well before 100.  At 4 requests, and that large of a stack, it seems like something else is amiss.

Do you have any ideas as to why, when I removed the std::string members from the FreeRTOS class (code was attached in previous comment), that this issue does not happen?  It seems to indicate that something is wrong with the handling of the std::string values and not necessarily with stack size.   

One of the most common stacktraces, and the first in my previous note, contains FreeRTOS.cpp:140 which is FreeRTOS::Semaphore::give() {
   ...
  m_owner = std::string("<N/A>");   // line 140
}

This line creates a std::string on the stack and then immediately goes out of scope.  Any future references to m_owner would be accessing invalid memory?

This time I just changed FreeRTOS::Semaphore:
    1. m_owner member to const char* 
    2. Three lines assigning "<N/A"> literal now
    3. Three lines that had std::string assignments I just use their c_str()

I then am able to run the example for many 1000's of requests without failure.   This is also with the original stack size for HttpServer.

nkolban, 2017-11-23:
One quick thought ... in your code you have:

```
void get_hello(HttpRequest *preq, HttpResponse *presp) {
	presp->sendData("bonjour ca va") ;
}
```

This is a callback handler for your client.  The thinking is that you are going to do work to "own/handle" the request.  And that includes determining if the request is complete.

What if you coded:

```
void get_hello(HttpRequest *preq, HttpResponse *presp) {
   presp->sendData("bonjour ca va") ;
   presp->close();
}
```

yaoyogi, 2017-11-23:
Thank you for the hint.

I will add the close()

As is, your framework must be helping in that regard somehow, as my tests
are in the 10k+ request range and all complete successfully!
My enclosing test framework checks the heap on each iteration and it didn't
register any leaks, but better to be complete as you suggest.

Thanks again for your work with the ESP 8266/32 community and have a great
Holiday!

    MrP

nkolban, 2017-11-23:
Thanks for providing the source of the application and the stack.  Looking carefully and slowly at the stack trace, it seems to show that the problem occurs at this source statement:

https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/SockServ.cpp#L233

Specifically, it appears when trying to create a std::string instance.  It also seems to show a memory assertion here:

https://github.com/espressif/esp-idf/blob/master/components/heap/multi_heap.c#L370

My "gut" is telling me the following ... and this could be VERY wrong ...

The HttpServer is multi threaded (tasks) and is allocating and releasing storage all over the place.  Within the ESP32 there is (logically) one chunk of memory called "the heap" from which are carved requested storage areas.  All the tasks must thus synchronize on these heap accesses.  I.e. if one task is allocating storage then if another task should also allocate storage it must wait for the first to complete before continuing.  Looking at the error, it almost "feels" like there might be a bug there and if requests are being made in parallel, we have a "window" in which a problem can be encountered where multiple tasks step on each other.  I'm going to try to write some tests to try and flush that out (assuming it exists).

nkolban, 2017-11-23:
I have fully recreated on my system.  Debugging now.

nkolban, 2017-11-25:
There is a phrase ... "two wrongs don't make a right".  I wrote a test and recreated what appeared to be your exact problem.  Perfect!!!  I then dug and dug and dug ... and eventually found a problem.  I made some changes and then all worked GREAT.    Like you, I was getting an exception about the 60-70 tests mark.  After making the changes, I have now run thousands of REST requests through without incident.  However ... here's the thing.  The change I had to make was in how I was managing WiFi and not the HTTP Server.  Can you post the source code of your `IZ_Example_Util` classes?  I want to see how you are initializing the WiFi environment.

yaoyogi, 2017-11-25:
Thanks for digging.

The wifi management for AP/STA/APSTA + DNS + MDNS + captive-portal (and
more) are in a package of utility classes that are part of a much larger
infrastructure - the tip is IZ_Example_Util classes.  Basically not
easy/obvious to export.

Would be nice if it was just WiFi environment issues.  I'm still wondering
though how the std::string swap outs I did alleviated the problem...

First things first.  I will create a pure IDF standalone that demonstrates
what the same behavior was minus my overall environment so you can verify
against your baseline.

    MrP

nkolban, 2017-11-25:
Okly ... rather than be mysterious on my part ... what I found was that in the code we were examining, since you hadn't specified how you were creating WiFi, I used my own technique and seemed to end up at the same place.  The thinking was then that you did what I did (or I did what you did).  Simply put, in my own test I had the following:

```
void myTask() {
   WiFi myWifi;
   // Code here to initialize WiFi and wait to connect to my access point.
   // Code here to create an HTTP Task and start it.
}
```

The problem ... (for me) ... was that after creating the Http task in the above pseudo code, the WiFi object goes out of scope.  While working with WiFi using this class, we must keep the instance of the class around for as long as we are using WiFi.   So ... are you using the WiFi class supplied with the cpp_utils and, if so, what is its scope?

The good news ... when I did that, my problem went away.  There is of course an opportunity that your puzzle and the one I recreated weren't the same but my symptoms were identical to those that you report.  I also found that I could do thousands of HTTP transports with no issues while before, I could only do about 70 before error.

yaoyogi, 2017-11-25:
Thank you for the mysterious de-ref :)

I am not using the WiFi type from cpp_utils.  I am just using the web
server portion.

Similar to your thinking, I am reusing my general utils (WIFI etc...) for
dozens of other cases/tests successfully so I have just stuck with them so
far.

My network instances are held in global structures that are available/valid
for lifetime of a power-cycle (config etc... goes in/out to NVS as needed).

I agree it's easy to forget that some code ends up in transient stack or
such but is not the case in this scenario.

I will do the decoupling anyways as it is better to isolate all aspects and
just concentrate on what the behavior is, without a lot of potential noise
for other code/tasks.

     MrP

nkolban, 2017-11-26:
The good news is that while studying this issue, I learned a ton about heap debugging within the ESP-IDF environment.  It has recently (within the last month or two) added some wonderful debugging capabilities.  They have to be enabled though and it takes a little work to get the results.  However, when enabled, we can detect writes to storage that were previously freed and detect storage over-runs.  When I had let the WiFi class fall out of scope, some data structures that had been registered with WiFi were released.  Since WiFi was still active, it was updating data at pointers it thought were still good ... and the result was stepping on storage that it no longer owned.  Since that storage was subsequently re-used (malloced) by the HTTP components, all kinds of weird things were happening.  My ideal goal would be that I could recreate your puzzle here.  Which I did ... but that turned out to be my mistake with prematurely releasing WiFi.  When I resolved that, I no longer had an issue and was able to run thousands of requests through the server before *I* stopped testing (no failure).  I think we both agree that is the goal we want to end up with on your machine.  If we can't recreate on my system, we might be able to look at switching on heap tracing on your system to perform an "in-situ" diagnosis ... however if you can recreate a simple version ... that would be ideal.

Here is the program I used which had no issues.

https://pastebin.com/vpsfKfb5

This listens on http://192.168.1.99:9080/helloWorld

I then wrote a Java app that sends a requests

https://pastebin.com/4R6U4Dbk

yaoyogi, 2017-11-26:
I heard about some of the ESP32 related tools but have not tried any yet.
Nice to know they are useful/worthwhile

So far I have roughed it without any type of debugging support, but really
need it now :)

I appreciate all your efforts and agree a small reproducible is best.

Thanks for the code frags, I will use them as well on my end just in case
there is some other environment issues unique to my world...

yaoyogi, 2017-11-26:
I have a decoupled example that represents my main infrastructure code
related to wifi connectivity.

https://pastebin.com/eCjqykBz -- standalone ESP32 IDF example using
cpp_utils webserver
https://pastebin.com/fvzymdEK -- ESP32WebServerTest, my Java test client.
Non-apache, slightly different...
I used your client as well and got same results.

Unfortunately, or at least more precisely, I see the exact same problems
are reported earlier.
Approximately 15-75 requests before failure.

When I replaced FreeRTOS.cpp line 97 which assigns member m_owner, and
change it to be a const char* (vs. std::string), the problem goes away and
I can continuously run requests to the server.  Keep reading after this,
there's better news...

 -- this is the consistent stack trace:
I (407) cpu_start: App cpu up.
I (427) heap_init: Initializing. RAM available for dynamic allocation:
I (434) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (440) heap_init: At 3FFBA698 len 00025968 (150 KiB): DRAM
I (446) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
I (453) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (459) heap_init: At 400905A4 len 0000FA5C (62 KiB): IRAM
I (465) cpu_start: Pro cpu start user code
I (149) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (205) wifi: wifi firmware version: 165c1a5
I (205) wifi: config NVS flash: enabled
I (205) wifi: config nano formating: disabled
I (205) system_api: Base MAC address is not set, read default base MAC
address from BLK0 of EFUSE
I (215) system_api: Base MAC address is not set, read default base MAC
address from BLK0 of EFUSE
I (245) wifi: Init dynamic tx buffer num: 32
I (245) wifi: Init data frame dynamic rx buffer num: 32
I (245) wifi: Init management frame dynamic rx buffer num: 32
I (245) wifi: wifi driver task: 3ffc7124, prio:23, stack:4096
I (255) wifi: Init static rx buffer num: 10
I (255) wifi: Init dynamic rx buffer num: 32
I (265) wifi: Init rx ampdu len mblock:7
I (265) wifi: Init lldesc rx ampdu entry mblock:4
I (265) wifi: wifi power manager task: 0x3ffcc590 prio: 21 stack: 2560
I (275) ex_webserver: ** Connecting to Access Point ssid: MrP, pwd: m0k1ew0m
I (305) phy: phy_version: 366.0, ba9923d, Oct 31 2017, 18:06:17, 0, 0
I (305) wifi: mode : sta (30:ae:a4:32:39:18)
I (305) wifi: sleep disable
I (305) ex_webserver: evthandler event_id: 2
I (315) ex_webserver: SYSTEM_EVENT_STA_START occurred
I (445) wifi: n:1 0, o:1 0, ap:255 255, sta:1 0, prof:1
I (1425) wifi: state: init -> auth (b0)
I (1425) wifi: state: auth -> assoc (0)
I (1435) wifi: state: assoc -> run (10)
I (1445) wifi: connected with MrP, channel 1
I (1445) ex_webserver: evthandler event_id: 4
I (2165) event: sta ip: 192.168.1.72, mask: 255.255.255.0, gw: 192.168.1.254
I (2165) ex_webserver: evthandler event_id: 7
I (2165) ex_webserver: SYSTEM_EVENT_STA_GOT_IP occurred
I (2165) ex_webserver: WiFi STA has IP blocking has completed...
I (2175) ex_webserver: Webserver started, path: /ESP32/hello, port: 9211
I (4435) wifi: pm start, type:0

assertion "next > (intptr_t)block" failed: file
"/Users/marko/esp/esp-idf/components/heap/./multi_heap.c", line 116,
function: get_next_block
abort() was called at PC 0x40129143 on core 0
0x40129143: __assert_func at
/Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdlib/../../../.././newlib/libc/stdlib/assert.c:63
[... 51 lines cut ...]
std::char_traits<char>, std::allocator<char> >) at
/Users/marko/git/iz_firmware/wobblezz/iz_components/kolban/./FreeRTOS.cpp:97

0x400fa331: SockServ::waitForNewClient() at
/Users/marko/git/iz_firmware/wobblezz/iz_components/kolban/./SockServ.cpp:134

0x400f9ce9: HttpServerTask::run(void*) at
/Users/marko/git/iz_firmware/wobblezz/iz_components/kolban/./HttpServer.cpp:293

0x400facf6: Task::runTask(void*) at
/Users/marko/git/iz_firmware/wobblezz/iz_components/kolban/./Task.cpp:58

I saw a number of repo changes in the cpp-utils so I did a fresh pull.  My
tests are now running as-is with no errors!
This is great news and whatever the diff between the two older cpp-utils
pulls I was working with and this fresh pull did the trick!

Thank you.
     MrP

PS.  I would like to debug the ESP32 and be able to do similar ESP32 heap
trace analysis as you are doing.  I am on MAC OS X Sierra 10.12.2, using
Eclipse Neon C++ IDE and the board is
https://www.adafruit.com/product/3405 (silicon
rev 1).  Could you please tell me what your debug environment looks like?

yaoyogi, 2017-11-26:
After much support from Kolban, and a fresh pull of cpp-utils, the issue no longer appears.

Thank you to all, this issue is resolved.

closed 2017-11-26 by yaoyogi

chegewara, 2017-11-26:
Sorry for not answering earlier but im occupied at work. One thing i can say, it is related to short stack. Ive been fighting with this for a while. Its 2 places where you need to set stack size very precisely, because too high stack also can cause the issues.