nkolban/esp32-snippets #219: libcurl failing to connect to server (testCurl.c)

Kniggebrot, 2017-11-23:
Hello everyone,
I recently tried to run the testCurl.c on my ESP32-DevKitC (revision 1) to test the functionality of libcurl on said ESP, because I was looking for a library which is able to do SMTP- and IMAP-TLS-E-Mail-sending and receiving and libcurl seems quite suited for the job whilst also available on the ESP (thanks @nkolban for that!)

And although app and library compile without an error (after removing system_init()), cURL prints the following error:

> * connect to 93.184.216.34 port 80 failed: Connection already in progress
> * Failed to connect to example.com port 80: Connection already in progress
> * Closing connection 0
> I (3342) main: curl_easy_perform failed: Couldn't connect to server 

The full build & monitor log can be found here: [https://pastebin.com/D9Ra0vdt](https://pastebin.com/D9Ra0vdt)

I used the latest cURL and ESP-IDF. I suspect it's an issue with the IDF since I tried various older versions...
Thank you for your time!

danielRM88, 2017-12-09:
Hi @Kniggebrot have you had any luck with this? I am facing the same problem.

Kniggebrot, 2017-12-17:
Hi @danielRM88, sorry for the late answer, had quite a busy week... Thus I didn't have much time to make further tests. I'll try to setup some older IDFs after Christmas, but I can't make any promises... Hopefully I don't need to go too many versions back :D

paulreimer, 2017-12-30:
I have the same issue! I will note that it does work on the current `esp-idf` release (tag `v2.1`), but does not work on the latest `master` at this time.

I ran `git bisect` on `esp-idf` and determined this to be the first breaking commit (curl requests did work just before this):
https://github.com/espressif/esp-idf/commit/3ebf7923d3d4370bc5e8a823fd1ae209bb2b8bc3

It is completely broken after that commit and for the next few commits (boot loop with the following error:)
```
assertion "socket has no netconn" failed: file "/esp-idf/components/lwip/api/sockets.c", line 3143, function: lwip_socket_drop_registered_memberships
abort() was called at PC 0x40108323 on core 0
0x40108323: __assert_func at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdlib/../../../.././newlib/libc/stdlib/assert.c:63 (discriminator 8)
...
```

and then the behaviour changes later to the one reported in this issue (Connection already in progress) at this commit:
https://github.com/espressif/esp-idf/commit/3f83914f7aa2916cbff9b03f5df8533bfed9f9f5

I'm not sure how/where to fix it beyond that, unfortunately.

paulreimer, 2017-12-30:
Alright well at least here is the minimal diff that broke it (I'll assume the middle commit 3ebf792 was WIP and the `assertion "socket has no netconn" failed` is a result of that)

https://github.com/espressif/esp-idf/compare/a320fed...3f83914

Mr-NVR, 2018-01-20:
See https://github.com/espressif/esp-idf/issues/1452

Kniggebrot, 2018-01-21:
Oh nice! Works well, lots of thanks to you guys for helping!

closed 2018-01-21 by Kniggebrot