nkolban/esp32-snippets #220: Compilation error when using cpp_utils

SalvaPedraza, 2017-11-23:
I am using cpp_utils (under <my_project>/components folder) in my ESP-IDF project.
But "make all" produce following error.
I found similar error on https://github.com/espressif/esp32-at/issues/48 and tried different xtensa-esp32-elf verssions, but all fails.
I am developing on Eclipse on Ubuntu 64bits.
What can I try?

make all
CXX GeneralUtils.o
In file included from /home/salva/esp/pool_sensor/components/cpp_utils/./GeneralUtils.cpp:16:0:
/home/salva/esp/pool_sensor/components/cpp_utils/FreeRTOS.h:46:3: error: 'pthread_mutex_t' does not name a type
   pthread_mutex_t   m_pthread_mutex;
   ^
/home/salva/esp/esp-idf/make/component_wrapper.mk:210: recipe for target 'GeneralUtils.o' failed
make[1]: *** [GeneralUtils.o] Error 1
/home/salva/esp/esp-idf/make/project.mk:386: recipe for target 'cpp_utils-build' failed
make: *** [cpp_utils-build] Error 2

chegewara, 2017-11-23:
Try to `make clean` and the `make all`

Also try to make in cli
`xtensa-esp32-elf-gcc -v`

When did you last time update esp-idf and esp32-snippets?

SalvaPedraza, 2017-11-24:
I updated to latest everything: esp-idf, xtensa-esp32-elf and esp32-snippets.
Now everithing worrks fine.
Thanks for your support

nkolban, 2017-12-09:
Closing for just now.  Re-open as needed.

closed 2017-12-09 by nkolban

programmer131, 2018-03-05:
I'm using Arduino IDE and getting exactly same issue, just installed latest ble arduino library and also pulled latest esp32 arduino stack. 

C:\Users\DELL\Documents\Arduino\libraries\ESP32_BLE_Arduino\src/FreeRTOS.h:46:3: error: 'pthread_mutex_t' does not name a type

   pthread_mutex_t   m_pthread_mutex;

gerisse, 2018-03-25:
same issue too for me : with BLE_notify example, in esp 32 BLE Arduino . Thanks

chegewara, 2018-03-25:
Hi,
have you updated arduino-esp32 and toolchain?
From `Arduino\hardware\espressif\esp32\tools\xtensa-esp32-elf\bin` folder use this command:
`xtensa-esp32-elf-gcc.exe --version` 
latest version is: 
`xtensa-esp32-elf-gcc.exe (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a5) 5.2.0`

gerisse, 2018-03-25:
yes, i have the same : 5.2.0 (on Linux)

chegewara, 2018-03-25:
`crosstool-ng-1.22.0-80-g6c4433a5`?

gerisse, 2018-03-25:
not exactly : xtensa-esp32-elf-gcc (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a) 5.2.0

chegewara, 2018-03-25:
@gerisse your toolchain is fresh enough to work with pthreads, please try @programmer131 solution

gerisse, 2018-03-25:
ok, i re-installed it, and all is ok, thanks :+1:

vjkanth, 2018-04-03:
In file included from C:\...\libraries\ESP32_BLE_Arduino-master\src/BLEDescriptor.h:16:0,

                 from C:\...\libraries\ESP32_BLE_Arduino-master\src/BLECharacteristic.h:17,

                 from C:\...\libraries\ESP32_BLE_Arduino-master\src/BLEServer.h:19,

                 from C:\...\libraries\ESP32_BLE_Arduino-master\src/BLEDevice.h:18,

                 from C:\arduino-1.8.5\sketches\BLE_server\BLE_server.ino:6:

C:\...\libraries\ESP32_BLE_Arduino-master\src/FreeRTOS.h:46:3: error: 'pthread_mutex_t' does not name a type

   pthread_mutex_t   m_pthread_mutex;

   ^

In file included from C:\...\libraries\ESP32_BLE_Arduino-master\src/BLECharacteristic.h:17:0,

                 from C:\...\libraries\ESP32_BLE_Arduino-master\src/BLEServer.h:19,

                 from C:\...\libraries\ESP32_BLE_Arduino-master\src/BLEDevice.h:18,

                 from C:\arduino-1.8.5\sketches\BLE_server\BLE_server.ino:6:

C:\...\libraries\ESP32_BLE_Arduino-master\src/BLEDescriptor.h:54:60: error: invalid conversion from 'int' to 'esp_gatt_perm_t' [-fpermissive]

  esp_gatt_perm_t      m_permissions = ESP_GATT_PERM_READ | ESP_GATT_PERM_WRITE;

                                                            ^

In file included from C:\...\libraries\ESP32_BLE_Arduino-master\src/BLEServer.h:19:0,

                 from C:\...\libraries\ESP32_BLE_Arduino-master\src/BLEDevice.h:18,

                 from C:\arduino-1.8.5\sketches\BLE_server\BLE_server.ino:6:

C:\...\libraries\ESP32_BLE_Arduino-master\src/BLECharacteristic.h:105:67: error: invalid conversion from 'int' to 'esp_gatt_perm_t' [-fpermissive]

  esp_gatt_perm_t             m_permissions = ESP_GATT_PERM_READ | ESP_GATT_PERM_WRITE;

                                                                   ^

In file included from C:\...\libraries\ESP32_BLE_Arduino-master\src/BLEDevice.h:20:0,

                 from C:\arduino-1.8.5\sketches\BLE_server\BLE_server.ino:6:

C:\...\libraries\ESP32_BLE_Arduino-master\src/BLEUtils.h:50:49: error: 'esp_gattc_service_elem_t' has not been declared

  static std::string gattcServiceElementToString(esp_gattc_service_elem_t *pGATTCServiceElement);

                                                 ^
Getting this when attempting to compile the BLE_Server sketch. Had to copy over files it couldn't find. Added a line "#define CONFIG_CXX_EXCEPTIONS 1" in teh sdkconfig.h file. Have Windows 10, ESP32 over USB, ensured the toolchain version is at "xtensa-esp32-elf-gcc.exe (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a5) 5.2.0", nevertheless did another get to reconfirm. Restarted IDE etc... Still no joy. Tried a simpler sketch at "https://github.com/espressif/arduino-esp32/blob/master/libraries/SimpleBLE/examples/SimpleBleDevice/SimpleBleDevice.ino" and works perfectly, can see the chip on my Android phone. This one even compile. What else can I try please?

chegewara, 2018-04-04:
https://github.com/nkolban/esp32-snippets/issues/220#issuecomment-375988331
Im guessing he meant to reinstall arduino