nkolban/esp32-snippets #222: SampleNotify.cpp issue

dorthonir, 2017-11-25:
Hi, I have an issue with this specific example. It compiles just fine, but seem to have problem with creating characteristic:

`D (1370) BLEService: << start()
D (1370) BLEAdvertising: >> start
D (1370) BLEAdvertising: - no services advertised
D (1380) BLEUtils: Received a GAP event: ESP_GAP_BLE_ADV_DATA_SET_COMPLETE_EVT
D (1380) BLEUtils: [status: 0]
D (1390) BLEServer: BLEServer ... handling GAP event!
D (1390) BLEUtils: Received a GAP event: ESP_GAP_BLE_SCAN_RSP_DATA_SET_COMPLETE_EVT
D (1400) BLEUtils: [status: 0]
D (1400) BLEServer: BLEServer ... handling GAP event!
D (1410) BLEUtils: Received a GAP event: ESP_GAP_BLE_ADV_START_COMPLETE_EVT
D (1420) BLEUtils: [status: 0]
D (1420) BLEServer: BLEServer ... handling GAP event!
D (1420) BLEAdvertising: << start
`
I have the latest esp-idf env, and esp32-snippets.

nkolban, 2017-11-26:
Howdy, That appears to be a fragment of trace from the overall trace.  When I start my copy of SampleNotify, that is also what I see at the end.   At this point, the ESP32 is being a BLE server.  The expectation now is that a BLE client will connect to it at which point the BLE server (the ESP32) will start to transmit notifications to the BLE client.  In your story, did you connect a BLE client?  I test with the nRFConnect tool on an Arduino.

dorthonir, 2017-11-26:
Thank you for the response. I tested it with the same app. After doing some changes, that are hard to trace back now, it finally started working. Maybe it had something to do with stack size. 
Yesterday , on the same app, the service and characteristic were just unavailable, app didn't seem to see it. It works just fine now, but the another problem emerged.
 I wanted to connect 2 esp32's one based on sampleNotify.cpp and second on sampleClient_Notify.cpp. I matched the UUID's in the examples, client is detecting the server, but it doesn't connect it. The sampleServer example, on the other hand, connects just fine. Is there something I missed?

nkolban, 2017-11-26:
Without seeing your code or a design there's not much that I can immediately offer.  What I'd suggest is get both parts working independently of each other.  I use an Android phone using nRFConnect to act as a BLE client (to an ESP32 being a BLE server).  The same nRFConnect can also be a BLE Server (to an ESP32 being a BLE client).  Once you have confidence/faith that the two ends are working correctly then you can point them to each other and see what happens.

The BLE C++ classes have trace built in so if we switch on the ESP-IDF tracing to verbose, we can see the log messages issued.  From there, we can draw a map from one device to the other and see the communications from one received from the other (hopefully).

dorthonir, 2017-12-04:
Hi, I tested both parts as you suggested, and they work. Sample_notify connects to my phone and sends notifications. Sample_client connects to the sample_server example. Sample_notify, on the other hand, doesn't cooperate with any of sample_client apps. This issue seem very familiar with issue no. #198 Client finds the other device, but then just ignores it:

> D (876) BLEUtils: Received a GAP event: ESP_GAP_BLE_SCAN_PARAM_SET_COMPLETE_EVT
D (886) BLEUtils: [status: 0]
D (886) BLEUtils: Received a GAP event: ESP_GAP_BLE_SCAN_START_COMPLETE_EVT
D (896) BLEUtils: [status: 0]
D (896) BLEUtils: Received a GAP event: ESP_GAP_BLE_SCAN_RESULT_EVT
D (906) BLEUtils: search_evt: ESP_GAP_SEARCH_INQ_RES_EVT, bda: 30:ae:a4:07:d5:ba, dev_type: ESP_BT_DEVICE_TYPE_
BLE, ble_addr_type: BLE_ADDR_TYPE_PUBLIC, ble_evt_type: ESP_BLE_EVT_CONN_ADV, rssi: -43, ble_adv: ??, flag: 6 (
[LE General Discoverable Mode] [BR/EDR Not Supported] ), num_resps: 1, adv_data_len: 22, scan_rsp_len: 22
D (936) BLEAdvertisedDevice: - setRSSI(): rssi: -43
D (936) BLEAdvertisedDevice: Type: 0x01 (ESP_BLE_AD_TYPE_FLAG), length: 1, data: 06
D (946) BLEAdvertisedDevice: Type: 0x09 (ESP_BLE_AD_TYPE_NAME_CMPL), length: 8, data: 4d59444556494345
D (956) BLEAdvertisedDevice: - setName(): name: MYDEVICE
D (966) BLEAdvertisedDevice: Type: 0x0a (ESP_BLE_AD_TYPE_TX_PWR), length: 1, data: eb
D (966) BLEAdvertisedDevice: - txPower: -21
D (976) BLEAdvertisedDevice: Type: 0x12 (ESP_BLE_AD_TYPE_INT_RANGE), length: 4, data: 20004000
D (986) BLEAdvertisedDevice: Type: adType: 18 - 0x12
D (986) BLEAdvertisedDevice: Type: 0x01 (ESP_BLE_AD_TYPE_FLAG), length: 1, data: 06
D (996) BLEAdvertisedDevice: Type: 0x09 (ESP_BLE_AD_TYPE_NAME_CMPL), length: 8, data: 4d59444556494345
D (1006) BLEAdvertisedDevice: - setName(): name: MYDEVICE
D (1016) SampleClientDisconnect: Advertised Device: Name: MYDEVICE, Address: 30:ae:a4:07:d5:ba, txPower: -21
D (1026) BLEUtils: Received a GAP event: ESP_GAP_BLE_SCAN_RESULT_EVT
D (1026) BLEUtils: search_evt: ESP_GAP_SEARCH_INQ_RES_EVT, bda: 30:ae:a4:07:d5:ba, dev_type: ESP_BT_DEVICE_TYPE
_BLE, ble_addr_type: BLE_ADDR_TYPE_PUBLIC, ble_evt_type: ESP_BLE_EVT_CONN_ADV, rssi: -43, ble_adv: ??, flag: 6
([LE General Discoverable Mode] [BR/EDR Not Supported] ), num_resps: 1, adv_data_len: 22, scan_rsp_len: 22
D (1056) BLEScan: Ignoring 30:ae:a4:07:d5:ba, already seen it.
D (1066) BLEUtils: Received a GAP event: ESP_GAP_BLE_SCAN_RESULT_EVT
D (1066) BLEUtils: search_evt: ESP_GAP_SEARCH_INQ_RES_EVT, bda: 30:ae:a4:07:d5:ba, dev_type: ESP_BT_DEVICE_TYPE
_BLE, ble_addr_type: BLE_ADDR_TYPE_PUBLIC, ble_evt_type: ESP_BLE_EVT_CONN_ADV, rssi: -44, ble_adv: ??, flag: 6
([LE General Discoverable Mode] [BR/EDR Not Supported] ), num_resps: 1, adv_data_len: 22, scan_rsp_len: 22
D (1096) BLEScan: Ignoring 30:ae:a4:07:d5:ba, already seen it.
D (1106) BLEUtils: Received a GAP event: ESP_GAP_BLE_SCAN_RESULT_EVT
D (1106) BLEUtils: search_evt: ESP_GAP_SEARCH_INQ_RES_EVT, bda: 30:ae:a4:07:d5:ba, dev_type: ESP_BT_DEVICE_