nkolban/esp32-snippets #223: WiFi.h/.c files are clashing with files from arduino-esp32

roamingthings, 2017-11-25:
The files named `cpp_utils/WiFi.c` and `cpp_utils/WiFi.h` clash with files from the esp32-arduinio component.
It's not possible to use cpp_utils in conjunction with esp32-arduino.

I suggest renaming theses files for example to `WiFiUtil.h` and `WiFiUtil.c`

nkolban, 2017-11-25:
Hmmm ... this is a puzzle.  My thinking was always that Arduino already has a first class eco system for C++ programming that builds on the Arduino APIs and existant Arduino libraries.  The existence of cpp_utils was primarily for working in ESP-IDF land.  The one place where they overlap is in BLE support where the BLE C++ classes found in cpp_utils have been packaged and distributed with the Arduino-ESP32 libraries.

My thinking is ... if one is programming in Arduino APIs then one would be using the WiFi headers provided by the Arduino WiFi support.   What I'm not seeing is why one would want to use cpp_utils WiFi classes in an Arduino environment which already has WiFi support?

roamingthings, 2017-11-27:
I‘m currently working on a small sideproject where I want to mainly use ESP-IDF but also want to use a driver (GxEPD) which relies on the Arduino framework and other similar dependencies. My plan is to get rid of the Arduino framework as soon as I understand the components I’m using better or have ported the driver to be independent of Arduino libraries.

I agree that if you are using the Arduino ecosystem you may not need the cpp_utils

nkolban, 2017-11-27:
I think the WiFi.cpp and WiFi.h source files supplied with cpp_utils are at the top level of any builds which means that if you bring in cpp_utils and use with Arduino ... then of you simply remove the WiFi.cpp and WiFi.h files from the cpp_utils folders they will all be "gone" and you can use the remainder of cpp_utils with Arduino.  Note that we haven't tested these with the exception of the BLE code.

roamingthings, 2017-11-28:
I have been thinking about that issue and can see your argument. Since I want to migrate away von the Arduino library it should be totally fine for me to remove (or in my case rename) the files as long as I have the Arduino dependency on my project.

So I will close this issue.
Thank you for the discussion.

closed 2017-11-28 by roamingthings