nkolban/esp32-snippets #224: Ringbuffer object not usable from C++ due to missing extern "C" declaration (linker error)

Sepp62, 2017-11-25:
This may be a duplicate from 
https://esp32.com/viewtopic.php?f=13&t=1301
but it seems not to be solved yet.

In 
...Arduino\hardware\espressif\esp32\tools\sdk\include\freertos\freertos\ringbuf.h
extern "C"-Declaration is missing:

```
#ifdef __cplusplus
extern "C" {
#endif
```

nkolban, 2017-11-25:
It seems that it was fixed 2 weeks ago in the latest ESP-IDF.  See:

https://github.com/espressif/esp-idf/commit/0b48d5978e4d7173eb7e9932b93234aeb3f2283a

Try and refresh your ESP-IDF to the latest version available and see if that resolves the puzzle.

Sepp62, 2017-11-25:
For any reason get.exe did not refresh my SDK-files.

After deleting the whole repository and cloning it anew, I have the latest files and the declaration is there.

Issue resolved.

Sorry for wasting your time. I should  have had a look into the files via Gitgub webgui.

closed 2017-11-25 by Sepp62