nkolban/esp32-snippets #225: BLE_Client crash on server disconnect

skyng22003, 2017-11-25:
First of all, a massive thank you for writing this library.

I have been using the ESP32 on the Arduino IDE with the latest ESP32_BLE library.

I made some modifications to your BLE_Client here's the code is below.

Basically what it is doing it there are 2 services after it connects to the server.
The server sends out notifications on the characteristic
rxUUID("0783b03e-8535-b5a0-7140-a304d2495cb8");
and the other is charUUID("0783b03e-8535-b5a0-7140-a304d2495cba"); which the client can write to.

There is a timeout timer "BLEtimeout" where if the client does not receive any notifications the client initiates a pClient->disconnect and re-scans to connect, this part works fine.

However, if the server initiates a disconnect ie. device is powered off etc the program grinds to a halt.

Here's the printout:
 Notify callback for characteristic 0783b03e-8535-b5a0-7140-a304d2495cb8 of data length 8
Data: 614000040020 //Last notifcation received, server disconnect here
Setting new characteristic value to "Time since boot: 230"
E (252590) BT: bta_gattc_conn_cback() - cif=3 connected=0 conn_id=3 reason=0x0008
connected: 1
doConnect: 0
Scanning: 0
BLETO: 1
Setting new characteristic value to "Time since boot: 240"
//After this there were no more printouts

I don't know if this is related to the other issues raised but I would appreciate your input.

Many Thanks, Sky

**************************************************************************
/**
* A BLE client example that is rich in capabilities.
*/

#include "BLEDevice.h"
//#include "BLEScan.h"
uint8_t BLEtimeout =0;
uint8_t txData[20];
// The remote service we wish to connect to.
static BLEUUID serviceUUID("0783b03e-8535-b5a0-7140-a304d2495cb7");
// The characteristic of the remote service we are interested in.
static BLEUUID    charUUID("0783b03e-8535-b5a0-7140-a304d2495cba");
static BLEUUID    rxUUID("0783b03e-8535-b5a0-7140-a304d2495cb8");

static BLEAddress *pServerAddress;
BLERemoteService* pRemoteService;
BLEClient*  pClient = BLEDevice::createClient();
static boolean doConnect = false;
static boolean connected = false;
bool Scanning = false;
static BLERemoteCharacteristic* pRemoteCharacteristic;
static void notifyCallback(
  BLERemoteCharacteristic* pBLERemoteCharacteristic,
  uint8_t* pData,
  size_t length,
  bool isNotify) {
  Serial.print("Notify callback for characteristic ");
  Serial.print(pBLERemoteCharacteristic->getUUID().toString().c_str());
  Serial.print(" of data length ");
  Serial.println(length);
  Serial.print("Data: ");
  Serial.print(pData[0]);
  Serial.print(pData[1]);
  Serial.print(pData[2]);
  Serial.print(pData[3]);
  Serial.print(pData[4]);
  Serial.print(pData[5]);
  Serial.print(pData[6]);
  Serial.print(pData[7]);
  Serial.print(pData[8]);
  Serial.print(pData[9]);
  Serial.println(pData[10]);
  BLEtimeout = 0;
}

bool connectToServer(BLEAddress pAddress) {
  Serial.print("Forming a connection to ");
  Serial.println(pAddress.toString().c_str());

  Serial.println(" - Created client");

  // Connect to the remove BLE Server.
  pClient->connect(pAddress);
  Serial.println(" - Connected to server");

  // Obtain a reference to the service we are after in the remote BLE server.
  pRemoteService = pClient->getService(serviceUUID);
  if (pRemoteService == nullptr) {
    Serial.print("Failed to find our service UUID: ");
    Serial.println(serviceUUID.toString().c_str());
    return false;
  }
  Serial.println(" - Found our service");

  pRemoteCharacteristic = pRemoteService->getCharacteristic(rxUUID);
  if (pRemoteCharacteristic == nullptr) {
    Serial.print("Failed to find RX UUID: ");
    Serial.println(rxUUID.toString().c_str());
    return false;
  }

  //   Serial.println(" - Found RX characteristic");
  // Read the value of the characteristic.
  //std::string value1 = pRemoteCharacteristic->readValue();
  // Serial.print("The characteristic value was: ");
 // Serial.println(value1.c_str());

  pRemoteCharacteristic->registerForNotify(notifyCallback);
  //pRemoteCharacteristic = pRemoteService->getCharacteristic(charUUID);

  // Obtain a reference to the characteristic in the service of the remote BLE server.
  pRemoteCharacteristic = pRemoteService->getCharacteristic(charUUID);
  if (pRemoteCharacteristic == nullptr) {
    Serial.print("Failed to find our characteristic UUID: ");
    Serial.println(charUUID.toString().c_str());
    return false;
  }
  Serial.println(" - Found our characteristic");

  //std::string value = pRemoteCharacteristic->readValue();
  //Serial.print("The characteristic value was: ");
  //Serial.println(value.c_str());

}
/**
* Scan for BLE servers and find the first one that advertises the service we are looking for.
*/
class MyAdvertisedDeviceCallbacks : public BLEAdvertisedDeviceCallbacks {
  /**
  * Called for each advertising BLE server.
  */
  void onResult(BLEAdvertisedDevice advertisedDevice) {
    Serial.print("BLE Advertised Device found: ");
    Serial.println(advertisedDevice.toString().c_str());

    // We have found a device, let us now see if it contains the service we are looking for.
    if (advertisedDevice.haveServiceUUID() && advertisedDevice.getServiceUUID().equals(serviceUUID)) {

      // 
      Serial.print("Found our device!  address: ");
      advertisedDevice.getScan()->stop();

      pServerAddress = new BLEAddress(advertisedDevice.getAddress());
      doConnect = true;

    } // Found our server
  } // onResult
}; // MyAdvertisedDeviceCallbacks

void setup() {
  Serial.begin(115200);
  Serial.println("Starting Arduino BLE Client application...");
  BLEDevice::init("");

  // Retrieve a Scanner and set the callback we want to use to be informed when we
  // have detected a new device.  Specify that we want active scanning and start the
  // scan to run for 30 seconds.
  Scanning = true;
  BLEScan* pBLEScan = BLEDevice::getScan();
  pBLEScan->setAdvertisedDeviceCallbacks(new MyAdvertisedDeviceCallbacks());
  pBLEScan->setActiveScan(true);
  pBLEScan->start(30);
} // End of setup.

  // This is the Arduino main loop function.
void loop() {

  // If the flag "doConnect" is true then we have scanned for and found the desired
  // BLE Server with which we wish to connect.  Now we connect to it.  Once we are 
  // connected we set the connected flag to be true.
  if (doConnect == true) {
    if (connectToServer(*pServerAddress)) {
      Serial.println("We are now connected to the BLE Server.");
      connected = true;
     // Scanning = false;
    }
    else {
      Serial.println("We have failed to connect to the server; there is nothin more we will do.");
    }
    doConnect = false;
    //Scanning = true;
  }

  // If we are connected to a peer BLE Server, update the characteristic each time we are reached
  // with the current time since boot.
  if (connected) {
    String newValue = "Time since boot: " + String(millis() / 1000);
    Serial.println("Setting new characteristic value to \"" + newValue + "\"");
    
    Scanning = false;
    // Set the characteristic's value to be the array of bytes that is actually a string.

    //pRemoteCharacteristic->writeValue(txData,20);
    //pRemoteCharacteristic = pRemoteService->getCharacteristic(charUUID);
    pRemoteCharacteristic->writeValue(newValue.c_str(), newValue.length());
    //std::string value3 = pRemoteCharacteristic->readValue();
    // Serial.print("Data:");
    //Serial.println(value3.c_str());
  }

  

  if (BLEtimeout >= 5) {
    
    connected = false;
    doConnect = false;
     pClient->disconnect();
    //Scanning = true;
    BLEtimeout = 6;
  }
  if(connected == false && doConnect == false){
    Scanning = true;
    BLEScan* pBLEScan = BLEDevice::getScan();
    pBLEScan->setAdvertisedDeviceCallbacks(new MyAdvertisedDeviceCallbacks());
    pBLEScan->setActiveScan(true);
    pBLEScan->start(30);
    BLEtimeout = 0;
  }
if (Scanning == false){
BLEtimeout++;
}

Serial.print("connected: ");
Serial.println(connected);

Serial.print("doConnect: ");
Serial.println(doConnect);

Serial.print("Scanning: ");
Serial.println(Scanning);

Serial.print("BLETO: ");
Serial.println(BLEtimeout);

  delay(1000); // Delay a second between loops.
} // End of loop
**********************************************************************************************

nkolban, 2017-11-25:
Many thanks for the detailed report.  Lets see if we can't capture a trace from your environment.  There are some instructions found here:

https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/ArduinoBLE.md#switching-on-debugging

Switching this on, rebuilding the code and re-running should produce trace of what is going inside the BLE libraries.  From there, we can take a look see and see if we get an indication of what might be going on.

skyng22003, 2017-11-26:
Thank you for your prompt reply, below is what I got from turning verbose debug on:
[D][BLEDevice.cpp:105] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 3] ... ESP_GATTC_NOTIFY_EVT
[D][BLEUtils.cpp:1344] dumpGattClientEvent(): GATT Event: ESP_GATTC_NOTIFY_EVT
[D][BLEUtils.cpp:1451] dumpGattClientEvent(): [conn_id: 0, remote_bda: f5:e1:02:80:05:f4, handle: 18 0x12, value_len: 8, is_notify: 1]
[D][BLERemoteCharacteristic.cpp:173] gattClientEventHandler(): Invoking callback for notification on characteristic Characteristic: uuid: 0783b03e-8535-b5a0-7140-a304d2495cb8, handle: 18 0x12, props: broadcast: 0, read: 0, write_nr: 0, write: 0, notify: 1, indicate: 0, auth: 0
Notify callback for characteristic 0783b03e-8535-b5a0-7140-a304d2495cb8 of data length 8
Data: 614000040019
E (78442) BT: bta_gattc_conn_cback() - cif=3 connected=0 conn_id=3 reason=0x0008
[D][BLEDevice.cpp:105] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 3] ... ESP_GATTC_WRITE_CHAR_EVT
[D][BLEUtils.cpp:1344] dumpGattClientEvent(): GATT Event: ESP_GATTC_WRITE_CHAR_EVT
[D][BLEUtils.cpp:1587] dumpGattClientEvent(): [status: ESP_GATT_ERROR, conn_id: 0, handle: 22 0x16]
[D][BLERemoteCharacteristic.cpp:566] writeValue(): << writeValue
connected: 1
doConnect: 0
Scanning: 0
BLETO: 1
[D][BLEDevice.cpp:105] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 3] ... ESP_GATTC_DISCONNECT_EVT
[D][BLEUtils.cpp:1344] dumpGattClientEvent(): GATT Event: ESP_GATTC_DISCONNECT_EVT
[D][BLEUtils.cpp:1389] dumpGattClientEvent(): [conn_id: 0, remote_bda: f5:e1:02:80:05:f4]
[D][BLEDevice.cpp:105] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 3] ... ESP_GATTC_CLOSE_EVT
[D][BLEUtils.cpp:1344] dumpGattClientEvent(): GATT Event: ESP_GATTC_CLOSE_EVT
[D][BLEUtils.cpp:1359] dumpGattClientEvent(): [status: ESP_GATT_INSUF_AUTHORIZATION, reason:ESP_GATT_CONN_TIMEOUT, conn_id: 0]
Setting new characteristic value to "Time since boot: 66"
[D][BLERemoteCharacteristic.cpp:544] writeValue(): >> writeValue(), length: 19

skyng22003, 2017-11-26:
That was the end of the print, there were no exception codes though....

nkolban, 2017-11-26:
That doesn't seem like enough trace ... was there more at the start?  Can you paste the complete set of trace to a pastebin?  We are seeing some errors in the trace ....  these are the errors pushed by the lower level ... which just now simply state "ESP_GATT_ERROR" ... I'm hoping that previous trace will give us more information.

skyng22003, 2017-11-26:
Sure, let me know if there's anything else I can do
[https://pastebin.com/itsE2J6f](url)

nkolban, 2017-11-26:
I have an initial guess at what is going on and have created an issue to guard ... see:

https://github.com/nkolban/esp32-snippets/issues/228

At a high level, I think what is happening is that the logic in the app is calling 

```
pRemoteCharacteristic->writeValue(newValue.c_str(), newValue.length());
```

Even though the connection to the BLE Server has now been lost.  It appears that the BLE logic blocks that API call waiting for a response that will never come.  The onus appears to be on the application logic to not attempt to read or write from remote characteristics after an indication that the connection has been dropped.

The BLEClient class has a method on it called "isConnected" which returns true if we are connected and false otherwise.  What I'd like you to do is test that flag is correct and only attempt to write to the remote characteristic if we are indeed currently connected.

skyng22003, 2017-11-26:
The code below fixed the issue, many thanks!

Is there an option to configure the "isConnected" timeout? 

 if(pClient->isConnected()==true){
    pRemoteCharacteristic->writeValue(newValue.c_str(), newValue.length());
    Serial.println("Setting new characteristic value to \"" + newValue + "\"");
    //std::string value3 = pRemoteCharacteristic->readValue();
    // Serial.print("Data:");
    //Serial.println(value3.c_str());
    }

nkolban, 2017-11-26:
This is only a circumvention issue #228 will fix it properly.  Currently there is no "timeout" ... we are either connected or not connected.  No timeout involved.  What I need to do is provide some improvements such as:

1. Throwing an exception when attempting to read or write from a disconnected characteristic.
2. Providing an "onDisconnected" callback

skyng22003, 2017-11-26:
I see, thank you very much for your help!

nkolban, 2017-12-14:
Closing for just now

closed 2017-12-14 by nkolban