nkolban/esp32-snippets #227: BLE exception

ifrew, 2017-11-25:
IDE: Visual Micro (Visual Studio) built on top of Arduino 1.8.5
ESPCore32  Build: 11/24/17
BLE library: BLE_Arduino shipped in ESP32 Core

I had  been working to overcome an exception in the ESP32 IDF Bluetooth stack this morning after calling this BLEDevice::Init See https://github.com/espressif/esp-idf/issues/1315 

I figured out that a malloc was failing  due to a lack of RAM. I have seen posts on that topic here along with boot loop issues which I also originally was getting. I was going to add a call in BLEDevice.cpp to release the classic Bluetooth stack after the controller was initialized as recommend by others. There I found that @nkolban had already put that in there but had commented it out

// esp_bt_controller_mem_release(ESP_BT_MODE_CLASSIC_BT);  //FIXME waiting for response from esp-idf issue

@nkolban Since I uncommented your code to get this far what IDF issue are you referring to in the line above? That may help me move forwards on this.

I uncommented this and sure enough I got past the initial exception and the ble server came up and my sketch started running. Awesome! However, it then died again with another exception shortly later. Decoding the stack gives this traceback. This  seems like another memory issue. Not sure how to release any more RAM. There was mention made of selecting wifi config options to be a low level but since I don't use arudnio ide directly I'm not sure what #define that would represent so I can change the sdkconfig.h file and recompile.

Thanks folks for any insight!

Decoding 21 results
0x4008b321: uxListRemove at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./list.c line 214
0x4008a6a2: xTaskRemoveFromEventList at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./tasks.c line 3485
0x40088fed: xQueueGenericSend at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./queue.c line 1995
0x4013a745: hci_hal_h4_task_post at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/bt/bluedroid/hci/hci_hal_h4.c line 180
0x4013a7c6: host_recv_pkt_cb at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/bt/bluedroid/hci/hci_hal_h4.c line 274
0x401a0469: vhci_send at ?? line ?
0x4019a87a: r_eif_send at ?? line ?
0x4019ae19: hci_tx_start at hci_tl.c line ?
0x4019b002: r_hci_tl_send at ?? line ?
0x4019eea2: r_llm_con_req_ind at ?? line ?
0x4019f725: r_llm_pdu_defer at ?? line ?
0x4019ded7: r_lld_pdu_check at ?? line ?
0x4019ca26: r_lld_evt_deffered_elt_handler at ?? line ?
0x4019a383: r_rw_schedule at ?? line ?
0x4019a3d3: btdm_controller_task at ?? line ?

Guru Meditation Error: Core  0 panic'ed (StoreProhibited)
. Exception was unhandled.
Register dump:
PC      : 0x4008b321  PS      : 0x00060233  A0      : 0x8008a6a5  A1      : 0x3fff88a0  
A2      : 0x3ffbd974  A3      : 0x00000000  A4      : 0x00000010  A5      : 0x3ffbd834  
A6      : 0x3ffc7fc0  A7      : 0x00000022  A8      : 0x000f0000  A9      : 0x00001f32  
A10     : 0x00000000  A11     : 0x00000000  A12     : 0x00060420  A13     : 0x00000001  
A14     : 0x00060223  A15     : 0x00000000  SAR     : 0x00000008  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x00001f3a  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  

Backtrace: 0x4008b321:0x3fff88a0 0x4008a6a2:0x3fff88c0 0x40088fed:0x3fff88e0 0x4013a745:0x3fff8920 0x4013a7c6:0x3fff8950 0x401a0469:0x3fff8970 0x4019a87a:0x3fff8990 0x4001791d:0x3fff89b0 0x4019ae19:0x3fff89d0 0x4019b002:0x3fff8a10 0x400188f5:0x3fff8a40 0x40044d49:0x3fff8a80 0x400448f1:0x3fff8ab0 0x4019eea2:0x3fff8ae0 0x4019f725:0x3fff8b40 0x4019ded7:0x3fff8b60 0x4019ca26:0x3fff8bb0 0x40019d11:0x3fff8bf0 0x40055b4d:0x3fff8c10 0x4019a383:0x3fff8c30 0x4019a3d3:0x3fff8c50 Stack decoding gives this error

nkolban, 2017-11-25:
Ideally what I'd like to see is a minimal Arduino sketch that is failing.  If possible, please post that as a pastebin and we can examine the source and try and recreate.

Do the example BLE sketches supplied with the distribution work?

We have the ability to switch on diagnostics ... see:

https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/ArduinoBLE.md#switching-on-debugging

chegewara, 2017-11-26:
It is commented in BLEDevice because its causing multi_heap crashes. I dont know how it will behave in arduino but you can try to comment it back and use this in 1st line in setup:
`esp_bt_controller_mem_release(ESP_BT_MODE_CLASSIC_BT);`

ifrew, 2017-11-27:
Ok. I dig more digging into this today..a lot of time. I have close the issue in esp core 32 and will use this moving forwards..

I do not believe the issue is RAM now. I had to release memory in your BLEDevice code in order to get the init to work. That gave the issue of the guru meditation error we have now. Enabling lower level debug statements my app started to run but as soon as I connected to it using the BLE Scanner android app it crashed with the same error. Progress! So the issue is in handling the connection.  I did however manage to get the simple ble_example modified to provide a very similar error. Using the android phone app BLE Scanner, I was able to get it to crash too.

Note that in both cases the BLEDevice code now issues esp_bt_controller_mem_release(ESP_BT_MODE_CLASSIC_BT) after the init.. I uncommented the line.

The trace for both apps is here
https://pastebin.com/KVXmQAis

[ble example.ino.txt](https://github.com/nkolban/esp32-snippets/files/1504623/ble.example.ino.txt)

 and I have attached the ble_example sketch so you can try and replicate.
Note also that I saw a similar issue reported here https://esp32.com/viewtopic.php?f=13&t=3513

chegewara, 2017-11-27:
Im just saying what i think. I spent 2-4 weeks fighting with memory shortage, heap issues etc. Few days ago i found last pieces of this puzzle, just im very busy at work and cant make PR. You can look at ESP32_Explorer to see. In this time with mr. @nkolban we were able to recover 80-100k of heap, but having about 100k free heap, and lowest available amount of heap about 60kB i had heap corruption issues. This commented line in BLEDevice is my piece of code and it is commented because ive got heap corruption in heap released by this command.

Now back to your code. You may be right and it is not related to ram shortage, but you are running ble at the same time with wifi. This is deadly combination, because you are working under arduino and you cant change options in menuconfig. Im working with esp-idf and im able to free about 40k from wifi library, about 12k from mbedtls, and some more in few other parts of code. Im here to help you and other users and this is only your choice to use or not my help.

ifrew, 2017-11-27:
Thanks for the feedback! I see there are a lot of issues using ble and WiFi and heap. Been seeing all your posts about it. For my project, what we do now is use a hm10 or hc05 board as an addition running on esp8266.  Simple and inexpensive solution. Was hoping to capitalize on using the esp32 Bluetooth. Looks like until this all settles that approach is the best at this time for esp32. Appreciate how easy the ble library is to use though. You and Neil have done a great job there. It was way to complex to use all the GATT stuff directly.

I am also using visual micro on top of Arduino ide so I have less control to easily changing configurations. Moving to idf may be better approach. Btw, what config did you change in WiFi to release 40k? I saw your post about choosing minimum WiFi settings in menuconfig but since I don't have that I wasn't sure what they were. I'm off on vacation on Wednesday so, won't have time to so much more digging. However, it would be great to test if the example sketch that fails I sent actually ran in idf with the ram you freed up and the comment put back to not release memory. If it fails then that would be good for the idf folks to look at fixing.

chegewara, 2017-11-27:
Its my first quick excercise. Try to run this code with commented and uncommented line which release memory from BT:

```
#include <esp_log.h>
#include <BLEDevice.h>
void setup() {
  Serial.begin(115200);
  // put your setup code here, to run once:
  esp_bt_controller_mem_release(ESP_BT_MODE_CLASSIC_BT);
  ESP_LOGE(LOG_TAG, "current free heap: %d, minimum ever free heap: %d", xPortGetFreeHeapSize(), xPortGetMinimumEverFreeHeapSize());
}

void loop() {
  // put your main code here, to run repeatedly:

}
```

Before:
`setup(): current free heap: 223296, minimum ever free heap: 218188
`
after:
`setup(): current free heap: 248056, minimum ever free heap: 243096
`

chegewara, 2017-11-27:
You can use this line to make some debugging in your code and see if its lack of heap issue or not:
`  ESP_LOGE(LOG_TAG, "current free heap: %d, minimum ever free heap: %d", xPortGetFreeHeapSize(), xPortGetMinimumEverFreeHeapSize());
`
It is just a begining.

chegewara, 2017-11-27:
All changes i made are in sdkconfig.h file and tested with simple wifi example file in arduino ide. All changes you made are on your own responsibility. You have to understand what you are changing, but we are always here to help you with issues.
To release some heap from WiFi stack you need to change this value:
`CONFIG_ESP32_WIFI_STATIC_RX_BUFFER_NUM` minimum value is 2
From experience i found that this is also too high stack in normal usage cases:
`CONFIG_BTC_TASK_STACK_SIZE`
Im usually setting it to 3000, but ive had to increase it to 4000 in case i setup about 6 or 10 characteristics (dont remember now). I have had no crash in app, just not all characteristics were visible by central. 
Another value im changing in my settings is:
`MBEDTLS_SSL_MAX_CONTENT_LEN` - minimal value is 512 but ive had to set it up a little bit higher in case of OTA image, with normal factory case 512 should have works.

With this line and above changes i could release about 46kB in arduino ide:
` esp_bt_controller_mem_release(ESP_BT_MODE_CLASSIC_BT);`

ifrew, 2017-11-27:
Awesome...ill get on that shortly and try. Thanks for the help and quick response..much appreciated.

ifrew, 2017-11-27:
Thanks for reminding me to test my ble_example that failed with the esp_bt_controller_mem_release(ESP_BT_MODE_CLASSIC_BT); commented out. Yes it does work once I do that. Releasing the memory causes the guru mediation error that I was getting. 

So it is heap related! Ok..only way for my main program to work now is to free up memory before initializing the ble stack. I'll try your wifi suggestions and see what happens!

Thanks!

ifrew, 2017-11-27:
Just as a final thought, I turned back the debug logging to level 1 to save more memory and it now runs!!!! So the wifi tweak and debug log level set to 1 got me going!!! Sweet for now. Hopefully in the bluedroid stack they can improve the memory consumption!!!! 

Happy the man! Many thanks for the insights!

chegewara, 2017-11-27:
Glad it works, but you have to remember its like a mine field. It may work now, but tommorow it may stop working "with no reason". 

Happy programming.

nkolban, 2017-11-27:
My hope is that when ESP-IDF 3.0 comes out we will be stable.  At that point there will be two branches of cpp_utils.  One that will be "supported" against ESP-IDF 3.0 and one that will build and run against "best so far" on ESP-IDF master.   The thinking being that folks will be building apps on a released ESP-IDF.  Today, folks wanting to use cpp_utils are forced to use the pre-3.0 ESP-IDF master rather than the released ESP-IDF 2.1.1.

ifrew, 2017-11-28:
All, starting to make sense now. Thanks Neil. . Ble is running but now since I've tweaked the WiFi stack i can't connect to the WiFi. Tried increasing the Rx buffer to 3 but still fails. Increasing to 4 it crashes., we only really use it to configure parameters for our app via a web page.  If we could just allow one connection as that is all there needs to be that would be great to get going.

chegewara, 2017-11-28:
I will work on it later today if i have time.

ifrew, 2017-12-11:
Just back from vacation. Has there been update/progress since I've been gone on 
? Has a new IDF core being updated? I'm not sure best way for how to check for that. Any info appreciated.

nkolban, 2017-12-11:
The 3.0 release is now at "Release Candidate 1" ... see:

https://github.com/espressif/esp-idf/releases

This means that Espressif is likely to be releasing it within the next month (guess).

chegewara, 2017-12-11:
I have an idea. We are almost ready to switch to newest toolchain, just we are waiting that c++ exceptions will be implemented in arduino stack, so we can start from point zero. Can we wait few more days? In mean time we can refresh info about issue.

ifrew, 2017-12-11:
Sounds good! Just didn't want to change things yet if there was a release soon that i can sync too.

Thanks for quick response!!! Much appreciated. Your plan sounds solid.

Cheers
Iain

ifrew, 2017-12-16:
Just a quick question, I'm assuming that the esp32 Arduino core will be updated once IDF3.0 is fully released? ie, they won't issue a esp32 Arduino pre release candidate will they?

nkolban, 2017-12-16:
The owner of the Arduino libraries for ESP32 is Mr @me-no-dev .  He will be the one to properly answer the question.  I would be astonished if he didn't release a new build when ESP-IDF 3.0 is fully released.  I don't know if he is holding off for just now or not.

ifrew, 2017-12-16:
Thanks for the update Neil..Finally got the i2c interface to be stable thanks to.work by @stickbreaker Just the BLE to get going and.finally on the home stretch.  Maybe download  the IDF and move the project there to try out.

nkolban, 2017-12-17:
That should work.  The Arduino environment allows one to build either using the Arduino IDE or have the Arduino libraries available as a component in ESP-IDF.

nkolban, 2018-01-04:
Closing for just now.  Re-open as needed.

ifrew, 2018-01-04:
FYI.. I had tried to upgrade to the new idf 3.0 core myself but too many errors. Will wait for the pros to do it :-). There are a few changes needed in your library Neil as I am sure you will know. I fixed them but then got a bunch more errors.

chegewara, 2018-01-04:
Yes, trying to use master branch with arduino-ide is a pain if you have to fix all errors. Im doing this from time to time, last time about 2 weeks ago and today im preparing new library for arduino because new HID class.

joey1442, 2018-04-23:
Hi I am running a wificlientsecure example on ESP32 Arduino latest commit.
I am attempting to release more free heap and have tried to include a directive in the main .ino 
#include <WiFiClientSecure.h>
#define CONFIG_MBEDTLS_SSL_MAX_CONTENT_LEN 4096
and also set  #define CONFIG_MBEDTLS_SSL_MAX_CONTENT_LEN 4096  in the sdkconfig.h , but neither seem to have any effect on the free heap compared to the default MBEDTLS_SSL_MAX_CONTENT_LEN 16384 . 
[V][ssl_client.cpp:48] start_ssl_client(): Free heap before TLS 150920 ......are practically identical in both cases ?

jsmith1717, 2019-02-19:
I'm having the same problem. The example BLE server (ble_server.ino) works great by itself. I have a large app that runs wifi, http server, client, etc which also works great and never crashes. When I put the ble_server code into my app, it crashes on BLEDevice::init("..."), just like it did for ifrew. Weirdly, it doesn't get a guru meditation error with a stackdump that I can trace. Instead, it only prints the reset code plus fatal exception:

```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:952
load:0x40078000,len:6084
load:0x40080000,len:7936
entry 0x40080310
Fatal exception (0): IllegalInstruction
epc1=0x40080342, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000 
```

I added `-Map=arduino.map` to our custom config.mk to generate a map file, so I could find out where it was crashing. It turns out that `epc1=0x40080342` is in the exception handler. So the error is no help in finding the problem. I only found out what line it crashes on by commenting out code til it stopped crashing.

I then ran ifrew's ble example.ino. I had to make a new partition .csv and new board in board.txt with more app space to compile it. Weirdly, I did not have to do that to compile my large app with knolban's ble_server code in it. I don't understand how ifrew's small example can be larger than my app + Neil's BLE.

Anyway, I tried uncommenting the `esp_bt_controller_mem_release` in BleDevice.cpp. Still crashed. Then I commented it, and added the line to `setup()` per chegewara's instructions. Still crashed. Then I edited sdkconfig.h, even though the top of the file says autogenerated and "do not edit". Still crashes. I can't get it to run long enough to print the free heap space. On my big app with no BLE I have 162K.

I downloaded the esp32 core for Arduino in August 2018. I'm reluctant to download the latest, because I have three big projects close to release that use the 2018 core.

I'd really appreciate some help in dealing with this. I'm about to release three products that all require Bluetooth. Everything else works great, and yet I'm stonewalled at the finish line.