nkolban/esp32-snippets #231: components/cpp_utils build error

HarrisonOfTheNorth, 2017-11-26:
Hello, I build from the bash and have cpp_utils in my components folder as instructed, when going make flash monitor it starts building but when it gets to HttpServer it throws as follows:

```
CXX build/cpp_utils/HttpServer.o
MyPath/components/cpp_utils/./HttpServer.cpp: In member function 'virtual void HttpServerTask::run(void*)':
MyPath/components/cpp_utils/./HttpServer.cpp:195:26 error: exception handling disabled, use -fexceptions to enable
    catch(std::exception &e) {
                          ^
make[1]: *** [HttpServer.o] Error 1
make: *** [component-cpp_utils-build] Error 2

```

Any suggestions?

Kind regards,

Anthony

chegewara, 2017-11-26:
Please check if you have enabled exception handling in menuconfig:
`make menuconfig`

menuconfig->compiler options-> C++ exceptions

nkolban, 2017-11-26:
Issue #232 was just created and code changed against it.  Now when we try and compile code that expects C++ exception handling to be present and it isn't switched on, the compilation checks for it to be on and, if note, logs a more instructive message telling the user to switch on with some hints on how to do that.

jestebanes, 2017-11-29:
Hello Neil, but how can we do the "make menuconfig". If I try in esp32-snippets/cpp_utils the system answer 
"There is no rule to the objective 'menuconfig'"

Kind regards

chegewara, 2017-11-29:
You need to to use `make menuconfig` in your project folder, not in cpp_utils.

nkolban, 2017-11-29:
See also:

http://esp-idf.readthedocs.io/en/latest/get-started/index.html#start-a-project

jestebanes, 2017-11-29:
thanks, but I am Arduino user and compile with makefile.arduino in cpp_utils and the failure is on the Arduino side but referred to BLE library. I try to do the menuconfig on other projects but didn't work.

Best regards

nkolban, 2017-11-29:
@jestebanes Have a read through this issue - #228 

In the latest BLE C++ classes we started to exploit C++ exception handling.  In the latest ESP-IDF, Espressif have given us the ability to throw an exception.  This is great.  It means that code can throw an exception if it fails rather than return a failing return code which must always be checked.  If your read #228 you will find that although exception handling is not part of ESP-IDF, it is not turned on by default.  The Arduino-ESP32 apparently ships a pre-compiled version of the ESP-IDF as part of its distribution.  If one is using Arduino IDE, then one uses that ESP-IDF ... which has not yet had exception handling switched on.  Mr @me-no-dev believes he will be able to turn that on in a future release of the Arduino-ESP32 libraries.  In the interim, your choices are:

1. Use the BLE C++ classes that come with the Arduino-ESP32 package which don't have exception handling in them.
2. Use the latest BLE C++ classes with the Arduino-ESP32 package but don't use the Arduino IDE but instead use the Arduino-ESP32 package as an ESP-IDF component.  This will allow you to change the configuration options of how the ESP-IDF is compiled including the ability to switch on C++ exception handling.

dstoiko, 2018-04-05:
So apparently it doesn't make it a default when compiling? We are using `make defconfig` in our CI script on a remote linux machine and are now getting a build error because of that. How can this be enabled without using the `menuconfig` UI?