nkolban/esp32-snippets #233: Link error : symbol mbedtls_debug_set_threshold not defined

4rk4ik, 2017-11-27:
Hello, I just started to use paho as mqtt client and I realized after including it to my project that an error occured at linking step.

Here is the trace :
`$ make
CC main.o
AR libmain.a
LD app-template.elf
<project_path>/build/paho_mqtt_embedded_c/libpaho_mqtt_embedded_c.a(MQTTLinux.o):(.literal.NetworkConnectSSL+0x38): undefined reference to `mbedtls_debug_set_threshold'
<project_path>/build/paho_mqtt_embedded_c/libpaho_mqtt_embedded_c.a(MQTTLinux.o): In function `NetworkConnectSSL':
<project_path>/components/paho_mqtt_embedded_c/MQTTClient-C/src/linux/MQTTLinux.c:203: undefined reference to `mbedtls_debug_set_threshold'
collect2: error: ld returned 1 exit status
<esp-idf_path>/make/project.mk:322 : la recette pour la cible « <project_path>/build/app-template.elf » a échouée
make: *** [<project_path>/build/app-template.elf] Erreur 1`

The error comes from the line 218 in MQTTClient-C/src/linux/MQTTLinux.c, once commented, everything is ok.

nkolban, 2017-11-27:
Howdy ... I think I may know the problem.  Ive added some quick instructions in the README which can be read here:

https://github.com/nkolban/esp32-snippets/tree/master/networking/mqtt/paho_mqtt_embedded_c

See the ESP32 Specific information at the bottom of the page.  Please let me know if this helps.

4rk4ik, 2017-11-28:
Didn't see it at first reading... Enabling mbedTLS debug did the trick.
You can close the issue.

closed 2017-11-28 by nkolban