nkolban/esp32-snippets #234: How to write Makefile

blueMoods, 2017-11-28:
For issues: https://github.com/espressif/esp-idf/issues/857 
I add Makefile at esp32-snippets/cpp_utils/tests/BLETests/
The contents of the Makefile are as follows:
PROJECT_NAME := BLETests

COMPONENT_ADD_INCLUDEDIRS := components/include

include $(IDF_PATH)/make/project.mk

But I make failed ,the log are as follows:
/home/baohongde/github/customer/esp32-snippets/cpp_utils/tests/BLETests/build/esp32/libesp32.a(cpu_start.o):(.literal.main_task+0xc): undefined reference to app_main' /home/baohongde/github/customer/esp32-snippets/cpp_utils/tests/BLETests/build/esp32/libesp32.a(cpu_start.o): In functionmain_task':
/home/baohongde/develop/esp-idf/components/esp32/./cpu_start.c:392: undefined reference to `app_main'
collect2: error: ld returned 1 exit status
/home/baohongde/develop/esp-idf/make/project.mk:359: recipe for target '/home/baohongde/github/customer/esp32-snippets/cpp_utils/tests/BLETests/build/BLETests.elf' failed
make: *** [/home/baohongde/github/customer/esp32-snippets/cpp_utils/tests/BLETests/build/BLETests.elf] Error 1

Could you please help to find where i was wrong. And how to write Makefile.

chegewara, 2017-11-28:
Im assuming your main file is cpp, then you need to add extern in it because linker is expecting that `void app_main(void)` is plain C function. Just add this in you main.cpp or whatever you call it:

```
extern "C" {
    void app_main(void);
}
```

blueMoods, 2017-11-28:
@chegewara Thanks,  this sentence already exists in my main.cpp. but stil failed.

chegewara, 2017-11-28:
Ok then. Do you have components.mk file in your main subfolder?

chegewara, 2017-11-28:
Basicaly in most cases its just an empty file

blueMoods, 2017-11-28:
Like the picture, I add the empty file , but still make failed
[image]

chegewara, 2017-11-28:
You need to use this as start of your project:
https://github.com/espressif/esp-idf-template
Then copy main.cpp and all Sample files you want to use, like SampleServer.cpp or SampleClient.cpp to main subfolder, add subfolder components in template project and copy or symlink into components cpp_utils. Then you will have tree like this:
[image]

nkolban, 2017-11-28:
Thanks for the questions mr @blueMoods  ... delighted that we were able to help you out.  When satisfied, please close this issue.  If you have other puzzles related to using the cpp_utils project, please go ahead and create any new issues you need.

closed 2017-11-29 by blueMoods