nkolban/esp32-snippets #236: Ｉ have a question. How to detect RSSI from BLE on ESP32?

minwinmin, 2017-11-29:
Ｉ have a question. 
How to detect RSSI from BLE on ESP32 ?
I want to know RSSI from ESP32.
But I wasn`t able to understand a program.
I am a beginner...sorry.

nkolban, 2017-11-29:
When you are connected to a BLE Server with the ESP32 acting as a BLEClient, there is a method on BLEClient called getRssi().  When called, it returns the RSSI value associated the partner.

If you are truly a beginner, then the above may not make overly much sense.  There is little I can offer in an issue response other than to offer encouragement to persevere.   There many great communities on programming, C programming, C++ programming and of course ESP32 specific.  I always follow the paradigm of crawl/walk/run ... get something really simple going and then build upon it.

minwinmin, 2017-11-30:
I appreciate for your advice.
I try to write a program like this picture.
I used getRssi();.
But error happened...
[image]

minwinmin, 2017-11-30:
#include <BLEDevice.h>
#include <BLEServer.h>
#include <BLEUtils.h>
#include <BLE2902.h>
#include <BLEAdvertisedDevice.h>

BLECharacteristic *pCharacteristic;
bool deviceConnected = false;
uint8_t value = 0;

// See the following for generating UUIDs:
// https://www.uuidgenerator.net/

#define SERVICE_UUID        "4fafc201-1fb5-459e-8fcc-c5c9c331914b"
#define CHARACTERISTIC_UUID "beb5483e-36e1-4688-b7f5-ea07361b26a8"

class MyServerCallbacks: public BLEServerCallbacks {
    void onConnect(BLEServer* pServer) {
      deviceConnected = true;
    };

    void onDisconnect(BLEServer* pServer) {
      deviceConnected = false;
    }
};

void setup() {
  Serial.begin(115200);

  // Create the BLE Device
  BLEDevice::init("MyESP32");

  // Create the BLE Server
  BLEServer *pServer = new BLEServer();
  pServer->setCallbacks(new MyServerCallbacks());

  // Create the BLE Service
  BLEService *pService = pServer->createService(SERVICE_UUID);

  // Create a BLE Characteristic
  pCharacteristic = pService->createCharacteristic(
                      CHARACTERISTIC_UUID,
                      BLECharacteristic::PROPERTY_READ   |
                      BLECharacteristic::PROPERTY_WRITE  |
                      BLECharacteristic::PROPERTY_NOTIFY |
                      BLECharacteristic::PROPERTY_INDICATE
                    );

  // https://www.bluetooth.com/specifications/gatt/viewer?attributeXmlFile=org.bluetooth.descriptor.gatt.client_characteristic_configuration.xml
  // Create a BLE Descriptor
  pCharacteristic->addDescriptor(new BLE2902());

  // Start the service
  pService->start();

  // Start advertising
  pServer->getAdvertising()->start();
  Serial.println("Waiting a client connection to notify...");
}

void loop() {
  float h = getRssi();
  Serial.printf(h);
  if (deviceConnected) {
    Serial.printf("*** NOTIFY: %d ***\n", value);
    pCharacteristic->setValue(&value, 1);
    pCharacteristic->notify();
    //pCharacteristic->indicate();
    value++;
  }
  delay(2000);
}

minwinmin, 2017-11-30:
I used a sample program, "BLE_notify".

nkolban, 2017-11-30:
Howdy,
The getRSSI() function is part of the BLEClient class ... 

https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/BLEClient.cpp#L281

What I need to do is study and see if I can use the same technology and expose a getRSSI() on a BLEServer object.  Flagging this issue for study.

venkyyalagala, 2018-01-22:
MINWINMIN , DID YOU GOT THE RSSI VALUE FOR THE CODE WHICH YOU HAVE MENTIONED ABOVE ?? 
 please help me it is throwing an error as getRssi() as not declared 

this is the code 
#include <BLEDevice.h>
#include <BLEServer.h>
#include <BLEUtils.h>
#include <BLE2902.h>
#include <BLEAdvertisedDevice.h>

BLECharacteristic *pCharacteristic;
bool deviceConnected = false;
uint8_t value = 0;

// See the following for generating UUIDs:
// https://www.uuidgenerator.net/

#define SERVICE_UUID        "4fafc201-1fb5-459e-8fcc-c5c9c331914b"
#define CHARACTERISTIC_UUID "beb5483e-36e1-4688-b7f5-ea07361b26a8"

class MyServerCallbacks: public BLEServerCallbacks {
    void onConnect(BLEServer* pServer) {
      deviceConnected = true;
    };

    void onDisconnect(BLEServer* pServer) {
      deviceConnected = false;
    }
};

void setup() {
  Serial.begin(115200);

  // Create the BLE Device
  BLEDevice::init("MyESP32");

  // Create the BLE Server
  BLEServer *pServer = new BLEServer();
  pServer->setCallbacks(new MyServerCallbacks());

  // Create the BLE Service
  BLEService *pService = pServer->createService(SERVICE_UUID);

  // Create a BLE Characteristic
  pCharacteristic = pService->createCharacteristic(
                      CHARACTERISTIC_UUID,
                      BLECharacteristic::PROPERTY_READ   |
                      BLECharacteristic::PROPERTY_WRITE  |
                      BLECharacteristic::PROPERTY_NOTIFY |
                      BLECharacteristic::PROPERTY_INDICATE
                    );

  // https://www.bluetooth.com/specifications/gatt/viewer?attributeXmlFile=org.bluetooth.descriptor.gatt.client_characteristic_configuration.xml
  // Create a BLE Descriptor
  pCharacteristic->addDescriptor(new BLE2902());

  // Start the service
  pService->start();

  // Start advertising
  pServer->getAdvertising()->start();
  Serial.println("Waiting a client connection to notify...");
}

void loop() {
  float h = getRssi();
  Serial.printf(h);
  if (deviceConnected) {
    Serial.printf("*** NOTIFY: %d ***\n", value);
    pCharacteristic->setValue(&value, 1);
    pCharacteristic->notify();
    //pCharacteristic->indicate();
    value++;
  }
  delay(2000);
}
 :-1: 1

venkyyalagala, 2018-02-08:
please help me to find

farabiahmed, 2018-04-05:
You can get RSSI by modifying BLE_scan example like below:

```
class MyAdvertisedDeviceCallbacks: public BLEAdvertisedDeviceCallbacks {
    void onResult(BLEAdvertisedDevice advertisedDevice) {
      Serial.printf("Advertised Device: %s", advertisedDevice.toString().c_str());
      
      if (advertisedDevice.haveRSSI()){
        Serial.printf("Rssi: %d \n", (int)advertisedDevice.getRSSI());
      }
      else Serial.printf("\n");
  
    }
};
```

It is advised to modify your BLEAdvertisedDevice.toString() method like given below:
Path of file: \espressif\esp32\libraries\BLE\src\BLEAdvertisedDevice.h

```
std::string BLEAdvertisedDevice::toString() {
	std::stringstream ss;
	ss << "Name: " << getName() << ", Address: " << getAddress().toString();
	if (haveAppearance()) {
		ss << ", appearance: " << getAppearance();
	}
	if (haveManufacturerData()) {
		char *pHex = BLEUtils::buildHexData(nullptr, (uint8_t*)getManufacturerData().data(), getManufacturerData().length());
		ss << ", manufacturer data: " << pHex;
		free(pHex);
	}
	if (haveServiceUUID()) {
		ss << ", serviceUUID: " << getServiceUUID().toString();
	}
	if (haveTXPower()) {
		ss << ", txPower: " << (int)getTXPower();
	}
        if (haveRSSI()) {
		ss << ", Rssi: " << (int)getRSSI();
	}	
		
	return ss.str();
} // toString
```

khevu8122, 2023-03-15:
Hello I want to ring the buzzer and send notification to my phone according to distance of esp 32 and my phone bluetooth but I am not able to run the advertising class can you help?