nkolban/esp32-snippets #237: Can't compile BLE stuff since IDF update

DonWasabi, 2017-11-29:
Hi, 
I updated IDF (and recursively the BLE submodule, but dunno if there were any changes) some days ago and since then i can't compile anymore.. Has someone encountered similar issues? Here are the error outputs:
https://pastebin.com/ft5CrRLh

Kind regards :)

nkolban, 2017-11-29:
When did you last refresh the BLE code from this repository?  A few months back, Espressif changed the exposed BLE APIs in their pre-3.0 release.  This necessitated a change in the BLE C++ classes to be compatible.  Is it possible that you were running a relatively old ESP-IDF from master and now that you have pulled the latest master, you haven't also pulled the latest BLE C++ classes?

DonWasabi, 2017-11-30:
I like problems that are none in the first place and thus can be solved easily :) 
Thank you for your fast and helpful reply

nkolban, 2017-11-30:
Pleasure my friend ... and keep coming back.  Open as many issues as desired.  When you are completely satisfied with this one, please close it out.  We walk each of the issues regularly to see if work needs done to make progress.

closed 2017-12-09 by DonWasabi