nkolban/esp32-snippets #238: Find/Set 'MAC' Address of the ESP32

Atzingen, 2017-11-30:
I have the need to set the bluetooth MAC address.  If this is not possible for some reason (like been hard coded to the chip and not accesible with the library), then I would like to be able to get this info in the arduino code (somethin like BLEDevice::init(bt_name, MAC) or MAC = BLEDevice.getMac() ). 
 
I'm currently able to set the name of the device, the service uuid an the characteristics but did not find this in the library code.

Thanks

nkolban, 2017-11-30:
We have now added a new static method called BLEDevice::getAddress().

While coded, it has not yet been tested other than to validate a clean compile.  When called after a BLEDevice::init(), it should return the BLE address of the ESP32.  To the best of my knowledge, the BDA is hard-coded.  I believe one can configure a "random" flag where the address is created dynamically but randomly ... but I haven't seen any APIs to set it.

This change is in the cpp_utils master and not yet packaged with the Arduino-ESP32 builds.  It is my understanding that the only way to use this currently is by using the Arduino-ESP32 as an ESP-IDF component until such time as the Arduino-ESP32 libs are rebuilt to switch on / enable C++ exception handling.  If you are coding in direct ESP-IDF, there should be no issues.

Atzingen, 2017-12-01:
Great, thanks alot for the super fast reply and help !

closed 2017-12-01 by Atzingen

ArTyGill, 2018-08-11:
https://esp-idf.readthedocs.io/en/latest/api-reference/system/base_mac_address.html 
You can change the base mac-address and wifi and bluetooth will set your value.

ArTyGill, 2018-10-11:
The last link does not work, it may have been removed from the latest documentation.
Here is the link to the old documentation.
https://docs.espressif.com/projects/esp-idf/en/v3.0.4/api-reference/system/base_mac_address.html

Find function: esp_base_mac_addr_set()

Galaxy-Man, 2019-07-02:
Hi 
I am new to BT, is there a simple Arduino IDe code to use this method?
BLEDevice::getAddress()
I just want to print the ESP32 BT address to the serial port.
Thank you for any help

chegewara, 2019-07-03:
If i remember there is no function to find or set own MAC.

m-menon, 2020-07-31:
> https://esp-idf.readthedocs.io/en/latest/api-reference/system/base_mac_address.html
> You can change the base mac-address and wifi and bluetooth will set your value.

@ArTyGill is there any example or documentation on how that is done? I have a peculiar case where I need to set the Mac address (say in this [example](https://github.com/nkolban/ESP32_BLE_Arduino/tree/master/examples/BLE_iBeacon)). Thanks

ArTyGill, 2020-08-01:
@m-menon https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/system.html?highlight=mac#base-mac-address

try this method.

m-menon, 2020-08-01:
Thanks @ArTyGill ... this worked

https://github.com/h2zero/NimBLE-Arduino/issues/108#issuecomment-667462767

ArTyGill, 2020-08-01:
@m-menon 👍

Squirrelbd, 2024-08-10:
[Find and Change the MAC Address of ESP32](https://www.programmingboss.com/2024/08/esp32-over-air-ota-programming-with.html) by following this guide which works for me.

mnmanas, 2024-08-15:
Hi