nkolban/esp32-snippets #239: ESP32 crashes constantly after loading BLE_write example (Wemos Lolin32 Lite)

legendary6, 2017-12-01:
Hello, I just loaded the example without any modification and the board constantly crashes and reboots during setup. The same kind of crash happens during BLE_server example too(although it does work sometimes). Here's the serial dump that I got -

```
rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0010,len:4
load:0x3fff0014,len:588
load:0x40078000,len:0
load:0x40078000,len:9768
entry 0x400789b8
1- Download and install an BLE scanner app in your phone
2- Scan for BLE devices in the app
3- Connect to MyESP32
4- Go to CUSTOM CHARACTERISTIC in CUSTOM SERVICE and write something
5- See the magic =)
abort() was called at PC 0x400d8823 on core 0

Backtrace: 0x400887fc:0x3ffdd190 0x400888fb:0x3ffdd1b0 0x400d8823:0x3ffdd1d0 0x400d9e92:0x3ffdd1f0 0x400da01e:0x3ffdd210 0x400da369:0x3ffdd250 0x400da3a9:0x3ffdd280 0x400da3b9:0x3ffdd2a0 0x400d62d6:0x3ffdd2c0 0x400d4f51:0x3ffdd2e0 0x400d3d65:0x3ffdd320 0x400f2da5:0x3ffdd340 0x400eefac:0x3ffdd380

Rebooting...
ets Jun  8 2016 00:22:57

```

nkolban, 2017-12-01:
The text in your report:

```
1- Download and install an BLE scanner app in your phone
2- Scan for BLE devices in the app
3- Connect to MyESP32
4- Go to CUSTOM CHARACTERISTIC in CUSTOM SERVICE and write something
5- See the magic =)
```
Seems to being produced by the application you installed in your ESP32.  However, that text isn't at all familiar to me.  Can you describe exactly which application that comes with this repository/library you are running?

nkolban, 2017-12-01:
Ahh .. found it ... it appears to be written by mr @copercini  here:

https://github.com/nkolban/esp32-snippets/commit/209d8ddf104d38e1623c602587731f54aa30b911

Our next puzzle is to determine where within the code the application is failing.  I believe you are running an Arduino environment.  If so, we want to install the back-trace analysis tool that will take the back trace codes and tell us where in the actual source the exception is occurring.

See:

https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/ArduinoBLE.md#decoding-an-exception-stack-trace

copercini, 2017-12-01:
@legendary6  Are you running the last Arduino core version?

legendary6, 2017-12-01:
Thank you @nkolban for replying so quickly! I'm haven't used ESP32 much so forgive me if I struggle with some basic things. So the board I'm using is Wemos Lolin32 and Arduino IDE doesn't show an option to change the debug level as you showed. I also have an ESP32 dev module which I suppose has that option so let me get back to you on that one. @copercini I searched but couldn't find the core version number anywhere in /Arduino/hardware/espressif/esp32. Is there any particular file which mentions the core version?

legendary6, 2017-12-01:
Ok so I tried uploading the same code to the dev module and it worked perfectly so there's certainly something wrong with just this module. I also uploaded [BLE_server](https://github.com/nkolban/ESP32_BLE_Arduino/blob/master/examples/BLE_server/BLE_server.ino) example and it still crashes but sometimes it starts working after crashing a few times. The serial output is like this -

```
Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0010,len:4
load:0x3fff0014,len:588
load:0x40078000,len:0
load:0x40078000,len:9768
entry 0x400789b8
Starting BLE work!
abort() was called at PC 0x400d872f on core 0

Backtrace: 0x400887fc:0x3ffde190 0x400888fb:0x3ffde1b0 0x400d872f:0x3ffde1d0 0x400d9d9e:0x3ffde1f0 0x400d9f2a:0x3ffde210 0x400da275:0x3ffde250 0x400da2b5:0x3ffde280 0x400da2c5:0x3ffde2a0 0x400d61f6:0x3ffde2c0 0x400d4e71:0x3ffde2e0 0x400d3c85:0x3ffde320 0x400f2cb1:0x3ffde340 0x400eeeb8:0x3ffde380

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0010,len:4
load:0x3fff0014,len:588
load:0x40078000,len:0
load:0x40078000,len:9768
entry 0x400789b8
Starting BLE work!
Characteristic defined! Now you can read it in your phone!
```

copercini, 2017-12-01:
@legendary6  there is not an exact way, check the files modification date and if have more than two weeks, try to update the core to the last version, following this: https://github.com/espressif/arduino-esp32/blob/master/docs/arduino-ide/windows.md#how-to-update-to-the-latest-code

 @nkolban I can update it to a more serious text if you like

legendary6, 2017-12-01:
Ok so I think I found the culprit. The core files were dated at 25/8/2017 so I updated the core to the latest version and now everything works great! Thank you very much @copercini and @nkolban  for your help!

copercini, 2017-12-01:
=)
yeah, I remember this error from the old versions, then @nkolban found that it need a small delay before start the BLE task on arduino

nkolban, 2017-12-01:
This is great news.  Thank you mr @copercini 

Note sure what you mean by "I can update it to a more serious text if you like"?  The text messages are fine ... we are all in this game together and if you like those messages, you go with them.

@legendary6 Good news you got it working.  If you think there is a place we should better document these steps (whether in this project or in the Arduino-ESP32 project), your "give back" can always be to work with the authors/documenters of those areas and give them input on improving the clarity or location of the information you needed.  Heck, you can ever write your own words in the repositories or create a blog post.

Close this issue when you are delighted.  You can always create new issues on new topic if and when they come up.

legendary6, 2017-12-01:
@nkolban I will certainly help this community in whatever way I can. I'm a student and I'm working on a project with BLE which is pretty new to me and I tried to get my head around ESP IDF's API but for a newcomer like me it was too convoluted to understand. If not for this library and your support it would take days for me to get a working prototype. Thank you again!

closed 2017-12-01 by legendary6

nkolban, 2017-12-01:
Its great you are a student.  We have all kinds of folks visiting ... some are completely new, some are students, some are hobbyists, some are professionals ... from the super young to grey beards like myself who remember encouragement and support from grey beards when "I were a lad" ... as long as we keep helping each other and passing on knowledge in friendly fashions ... all is well.