nkolban/esp32-snippets #242: Provide convenience functions

nkolban, 2017-12-02:
While studying BLE Noble, I found the existence of convenience functions in that library.  What a great idea!!  The following are candidates for conveniences in this library:

```
BLEDevice::write(bdAddress, serviceUuid, characteristicUuid, value)
value BLEDevice::read(bdAddress, serviceUuid, characteristicUuid)
BLEClient::write(serviceUuid, characteristicUuid, value)
value BLEClient::read(serviceUuid, characteristicUuid)
BLERemoteService::write(characteristicUuid, value)
value BLERemoteService::read(characteristicUuid)
```

nkolban, 2017-12-02:
Easiest ones now implemented:

```
BLERemoteService::setValue(BLEUUID characteristicUuid, std::string value)
std::string BLERemoteService::getValue(BLEUUID characteristicUuid);
```

nkolban, 2017-12-03:
We have now added:

```
BLEClient::setValue(BLEUUID serviceUUID, BLEUUID characteristicUuid, std::string value)
std::string BLEClient::getValue(BLEUUID serviceUUID, BLEUUID characteristicUuid)

BLEDevice::setValue(BLEAddress bdAddress, BLEUUID serviceUUID, BLEUUID characteristicUuid, std::string value)
std::string BLEDevice::getValue(BLEAddress bdAddress, BLEUUID serviceUUID, BLEUUID characteristicUuid)
```

Coding is now likely to be complete but testing needs performed.  We can be loose on testing as the changes affect only new code and not existing.

VDelport, 2020-09-04:
Thank you for your great project. :)
Would you please add getValue and setValue functions for BLEDevice that can be used to read or write byte arrays?

chegewara, 2020-09-06:
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/BLERemoteCharacteristic.h#L49
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/BLERemoteCharacteristic.h#L53