nkolban/esp32-snippets #245: Error in BootWiFi.cpp

squonk11, 2017-12-04:
I tried to compile BootWifi and get the error message that "GeneralUtils::restart();" is not defined. I think the restart() function is included in System.cpp.
Best regards
   Lothar

nkolban, 2017-12-06:
Thank you sir.  I really am an awful maintainer ... sigh ... sorry about the hassle.  I have now corrected the code and resolved the issue.  Again ... humble and sincere apologies.

alfredomrh, 2017-12-06:
Hi, I get this error now:

/components/cpp_utils/./PWM.cpp:36:13: error: 'struct ledc_timer_config_t' has no member named 'duty_resolution'
  timer_conf.duty_resolution    = dutyResolution;
Thank you

nkolban, 2017-12-06:
Howdy,
What that means is that you have a back-level version of the ESP-IDF.  In the last few weeks, Espressif changed one of the major data structure names which required that I change the source code.  Please review your procedures and download the latest ESP-IDF and test again.

squonk11, 2017-12-06:
Now I have the BootWiFi example running. But unfortunately it only works if I connect with my tablet which only has Wifi and no GSM. If I try to connect to the accesspoint (for setting the station credentials) using my mobile phone (Android 7.1) it does not work. I think that the mobile phone routes the access to 192.168.4.1 into the GSM network. If I deactivate GSM on my mobile phone it works like with my tablet.
What is missing here? Do I need to activate mDNS or set some DNS server?

nkolban, 2017-12-06:
My experience has been that when I am running an ESP32 and it boots up as an access point I can then find it in my list of available WiFi networks on my phone.  If I then connect to the ESP32 as an access point, and then open a browser to http://192.168.4.1 ... my phone uses WiFi to connect to the ESP32 to show the page.

squonk11, 2017-12-06:
...hm - my phone unfortunately in this situation only tries to connect via GSM and I get the error message that the site 192.168.4.1 is not reachable. Do you have a hint what else I could try?

nkolban, 2017-12-07:
We could attach your phone to Vysor and then screen share what you see on your phone?  By virtue of the fact that the tablet "seems to work as desired" ... we might be looking at a setting on the phone perhaps.  I am assuming that in your browser you are trying to reach the URL:

http://192.168.4.1

What flavor of browser are you using on your phone?  Might you be able to try a different one?  Do you have proxy settings configured that might be routing you to the Internet?

alfredomrh, 2017-12-07:
Hello, first of all thank you for your time and work, it helps me a lot to understand the world of ESP32.

I would like to use this component (bootWifi) but I still have problems compiling. I've updated the esp-idf and toolchain. I have copied bootwifi and cpp_utils from their repository to the components folder of my project and both subfolders have their component.mk.

The first time I've compiled, I get this error:

components/cpp_utils/. /JSON. cpp: In member function' void JsonArray:: addInt (int)':
/home/user/workspace/myproject/components/cpp_utils/. /JSON. cpp: 100:70: error:' cJSON_CreateDouble' was not declared in this scope
  cJSON_AddItemToArray (m_node, cJSON_CreateDouble ((double)value, value));
                                                                      ^
/home/user/workspace/myproject/components/cpp_utils/. /JSON. cpp: In member function' void JsonObject:: setInt (std:: __cxx11:: string, int)':
/home/user/workspace/myproject/components/cpp_utils/. /JSON. cpp: 329:85: error:' cJSON_CreateDouble' was not declared in this scope
 ddItemToObject (m_node, name. c_str (), cJSON_CreateDouble ((double)value, value));

Then I copied the cJSON. h and cJSON. c and cJSON_Utils libraries to the cpp_utils folder, but I noticed that the cJSON_CreateDouble function is not defined, so I chose to comment on the lines in JSON. cpp.

When I comment on both lines, that error disappears, but it throws me this other one:

BootWiFi. h: 11:0,
                 from /home/user/workspace/myproject/main/. /main. c: 15:
/home/user/workspace/myproject/components/cpp_utils/. /WiFi. h: 12:18: fatal error: string: No such file or directory.
compilation terminated.

I don't understand why it includes a string file and not string. h, which I don't have on my system.

I would very much appreciate any help in using this component.

Thank you very much.

chegewara, 2017-12-07:
About 4 days ago there was esp-idf cJSON library refactored. We need to wait for response now from this issue
https://github.com/espressif/esp-idf/issues/1363

chegewara, 2017-12-07:
Now the other issues you have. 
Its how im using bootWiFi library:
- coppy all files from bootWiFi to main folder,
- add cpp_utils to components folder (sym link in my case),
- minimal main.cpp looks like this:
```
#include "BootWiFi.h"
#include "sdkconfig.h"

extern "C" {
	void app_main(void);
}

BootWiFi *boot;

void app_main(void) {
	boot = new BootWiFi();
	boot->boot();
}
```
Program is compiling and working just fine.

alfredomrh, 2017-12-07:
> Now the other issues you have.
Its how im using bootWiFi library:

    coppy all files from bootWiFi to main folder,
    add cpp_utils to components folder (sym link in my case),
    minimal main.cpp looks like this:

#include "BootWiFi.h"
#include "sdkconfig.h"

extern "C" {
	void app_main(void);
}

BootWiFi *boot;

void app_main(void) {
	boot = new BootWiFi();
	boot->boot();
}

Program is compiling and working just fine.

> 
Thanks for your help, Chegewara, but the issue remains. In the first error (cJSON. h) I have commented on the two lines and we will wait for an answer.

This second mistake:
`/components/cpp_utils/./WiFi.h:12:18: fatal error: string: No such file or directory
compilation terminated.
`

I have copied the 6 bootWifi files to my main folder and overwritten the component. mk file, the cpp_utils folder is in the component folder of my project, but I still can't compile.
My project is written in c, so the app_main () function is in my main. c file and not in main. cpp, I don't know if this will have anything to do with it.
Greetings

chegewara, 2017-12-07:
This is the problem. You are trying to use c++ library in c file. This wont work. You have to switch to c++ or create a wrapper to use c++ libraries in c project. But its hell a lot of work. 
You can write your project in c, this is not issue, but you cant use objects created from c++ class in c code.
I dont know if this sounds simple enough.

alfredomrh, 2017-12-07:
Now I understand, I thought that as the compiler is cross gcc and the C/C++ projects, you could use code from both languages. I haven't used C or C++ for 15 years and now I'm trying to catch up.

I suppose the only solution is to use smartconfig Example from esp-idf, but it doesn't look as complete as bootWifi, really??

chegewara, 2017-12-07:
You can use both C and C++. But when you search thru stackoverflow you will understand it more.
You can use C code in C++ with no problem. The problem is when you want to use C++ classes in C code.
This is first link from google search, may be helpfull
https://stackoverflow.com/questions/199418/using-c-library-in-c-code

alfredomrh, 2017-12-07:
Thank you very much, I'll read about it.

Greetings

squonk11, 2017-12-07:
> We  could attach your phone to Vysor and then screen share what you see on  your phone?  By virtue of the fact that the tablet "seems to work as  desired" ... we might be looking at a setting on the [...]
--

Now I tested with Firefox, Chrome and Android browser: the result is always the same: Time out error when trying to connect to 192.168.4.1 and as soon as I switch off GSM it works. There are no proxy settings in the browsers. The operating system I am using is not Android directly, my phone has LineageOS 7.1.1. I also tried to activate mDNS - but also this did not work. Is there anything else I could try? Do you have any more hints?
Greetings
   Lothar

chegewara, 2017-12-07:
This is obvious issue with your phone.

squonk11, 2017-12-07:
Now I tried with another phone, same type (Galaxy S4) but with original Samsung Android ROM: this works perfectly. So obviously it is a problem with LineageoOS.
Thanks for your support.

nkolban, 2017-12-08:
When you connect your phone with this "LineageoOS" to a local WiFi hub, can you ping that local WiFi hub from your phone?  eg. 192.168.1.1 or 192.168.0.1 ... I'm wondering if your phone exclusively uses the cell network even when you think you are using WiFi?

squonk11, 2017-12-08:
Yes, I can ping the local wifi hub at address 192.168.1.1

closed 2017-12-10 by squonk11