nkolban/esp32-snippets #246: BLEClient.ino ESP_GATT_INSUF_AUTHENTICATION

jttaylor86, 2017-12-05:
Hi Folks,

I could do with a bit of help.

I am using a ESP32 as a BLE Client, using the BLE Client.ino code from Neil Kolban and it successfully connects to a BLE Server, however, writing and reading the Remote Characteristic doesn't work currently and I don't know why.

I have the debug option turned on and a selected section pf output in the serial monitor is here:

[D][BLERemoteCharacteristic.cpp:433] readValue(): >> readValue(): uuid: fdce2345-1013-4120-b919-1dbb32a2d132, handle: 27 0x1b
[D][BLEDevice.cpp:105] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 3] ... ESP_GATTC_READ_CHAR_EVT
[D][BLEUtils.cpp:1344] dumpGattClientEvent(): GATT Event: ESP_GATTC_READ_CHAR_EVT
[D][BLEUtils.cpp:1493] dumpGattClientEvent(): [status: ESP_GATT_INSUF_AUTHENTICATION, conn_id: 0, handle: 27 0x1b, value_len: 0]
[D][BLERemoteCharacteristic.cpp:455] readValue(): << readValue(): length: 0

Does anyone know anything about ESP_GATT_INSUF_AUTHENTICATION? I presume this is a problem.

Thanks,

nkolban, 2017-12-05:
Howdy,
There is little in the way of security code added to the current BLE C++ classes because of a simple and stupid problem (on my part) ... namely I haven't found any BLE components that apparently need/use security.  What that means is that I haven't had anything to test again.  What I'd like to do is work with you to get this going ... it may take a little bit ... but it seems that you may actually have a BLE server device that requires security ... and that's something Ive been looking for.

My first question ... what kind of device is the target?  Please share as much of its details as possible.

jttaylor86, 2017-12-06:
Hello Neil,

it is a Hosiden HRM1062 chip.

jttaylor86, 2017-12-06:
Hello,
was is the difference between connecting to a BLE server and being paired with the BLE server?

nkolban, 2017-12-08:
My loose understanding is that "pairing" occurs when one or both parties in the connection originally "distrust" the other.  For example, imagine your cell phone and a bluetooth headset.  We don't want ANY bluetooth headset to listen into your phone ... so when a device wishes to connect, it will explicitly require a positive confirmation from one or both ends.  It is common that a device, once trusted the first time, will want to be trusted from then on.  To achieve that, pairing occurs.  My loose understanding is that the devices exchange some private information that identifies each other and that should they wish to connect again in the future, they will examine their history and "remember" the pairing that occurred in the past and allow the connection to proceed without question.

closed 2017-12-08 by jttaylor86