nkolban/esp32-snippets #249: esp32 interfacing with lsm6ds3

Goutham8128, 2017-12-08:
Hi

I want to interface lsm6ds3 with esp32 and compile it in arduino IDE. I used the sparkfun lsm6ds3 library but the headerfiles were not suitable for esp32.Can i get any guidance?

nkolban, 2017-12-08:
As much as we'd love to attempt to address all questions ... I'm afraid these issues on this Github repository only related to the code found here.  It appears that the device you mentioned is described here:

http://www.st.com/en/mems-and-sensors/lsm6ds3.html

And there is no code in this repository that relates to that.  Your best bet is to frame a question on the general ESP32 forum found here:

https://esp32.com/index.php

chegewara, 2017-12-08:
We has been talking about it on irc channel and asked @Goutham8128 to open issue. I will investigate existing library and see what i can do.

Goutham8128, 2017-12-08:
okay!
I will raise an issue in espressif too.

chegewara, 2017-12-08:
We need to start at some point. To do that, can you try to use any example from this repository https://github.com/sparkfun/SparkFun_LSM6DS3_Arduino_Library
and use this to init wire library:
`Wire.begin(sda, scl, freq);`
sda, scl and freq need to be values that are proper in your app.
Then if its not compiling or compiling but not running paste some logs please.

nkolban, 2017-12-09:
See also:

https://esp32.com/viewtopic.php?f=13&t=3862

Goutham8128, 2017-12-09:
I will try that and get back to you @chegewara .

I am looking into that @nkolban .
Thanks both of you for your assistance

chegewara, 2017-12-09:
You are right, this issue can be closed because its not related to BLE stack and we can resolve it somewhere else.

Goutham8128, 2017-12-09:
yeah thanks for your positive response @chegewara 

And here I paste the logs Mr.kolban @nkolban 
[image]

chegewara, 2017-12-09:
You are trying to use I2C interface and not SPI. Just fork this library and get rig off all related to SPI

Goutham8128, 2017-12-09:
ok

chegewara, 2017-12-09:
By the way. You should nt have SPI library imported to cpp_utils in arduino environment

Goutham8128, 2017-12-09:
I removed the SPI library i.e i commented that line include<SPI.h>
Though i am getting error. I will rebuild and get back to you.

Goutham8128, 2017-12-11:
Previously I have done in windows.Now i have done everything fresh in Linux environment.
I forked out the SPI library.
And You told to use Wire.begin(sda, scl, freq).At which part of the code I have to use this command.
[image]

chegewara, 2017-12-11:
I think here:
https://github.com/sparkfun/SparkFun_LSM6DS3_Arduino_Library/blob/master/src/SparkFunLSM6DS3.cpp#L72

Goutham8128, 2017-12-11:
The same error.

chegewara, 2017-12-11:
Because you didnt remove all SPI instructions from this library. No worries.
Add this line `#define MSBFIRST` in here and check again
https://github.com/sparkfun/SparkFun_LSM6DS3_Arduino_Library/blob/master/src/SparkFunLSM6DS3.cpp#L34

Goutham8128, 2017-12-11:
No change.
1)I commented out #include<SPI.h> in Sparkfun.cpp and the program which i am running
2I added Wire.begin(sda, scl, freq) in Sparkfun.cpp
3)Added #define MSBFIRST as you said in Sparkfun.cpp
Am i going correct?

chegewara, 2017-12-11:
1) is not needed anymore

2 and 3 are correct
is there any new error or the same?

Goutham8128, 2017-12-11:
No
The same one.

chegewara, 2017-12-11:
Then maybe `#define MSBFIRST 1`

Goutham8128, 2017-12-11:
no

it is showing the same error.

chegewara, 2017-12-11:
i will try it today and get back to you later

Goutham8128, 2017-12-11:
okay

Yeah and one doubt it is showing the error as MSBFIRST was not declared in the scope.But when I switch to esp8266 or Arduino UNo board in the arduino Environment it is not showing that error.

chegewara, 2017-12-11:
stupid arduino ide ;)

Goutham8128, 2017-12-11:
Lol yeah!

chegewara, 2017-12-11:
Or maybe its not arduino ide fault, just you guys are finding some weird peripherals?

Goutham8128, 2017-12-11:
No. If its not arduino IDE fault then it should not compile in Uno/esp8266 as well.

What do you mean by  weird peripheral?

chegewara, 2017-12-11:
If you change https://github.com/sparkfun/SparkFun_LSM6DS3_Arduino_Library/blob/master/src/SparkFunLSM6DS3.h like this:
```
#ifndef __LSM6DS3IMU_H__
#define __LSM6DS3IMU_H__

#include "stdint.h"

#define I2C_MODE 0
#define SPI_MODE 1
#define MSBFIRST 1   <------ADD THIS LINE
// Return values 
...
```
all should works. At least this is output log from MinimalistExample:
```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)

ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)

configsip: 0, SPIWP:0xee

clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00

mode:DIO, clock div:2

load:0x3fff0018,len:4

load:0x3fff001c,len:812

load:0x40078000,len:0

load:0x40078000,len:11404

entry 0x40078aa0

Processor came out of reset.

[W][esp32-hal-i2c.c:235] i2cWrite(): Ack Error! Addr: 6b

[W][esp32-hal-i2c.c:334] i2cRead(): Ack Error! Addr: 6b

[W][esp32-hal-i2c.c:235] i2cWrite(): Ack Error! Addr: 6b

[W][esp32-hal-i2c.c:235] i2cWrite(): Ack Error! Addr: 6b

[W][esp32-hal-i2c.c:334] i2cRead(): Ack Error! Addr: 6b

[W][esp32-hal-i2c.c:235] i2cWrite(): Ack Error! Addr: 6b

[W][esp32-hal-i2c.c:235] i2cWrite(): Ack Error! Addr: 6b

[W][esp32-hal-i2c.c:235] i2cWrite(): Ack Error! Addr: 6b

[W][esp32-hal-i2c.c:334] i2cRead(): Ack Error! Addr: 6b

Accelerometer:
 X = [W][esp32-hal-i2c.c:235] i2cWrite(): Ack Error! Addr: 6b

7.9934

 Y = [W][esp32-hal-i2c.c:235] i2cWrite(): Ack Error! Addr: 6b

15.9844

 Z = [W][esp32-hal-i2c.c:235] i2cWrite(): Ack Error! Addr: 6b

15.9844

Gyroscope:
 X = [W][esp32-hal-i2c.c:235] i2cWrite(): Ack Error! Addr: 6b

2292.8501

 Y = [W][esp32-hal-i2c.c:235] i2cWrite(): Ack Error! Addr: 6b

17.9200

 Z = [W][esp32-hal-i2c.c:235] i2cWrite(): Ack Error! Addr: 6b

0.0000

Thermometer:
 Degrees C = [W][esp32-hal-i2c.c:235] i2cWrite(): Ack Error! Addr: 6b

25.0000

 Degrees F = [W][esp32-hal-i2c.c:235] i2cWrite(): Ack Error! Addr: 6b

77.0000

Accelerometer:
 X = [W][esp32-hal-i2c.c:235] i2cWrite(): Ack Error! Addr: 6b

0.0000

 Y = [W][esp32-hal-i2c.c:235] i2cWrite(): Ack Error! Addr: 6b

0.0000

 Z = [W][esp32-hal-i2c.c:235] i2cWrite(): Ack Error! Addr: 6b

0.0000

Gyroscope:
 X = [W][esp32-hal-i2c.c:235] i2cWrite(): Ack Error! Addr: 6b

0.0000

 Y = [W][esp32-hal-i2c.c:235] i2cWrite(): Ack Error! Addr: 6b

0.0000

 Z = [W][esp32-hal-i2c.c:235] i2cWrite(): Ack Error! Addr: 6b

17.9200

```

Goutham8128, 2017-12-12:
ok

I will get back to you. I want the values of LSM6dS3 to be sent to nrfconnect via BLE

Goutham8128, 2017-12-12:
ok. Now you have done changes in sparkfun.h  i.e headerfile.

When i declare wire.begin(sda, scl, freq) .It is saying it is not declared in the scope.

chegewara, 2017-12-12:
You may forget about wire.begin(sda, scl, freq), just make changes from my last post and try MinimalistExample from sparkfun library

Goutham8128, 2017-12-12:
ok.No need to comment SPI?

chegewara, 2017-12-12:
No need. Nothing more just last my post for now counts at the moment

Goutham8128, 2017-12-12:
Ok..Code is compiled.Thank you
Now I will interface it with LSM and check if it is running

Goutham8128, 2017-12-12:
U just uploaded the code to the esp32 board?

chegewara, 2017-12-12:
Yes i did, but i dont have this device so i have a bunch of errors

Goutham8128, 2017-12-12:
But I Doubt without mentioning the pins which are interfaced in the code. How will it run

chegewara, 2017-12-12:
https://raw.githubusercontent.com/gojimmypi/ESP32/master/images/myESP32%20DevKitC%20pinout.png
pins gpio22, gpio23

Goutham8128, 2017-12-12:
[image]

Without interfacing , this is the output log. I will connect and paste you the log.
I am looking where to connect the interrupt pin.

Goutham8128, 2017-12-12:
No need to mention the pins in the program??

Goutham8128, 2017-12-12:
Thank you very much.I have interfaced interrupt pin to D5 of esp32 and pins used 21 , 22 for sda and scl respectively.Now it is working.
Now the interface is completed.Just I want to make the esp32 which is interfaced with LSM6DS3 act as BLE server and a certain BLE client should receive the sensor data .The client may be a generic app like nrfConnect.

nkolban, 2018-01-04:
Closing for just now.

closed 2018-01-04 by nkolban