nkolban/esp32-snippets #250: GATT_RSP_TIMEOUT

jttaylor86, 2017-12-08:
Hi folks,

I am having a wee problem with the GATT_RSP_TIMEOUT when writing to my BLE Server using the ESP32 as a client.

Using the BLE Client.ino
.....
String newValue = "0x070032000000000000000000000000";
pRemoteCharacteristic->writeValue(newValue.c_str(),newValue.length());

always leads to a GATT_RSP_TIMEOUT.

However when I reduce the size of the string to 18 characters I don't receive a timeout, however, this is not helpful as this then isn't the desired command.

I would be very grateful for any help with this.

jttaylor86, 2017-12-12:
Hi Kolban,

I realised that my GATT_RSP_TIMEOUT was due to writing more than the 18 bytes expected by the server.

I am a little bit confused by this statement. Pardon my lack of knowledge.

**// Set the characteristic's value to be the array of bytes that is actually a string.**

What does this mean? Does this mean we pass remote characteristic a string and remote characteristic converts this to an array of bytes?

pRemoteCharacteristic->writeValue(newValue.c_str(), newValue.length());

Following on from this, I have a hex string value = "330841000000000000000000000000000000". How can I write this to my server if I can only send a string of length 18?

I would be very grateful for any help with this.

nkolban, 2017-12-12:
Howdy,
In these classes we are programming C++ which means that we have available to us the whole of the C++ standard library.  This is a huge library and complex at times.  I don't pretend to know all of it.

https://en.wikipedia.org/wiki/C%2B%2B_Standard_Library

One of the classes in standard library is called `std::string`.  The purpose of this class is to hold a string ... or is it :-) ...

If we study the class we find that its purpose is to hold a "string (read as array) of single byte characters".

http://www.cplusplus.com/reference/string/string/

What this means is that an instance of std::string can hold an arbitrary array of characters ... since std::string also manages its own memory ... it has all the capabilities we need to hold a data buffer and that is how it is commonly used.

In C programming, there is no string data type ... instead we have a convention that a C String is a pointer to storage where the string is terminated by a NULL byte.  With std::string, we can hold all values of a bytes (0-255) because the string knows its own length.

So in our BLE classes, when we want to write a value to a remote characteristic ...  we can pass `writeValue` a std::string and write value will send the complete contents of the binary data contained in the std::string

Now to your specific technical question.
It is up to the target BLE server as to what kinds of data it allows.  It sounds like you have a BLE server that expects 18 bytes of data.  Ok ... that's fair enough.
Now lets assume you have a string that is:

"330841000000000000000000000000000000"

That isn't 18 bytes of data ... however, if we treat that as an encoding of 18 bytes of data that are hex encoded (guess on my part) then we have

0x33 0x08 0x41 0x00 ... 0x00

which probably is 18 bytes of data.

So your goal will be to write a function/method that takes as input a string of concatenated 2 character hex encoded values and returns an array (std::string) of byte values where each byte is composed of the decoding of 2 characters worth of hex data.

nkolban, 2018-01-04:
Closing for just now.  Reopen as needed.

closed 2018-01-04 by nkolban