nkolban/esp32-snippets #252: Problem with serving bigger files using HttpServer

squonk11, 2017-12-10:
Now I am trying to use the HttpServer class. Serving small files works perfectly. But as soon as I try to serve bigger files (jquery library) the HttpServer crashes; see log:

> D (11767) Socket:  - accept:  Received new client!: sockFd: 4097
D (11768) Socket: << accept: sockFd: 4097
D (11768) HttpServerTask: HttpServer listening on port 80 received a new client
connection; sockFd=4097
D (11778) HttpParser: >> parseRequestLine: "GET /index.shtml HTTP/1.1" [25]
D (11783) HttpParser: << parseRequestLine: method: GET, url: /index.shtml, versi
on: HTTP/1.1
D (11810) HttpRequest: Method: GET, URL: "/index.shtml", Version: HTTP/1.1
D (11811) HttpRequest: name="Accept", value="text/html,application/xhtml+xml,app
lication/xml;q=0.9,*/*;q=0.8"
D (11816) HttpRequest: name="Accept-Encoding", value="gzipp, deflate"
D (11823) HttpRequest: name="Accept-Language", value="de,en-US;q=0.7,en;q=0.3"
D (11830) HttpRequest: name="Connection", value="keep-alive"
D (11836) HttpRequest: name="Host", value="192.168.1.8"
D (11841) HttpRequest: name="Upgrade-Insecure-Requestts", value="1"
D (11847) HttpRequest: name="User-Agent", value="Mozilla/5.0 (Windows NT 10.0; W
in64; x64; rv:57.0) Gecko/20100101 Firefox/57.0"
D (11859) HttpRequest: Body: ""
D (11862) HttpServerTask: >> processRequest: Method: GET, Path: /index.shtml
D (11869) PathHandler: plain matching: /helloWorld with /index.shtml
D (11876) PathHandler: plain matching: /http with /index.shtml
D (11881) HttpServerTask: No Path handler found
D (11889) HttpServerTask: Opening file: /sdcard/http/index.shtml
D (11899) Socket: send: Binary of length: 19
D (11899) Socket: send: Raw binary of length: 19
D (11904) Socket: send: Binary of length: 3366
D (11906) Socket: send: Raw binary of length: 3366
D (11913) Socket: close: m_sock=4097, ssl: 0
D (11915) Socket: Calling lwip_close on 4097
D (11923) HttpServerTask: Waiting for new peer client
D (11925) Socket: >> accept: Accepting on 0.0.0.0 [80]; sockFd:: 4096, using SSL
: 0
D (111960) Socket:  - accept: Received new client!: sockFd: 4098
D (11961) Socket: << accept: sockFd: 4098
D (11961) HttpServerTask: HttpServer listening on port 80 received a new client
connection; sockFd=4098
D (11971) HttpParser: >> parseRequestLine: "GET /jquery-1.11.3.min.js HTTP/1.1"
[34]
D (11977) HttpParser: << parseRequestLine: method: GET, url: /jquery-1.11.3.min.
js, version: HTTP/1.1
D (12002) HttpRequest: Method: GET, URL: "/jquery-1.11.3.min.js", Version: HTTP/
1.1
D (12002) HttpRequest: name="Accept", value="*/*"
D (12004) HttpRequest: name="Accept-Encoding", value="gzip, deflate"
D (12010) HttpRequest: name="Accept-Language", value="de,en-US;q=0.7,en;q=0.3"
D (12017)) HttpRequest: name="Connection", value="keep-alive"
D (12023) HttpRequest: name="Host", value="192.168.1.8"
D (12028) HttpRequest: name="Referer", value="http://192.168.1.8/index.shtml"
D (12036) HttpRequest: name="User-Agent", value="Mozilla/5.0 (Windows NT 10.0; W
in64; x64; rv:57.0) Gecko/20100101 Firefox/57.0"
D (12049) HttpRequest: Body: ""
D (12050) HttpServerTask: >> processRequest: Method: GET, Path: /jquery-1.11.3.m
in.js
D (12058) PathHandler: plain matching: /helloWorld with /jquery-1.11.3.min.js
D (12065) PathHandler: plain matching: /http with /jquery-1.11.3.min.js
D (12072) HttpServerTask: No Path handler found
D (12080) HttpServerTask: Opening file: /sdcard/http/jquery-1.11.3.min.js
abort() was called at PC 0x40122d1a on core 0
0x40122d1a: operator new(unsigned int) at /home/ivan/e/crosstool-NG/.build/src/g
cc-5.2.0/libstdc++-v3/libsupc++/new_op.cc:54

> Backtrace: 0x400876c8:0x3ffd4050 0x400877c7:0x3ffd4070 0x40122d1a:0x3ffd4090 0x4
0126341:0x3ffd40b0 0x4012655b:0x3ffd40d0 0x4017188d:0x3ffd4100 0x401244d6:0x3ffd
4140 0x40174c69:0x3ffd4160 0x401341ca:0x3ffd4180 0x4012460d:0x3ffd41a0 0x40124a8
d:0x3ffd41d0 0x4014216e:0x3ffd4200 0x40142825:0x3ffd4640 0x4014589c:0x3ffd57a0
0x400876c8: invoke_abort at E:/msys32/opt/esp/esp-idf/components/esp32/panic.c:5
72

> 0x400877c7: abort at E:/msys32/opt/esp/esp-idf/components/esp32/panic.c:572
> 0x40122d1a: operator new(unsigned int) at /home/ivan/e/crosstool-NG/.build/src/g
cc-5.2.0/libstdc++-v3/libsupc++/new_op.cc:54

> 0x40126341: __gnu_cxx::new_allocator<char>::allocate(unsigned int, void const*)
at /home/ivan/e/crosstool-NG/.build/xtensa-esp32-elf/build/build-cc-gcc-final/xt
ensa-esp32-elf/libstdc++-v3/include/bits/basic_string.h:2263
 (inlined by) std::allocator_traits<std::allocator<char> >::allocate(std::alloca
tor<char>&, unsigned int) at /home/ivan/e/crosstool-NG/.build/xtensa-esp32-elf/b
uild/build-cc-gcc-final/xtensa-esp32-elf/libstdc++-v3/include/bits/alloc_traits.
h:360
 (inlined by) std::__cxx11::basic_string<char, std::char_traits<char>, std::allo
cator<char> >::_M_create(unsigned int&, unsigned int) at /home/ivan/e/crosstool-
NG/.build/xtensa-esp32-elf/build/build-cc-gcc-final/xtensa-esp32-elf/libstdc++-v
3/include/bits/basic_string.tcc:157
> 0x4012655b: std::__cxx11::basic_string<char, std::char_traits<char>, std::alloca
tor<char> >::reserve(unsigned int) at /home/ivan/e/crosstool-NG/.build/xtensa-es
p32-elf/build/build-cc-gcc-final/xtensa-esp32-elf/libstdc++-v3/include/bits/basi
c_string.h:2263
> 0x4017188d: std::__cxx11::basic_stringbuf<char, std::char_traits<char>, std::all
ocator<char> >::overflow(int) at /home/ivan/e/crosstool-NG/.build/xtensa-esp32-e
lf/build/build-cc-gcc-final/xtensa-esp32-elf/libstdc++-v3/include/sstream:180
> 0x401244d6: std::basic_streambuf<char, std::char_traits<char> >::xsputn(char con
st*, int) at /home/ivan/e/crosstool-NG/.build/xtensa-esp32-elf/build/build-cc-gc
c-final/xtensa-esp32-elf/libstdc++-v3/include/streambuf:436
> 0x40174c69: std::basic_streambuf<char, std::char_traits<char> >::sputn(char cons
t*, int) at /home/ivan/e/crosstool-NG/.build/xtensa-esp32-elf/build/build-cc-gcc
-final/xtensa-esp32-elf/libstdc++-v3/include/streambuf:436
> 0x401341ca: int std::__copy_streambufs_eof<char, std::char_traits<char> >(std::b
asic_streambuf<char, std::char_traits<char> >*, std::basic_streambuf<char, std::
char_traits<char> >*, bool&) at /home/ivan/e/crosstool-NG/.build/src/gcc-5.2.0/l
ibstdc++-v3/src/c++98/streambuf.cc:49
> 0x4012460d: int std::__copy_streambufs<char, std::char_traits<char> >(std::basic
_streambuf<char, std::char_traits<char> >*, std::basic_streambuf<char, std::char
_traits<char> >*) at /home/ivan/e/crosstool-NG/.build/xtensa-esp32-elf/build/bui
ld-cc-gcc-final/xtensa-esp32-elf/libstdc++-v3/include/streambuf:436
> 0x40124a8d: std::ostream::operator<<(std::basic_streambuf<char, std::char_traits
<char> >*) at /home/ivan/e/crosstool-NG/.build/xtensa-esp32-elf/build/build-cc-g
cc-final/xtensa-esp32-elf/libstdc++-v3/include/ostream:570
> 0x4014216e: HttpServerTask::processRequest(HttpRequest&) at E:/msys32/opt/devel/
HttpServer/components/cpp_utils/HttpServer.cpp:294
> 0x40142825: HttpServerTask::run(void*) at E:/msys32/opt/devel/HttpServer/compone
nts/cpp_utils/HttpServer.cpp:294
> 0x4014589c: Task::runTask(void*) at E:/msys32/opt/devel/HttpServer/components/cp
p_utils/Task.cpp:58

> Rebooting...

Any idea what is going wrong here?

Kind regards
   Lothar

chegewara, 2017-12-10:
Actually yes, i do have an idea. Because we are trying to save some heap we have setup not so much stack size in HttpServerTask class and SocketServ class:
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/HttpServer.cpp#L88
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/SockServ.cpp#L176

Also you may have to change settings in menuconfig -> wifi for TX buffer

nkolban, 2017-12-10:
This one is a good one ... ugly as sin ... here is the code that exists today:

https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/HttpServer.cpp#L166

It looks like:

```
std::stringstream ss;
ss << ifStream.rdbuf();
response.sendData(ss.str());
ifStream.close();
```

If I'm reading it correctly, we read the file from the file system into RAM and then send it to the browser.  But wait ... what if the file is bigger than available RAM?  We fail in the way you describe.  What I need to do is chunk up the file and send it piece by piece so that at any one time we only have part of the file in RAM before it is transmitted to the partner.

nkolban, 2017-12-10:
Attempted to create myself a baseline of the problem so placed a 200K on flash and tried to serve it up.  Darn if it didn't work (it shouldn't have).  Then I realized I am using a WROVER board with 4.5MBytes of RAM so switched that off to become a normal ESP32 and recreated the problem as described.  I now have a base line to test again.

To be continued ...

nkolban, 2017-12-10:
Changes have now been made to the logic.  We now send in 4K chunks from the server to the browser.  The size of the chunk can be changed using `HTTPServer#setFileBufferSize(size_t)` if needed.

When you get a chance, pull down the latest code and see if we have a resolution.

squonk11, 2017-12-10:
What do I have to change in menuconfig? Type of TX buffers? or Max. number of TX buffers. And which value?

squonk11, 2017-12-10:
Now it works! Thany you very much for your precious support.

squonk11, 2017-12-10:
one more thing: I am setting the send buffer size to 2048 like this:
`	HttpServer* pHttpServer = new HttpServer();
	pHttpServer->setFileBufferSize(2048);
	pHttpServer->setRootPath("/sdcard/http");
`
The log shows that still:
> D (10633) HttpResponse: >> sendData: 0x3ffd82f8, size: 1000
the buffer size is 1000? Am I missing something?

nkolban, 2017-12-10:
<sigh> ... I messed up again.  While testing out the fixes, I had hard-coded 1000 bytes as the buffer size.  Found that all was good but then realized that I had to expose a mechanism to make it configurable.  At that point I added setFileBufferSize() and getFileBufferSize() and then completly failed to use them in the logic that reads file data into the buffer ... as such, the buffer size in the code you were using was 1000 bytes and ignored the file buffer size attribute.  I have just uploaded a new fix and tested and seen with my own eyes that the segmented data is as expected.  Apologies again my friend.

squonk11, 2017-12-11:
No reason to apologize - I have to thank you.
I tested again and now it works perfectly.
But there are three more issues: 

1. when I open a directory in the browser (e.g. "http://192.168.1.8/FW7") I can see the directory contents because I set:
`pHttpServer->setDirectoryListing(true);` 
When I click on a link in this directory (e.g. "index.html") then this file is not opened because the browser attepts to open: "http://192.168.1.8/sdcard/http/FW7/index.html". So, it inserts "/sdcard/http" which I previously set as root path using:
`pHttpServer->setRootPath("/sdcard/http");`
2. when opening/requesting a  directory there are several error messages in the log:
> E (50827) HttpResponse: << sendData

3. Sometimes if I open several directories or pages concecutively after some time the HttpServer does not respond any more - until now I was not able to reproduce this.

Do you have fixes for that?

Kind regards

nkolban, 2017-12-11:
You are NOT being annoying at all.  I actually enjoy trying to solve these puzzles.  However, let me ask a favor.  Can you create a separate issue for tracking each distinct puzzle?  This issue talks about serving large files.  I track each puzzle in its own issue.

squonk11, 2017-12-11:
thank you for opening these issues for me - it should have been my task... 
On the third topic I need to do some more investigation in order to be able to reproduce it. Then I will open a new issue if necessary.

squonk11, 2017-12-16:
It is working perfect now.

closed 2017-12-16 by squonk11