nkolban/esp32-snippets #253: #include "sdkconfig.h" - No such directory

RobertYuwono, 2017-12-10:
C:\Users\Documents\Arduino\libraries\ESP32_BLE_Arduino\src/BLEDevice.h:10:23: fatal error: sdkconfig.h: No such file or directory

 #include "sdkconfig.h"

                       ^

compilation terminated.

exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).

I got this error, do you have any idea how I can solve it?
Thanks,
Robert

nkolban, 2017-12-10:
Howdy Robert  .... I'm probably going to need some more context.  What version of Arduino IDE are you using?  When was the last time you refreshed the ESP32-Arduino install?  Are you running a custom build of the BLE C++ classes?  (the latest versions of Arduino-ESP32 contain the BLE C++ classes meaning that you should delete private copies).

chegewara, 2017-12-10:
But why do you trying to compile this library with esp8266? It wont work.

RobertYuwono, 2017-12-10:
I am new to this programming world.
I am using  Arduino IDE 1.8.5 and I am using NodeMCU ( new, so never been refreshed).
I am just adding the library directly from the IDE and try to run the scan example.

The target of this:
- try to read estimote beacon using nodeMCU connected to bluetooth module (HC-05).

chegewara, 2017-12-10:
Most important question is: do you have NodeMCU-32S or NodeMCU 1.0? From logs i can read you have second one or at least this one you are trying to flash.
`Error compiling for board NodeMCU 1.0 (ESP-12E Module).`

nkolban, 2017-12-31:
Closing for just now.  Re-open as needed.

closed 2017-12-31 by nkolban

buteomont, 2019-02-09:
I know that this is an old thread, but I'm getting the exact same error and pulling out my hair.  I updated to the latest IDE (1.8.8) and the latest ESP32_BLE_Arduino (1.0.1) but neither helped. Would love a pointer to a fix.

chegewara, 2019-02-09:
Check if you have this file:
`C:\Users\user\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.1\tools\sdk`

buteomont, 2019-02-11:
Thanks.  In case anyone else is having this problem, there is apparently a conflict between the _expressif ESP32_ board libraries and the _ESP8266 community_ ones.  Removing the latter fixed the problem.  I haven't dug into it deep enough to be more specific.

Credesence, 2020-07-31:
this error is due to wrong board selection, change it to TinyPICO. problem solved for me

CodeWreckPro, 2021-09-20:
I tried this but couldn't fix it
i'm getting the error
C:\Users\Rakshiiii\Documents\Arduino\libraries\AsyncTCP-master\src/AsyncTCP.h:26:10: fatal error: sdkconfig.h: No such file or directory
26 | #include "sdkconfig.h"
I installed the library from git repository of me-no-dev.
could you please help me out.

Some insights from you @me-no-dev  would also be really great.

me-no-dev, 2021-09-20:
you building it for ESP32? because that file do exist for all ESP32 based boards

pvelesko, 2023-02-20:
For others: I was trying to program ESP8266 whereas my intention was ESP32

zagoorland, 2024-04-14:
I also have this error and I'm 100% sure I selected correct board.

C-PLUS-PLUSandPYTHON, 2024-11-25:
Hi, got the same error:

> [!WARNING]
> In file included from C:\Users\docsc\OneDrive\Niko\Elektronik\Arduino\WLAN_BLUETOOTH\BLEmouse_keyboard\BLEmouse_keyboard.ino:1:0:
c:\Users\docsc\Documents\Arduino\libraries\ESP32_BLE_Keyboard/BleKeyboard.h:3:10: fatal error: sdkconfig.h: No such file or directory
 #include "sdkconfig.h"
          ^~~~~~~~~~~~~
compilation terminated.

I tried to make a ble keyboard with esp32.
Board Selected: NodeMCU 1.0 (ESP-12E Module)

Code: 
```c++
#include <BleKeyboard.h>

// First param is name
// Second is manufacturer
// Third is initial batter level
BleKeyboard bleKeyboard("BLE_ESP32", "Arduino", 100);

void setup() {
  Serial.begin(9600);
  Serial.println("Starting, Connecting to Device");
  bleKeyboard.begin();
  while (!bleKeyboard.isConnected()) {
    Serial.print(".");
  }
}

void loop() {
  if (bleKeyboard.isConnected()) {
    Serial.println("Sending Hello World");
    bleKeyboard.println("Hello World!");
    delay(100);
  }
}
```

Things I've already tried:
-use other/example sketches
-reinstall ArduinoBle lib now Version 1.3.7
   and ESP32-BLE-Keyboard now v0.3.0

Thanks in advance!

chegewara, 2024-11-26:
> ESP-12E Module

Are you sure this is esp32? As far as i know it is esp8266 module

And your issue about missing file, im can you show screenshoot how did you select module from menu?

pvelesko, 2024-11-26:
> Are you sure this is esp32? As far as i know it is esp8266 module

this was the issue for me

C-PLUS-PLUSandPYTHON, 2024-11-27:
How Canon I fix it? I Downloaded the lib from the official repo