nkolban/esp32-snippets #254: ESP32_BLE_Arduino

AlanLL, 2017-12-10:
The example file "BLE_Notify" does not compile.
The error I get is:-
--------------------------------------------------------------------------------------
Arduino: 1.8.3 (Windows 10), Board: "ESP32 Dev Module, 80MHz, 115200, None"

In file included from C:\Users\Alan\Documents\Arduino\libraries\ESP32_BLE_Arduino-master\src/BLEDescriptor.h:16:0,

                 from C:\Users\Alan\Documents\Arduino\libraries\ESP32_BLE_Arduino-master\src/BLECharacteristic.h:16,

                 from C:\Users\Alan\Documents\Arduino\libraries\ESP32_BLE_Arduino-master\src/BLEServer.h:19,

                 from C:\Users\Alan\Documents\Arduino\libraries\ESP32_BLE_Arduino-master\src/BLEDevice.h:18,

                 from C:\Users\Alan\Documents\Arduino\libraries\ESP32_BLE_Arduino-master\examples\BLE_notify\BLE_notify.ino:21:

C:\Users\Alan\Documents\Arduino\libraries\ESP32_BLE_Arduino-master\src/FreeRTOS.h:46:3: error: 'pthread_mutex_t' does not name a type

   pthread_mutex_t   m_pthread_mutex;

   ^

In file included from C:\Users\Alan\Documents\Arduino\libraries\ESP32_BLE_Arduino-master\src/BLEDevice.h:20:0,

                 from C:\Users\Alan\Documents\Arduino\libraries\ESP32_BLE_Arduino-master\examples\BLE_notify\BLE_notify.ino:21:

C:\Users\Alan\Documents\Arduino\libraries\ESP32_BLE_Arduino-master\src/BLEUtils.h:50:49: error: 'esp_gattc_service_elem_t' has not been declared

  static std::string gattcServiceElementToString(esp_gattc_service_elem_t *pGATTCServiceElement);

                                                 ^

In file included from C:\Users\Alan\Documents\Arduino\libraries\ESP32_BLE_Arduino-master\examples\BLE_notify\BLE_notify.ino:21:0:

C:\Users\Alan\Documents\Arduino\libraries\ESP32_BLE_Arduino-master\src/BLEDevice.h:57:23: error: 'esp_power_level_t' has not been declared

  static void setPower(esp_power_level_t powerLevel);

                       ^

exit status 1
Error compiling for board ESP32 Dev Module.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
------------------------------------------------------
I have Deleted the Library file and Downloaded the ESP32_BLE_Arduino-master.zip twice and I still get the same error.

Please Help
AlanLL

chegewara, 2017-12-10:
Did you try this to instal BLE_Arduino?
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/ArduinoBLE.md

AlanLL, 2017-12-10:
I have removed the "Master" and installed esp32 ble arduino via the library, restarted the IDE and now get thie:
----------------------------------------------------------
Arduino: 1.8.3 (Windows 10), Board: "ESP32 Dev Module, 80MHz, 115200, None"

C:\Users\Alan\Documents\Arduino\libraries\ESP32_BLE_Arduino\src\BLEUtils.cpp: In static member function 'static std::__cxx11::string BLEUtils::gattClientEventTypeToString(esp_gattc_cb_event_t)':

C:\Users\Alan\Documents\Arduino\libraries\ESP32_BLE_Arduino\src\BLEUtils.cpp:344:8: error: 'ESP_GATTC_CONNECT_EVT' was not declared in this scope

   case ESP_GATTC_CONNECT_EVT:

        ^

C:\Users\Alan\Documents\Arduino\libraries\ESP32_BLE_Arduino\src\BLEUtils.cpp:346:8: error: 'ESP_GATTC_DISCONNECT_EVT' was not declared in this scope

   case ESP_GATTC_DISCONNECT_EVT:

        ^

C:\Users\Alan\Documents\Arduino\libraries\ESP32_BLE_Arduino\src\BLEUtils.cpp: In static member function 'static void BLEUtils::dumpGattClientEvent(esp_gattc_cb_event_t, esp_gatt_if_t, esp_ble_gattc_cb_param_t*)':

C:\Users\Alan\Documents\Arduino\libraries\ESP32_BLE_Arduino\src\BLEUtils.cpp:684:8: error: 'ESP_GATTC_CONNECT_EVT' was not declared in this scope

   case ESP_GATTC_CONNECT_EVT: {

        ^

In file included from C:\Users\Alan\Documents\Arduino\libraries\ESP32_BLE_Arduino\src\BLEUtils.cpp:22:0:

C:\Users\Alan\Documents\Arduino\libraries\ESP32_BLE_Arduino\src\BLEUtils.cpp:686:44: error: 'union esp_ble_gattc_cb_param_t' has no member named 'connect'

     BLEUtils::gattStatusToString(evtParam->connect.status).c_str(),

                                            ^

C:\Users\Alan\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/log/esp_log.h:144:168: note: in definition of macro 'ESP_LOGD'

 #define ESP_LOGD( tag, format, ... )  if (LOG_LOCAL_LEVEL >= ESP_LOG_DEBUG)   { esp_log_write(ESP_LOG_DEBUG,   tag, LOG_FORMAT(D, format), esp_log_timestamp(), tag, ##__VA_ARGS__); }

                                                                                                                                                                        ^

C:\Users\Alan\Documents\Arduino\libraries\ESP32_BLE_Arduino\src\BLEUtils.cpp:687:15: error: 'union esp_ble_gattc_cb_param_t' has no member named 'connect'

     evtParam->connect.conn_id,

               ^

C:\Users\Alan\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/log/esp_log.h:144:168: note: in definition of macro 'ESP_LOGD'

 #define ESP_LOGD( tag, format, ... )  if (LOG_LOCAL_LEVEL >= ESP_LOG_DEBUG)   { esp_log_write(ESP_LOG_DEBUG,   tag, LOG_FORMAT(D, format), esp_log_timestamp(), tag, ##__VA_ARGS__); }

                                                                                                                                                                        ^

C:\Users\Alan\Documents\Arduino\libraries\ESP32_BLE_Arduino\src\BLEUtils.cpp:688:26: error: 'union esp_ble_gattc_cb_param_t' has no member named 'connect'

     BLEAddress(evtParam->connect.remote_bda).toString().c_str()

                          ^

C:\Users\Alan\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/log/esp_log.h:144:168: note: in definition of macro 'ESP_LOGD'

 #define ESP_LOGD( tag, format, ... )  if (LOG_LOCAL_LEVEL >= ESP_LOG_DEBUG)   { esp_log_write(ESP_LOG_DEBUG,   tag, LOG_FORMAT(D, format), esp_log_timestamp(), tag, ##__VA_ARGS__); }

                                                                                                                                                                        ^

C:\Users\Alan\Documents\Arduino\libraries\ESP32_BLE_Arduino\src\BLEUtils.cpp:700:8: error: 'ESP_GATTC_DISCONNECT_EVT' was not declared in this scope

   case ESP_GATTC_DISCONNECT_EVT: {

        ^

[... 19 lines cut ...]

C:\Users\Alan\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/log/esp_log.h:144:168: note: in definition of macro 'ESP_LOGD'

 #define ESP_LOGD( tag, format, ... )  if (LOG_LOCAL_LEVEL >= ESP_LOG_DEBUG)   { esp_log_write(ESP_LOG_DEBUG,   tag, LOG_FORMAT(D, format), esp_log_timestamp(), tag, ##__VA_ARGS__); }

                                                                                                                                                                        ^

C:\Users\Alan\Documents\Arduino\libraries\ESP32_BLE_Arduino\src\BLEUtils.cpp:704:26: error: 'union esp_ble_gattc_cb_param_t' has no member named 'disconnect'

     BLEAddress(evtParam->disconnect.remote_bda).toString().c_str()

                          ^

C:\Users\Alan\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/log/esp_log.h:144:168: note: in definition of macro 'ESP_LOGD'

 #define ESP_LOGD( tag, format, ... )  if (LOG_LOCAL_LEVEL >= ESP_LOG_DEBUG)   { esp_log_write(ESP_LOG_DEBUG,   tag, LOG_FORMAT(D, format), esp_log_timestamp(), tag, ##__VA_ARGS__); }

                                                                                                                                                                        ^

exit status 1
Error compiling for board ESP32 Dev Module.

--------------------------------------------

Am I doing something wrong?

chegewara, 2017-12-10:
Can you pastebin this file:
C:\Users\Alan\Documents\Arduino\hardware\espressif\esp32\package\package_esp32_index.template.json

How did you install esp32 ble library? From zip file or from library manager?

AlanLL, 2017-12-10:
Here is the requested file:
{
  "packages": [
    {
      "name": "esp32",
      "maintainer": "Espressif Systems",
      "websiteURL": "https://github.com/espressif/arduino-esp32",
      "email": "hristo@espressif.com",
      "help": {
        "online": "http://esp32.com"
      },
      "platforms": [
        {
          "name": "esp32",
          "architecture": "esp32",
          "version": "",
          "category": "ESP32",
          "url": "",
          "archiveFileName": "",
          "checksum": "",
          "size": "",
          "help": {
            "online": ""
          },
          "boards": [
            {
              "name": "ESP32 Dev Module"
            },
            {
              "name": "WEMOS LoLin32"
            }
          ],
          "toolsDependencies": [
            {
              "packager": "esp32",
              "name": "xtensa-esp32-elf-gcc",
              "version": "1.22.0-61-gab8375a-5.2.0"
            },
            {
              "packager": "esp32",
              "name": "esptool",
              "version": "96698a3"
            }
          ]
        }
      ],
      "tools": [
        {
          "name": "xtensa-esp32-elf-gcc",
          "version": "1.22.0-61-gab8375a-5.2.0",
          "systems": [
            {
              "host": "i686-mingw32",
              "url": "https://dl.espressif.com/dl/xtensa-esp32-elf-win32-1.22.0-61-gab8375a-5.2.0-2.zip",
              "archiveFileName": "xtensa-esp32-elf-win32-1.22.0-61-gab8375a-5.2.0-2.zip",
              "checksum": "SHA-256:c00dbdab8e8acc273d4aa319a74f7bb1d9496c843159823201d685359174168f",
              "size": "76346909"
            },
            {
              "host": "x86_64-apple-darwin",
              "url": "https://dl.espressif.com/dl/xtensa-esp32-elf-osx-1.22.0-61-gab8375a-5.2.0.tar.gz",
              "archiveFileName": "xtensa-esp32-elf-osx-1.22.0-61-gab8375a-5.2.0.tar.gz",
              "checksum": "SHA-256:89a9a6084ec032ddcd4bdae37b428e533ed5fd0c0ab6b66ddf703e4b3f83d6e4",
              "size": "40481812"
            },
            {
              "host": "x86_64-pc-linux-gnu",
              "url": "https://dl.espressif.com/dl/xtensa-esp32-elf-linux64-1.22.0-61-gab8375a-5.2.0.tar.gz",
              "archiveFileName": "xtensa-esp32-elf-linux64-1.22.0-61-gab8375a-5.2.0.tar.gz",
              "checksum": "SHA-256:b74333fd7f622f035fee7c4e0ca57c3e2ccb2515de5d901f33e87e79cdf0d4f9",
              "size": "37674568"
            },
            {
              "host": "i686-pc-linux-gnu",
              "url": "https://dl.espressif.com/dl/xtensa-esp32-elf-linux32-1.22.0-61-gab8375a-5.2.0.tar.gz",
              "archiveFileName": "xtensa-esp32-elf-linux32-1.22.0-61-gab8375a-5.2.0.tar.gz",
              "checksum": "SHA-256:dbb342b7c377a7c3a1fcccc60fa1193332d0c37e6a90c5075d3471c13c9e94ad",
              "size": "41032647"
            }
          ]
        },
        {
          "name": "esptool",
          "version": "96698a3",
          "systems": [
            {
              "host": "i686-mingw32",
              "url": "https://dl.espressif.com/dl/esptool-96698a3-windows.zip",
              "archiveFileName": "esptool-96698a3-windows.zip",
              "checksum": "SHA-256:12f026c84869f0bd2f6a863191ec84d27b91f21d17e2e2e9e11c79be50f2d0fc",
              "size": "3390935"
            },
            {
              "host": "x86_64-apple-darwin",
              "url": "https://dl.espressif.com/dl/esptool-96698a3-macos.tar.gz",
              "archiveFileName": "esptool-96698a3-macos.tar.gz",
              "checksum": "SHA-256:e3154dbf32c3ba5e5ebd45e913178b167cf761e5ce0152b2bb069a2453f6839b",
              "size": "3805789"
            }
          ]
        }
      ]
    }
  ]
}

chegewara, 2017-12-10:
You have installed old version of esp32-arduino. Its version 2 or 2.1. My suggestion is to delete it and install new version:
https://github.com/espressif/arduino-esp32/blob/master/docs/arduino-ide/windows.md

AlanLL, 2017-12-11:
I have removed ESP32_BLE_Arduino and done the above.
Everything now OK.
Thanks very much for your assistance.

closed 2017-12-11 by AlanLL