nkolban/esp32-snippets #257: HttpServer with directory listing shows wrong data when setRootPath is used

nkolban, 2017-12-11:
Imagine we have files on the local ESP32 file system at:
```
/spiflash/dir1/x.txt
```

If we call `HttpServer#setRootPath("/spiflash")` and then navigate to:

```
http://server:port/dir1
```

we are shown the directory listing which includes `x.txt`.  However the link to `x.txt` is now the file system path:
```
/spiflash/dir1/x.txt
```

when it should be the normalized path of `/dir1/x.txt`.

See also:
#252

chegewara, 2017-12-11:
It deppends what you are trying to achieve. If you want to download file then it requires some fix. I think i had to make some workaround, but if you want to display file contents in web browser then it require request path

squonk11, 2017-12-11:
But I thnik it normally should be possible to navigate in the directory listing by clicking on the links...

nkolban, 2017-12-11:
Here is the first pass at a resolution.  The file entries now shown in a directory listing are URL links to the files and no longer file system paths.  This will become apparent only when we have set a `setRootPath()`.

chegewara, 2017-12-11:
To navigate betwen directories it is possible, but like i said, if you want to open, let say, index.html file this way, then you need to add path handler. What youcan do is to create path handler with regex to all html files etc which will be sent to the browser.

squonk11, 2017-12-16:
It seems to be working now!

nkolban, 2017-12-16:
Great news my friend.  I'll close the issue for just now but if anything comes up, post back and we'll either re-open it or create a new one.  Many thanks for the feedback.

closed 2017-12-16 by nkolban