nkolban/esp32-snippets #258: BLE Observer and Client in the same time

tendriu, 2017-12-11:
Is it possible to maintain a connection with the service (central role) and receive advertising packets at the same time?

Regards,
Andrew

nkolban, 2017-12-11:
Its a good question.  The BLE C++ classes found in this repository basically act as wrappers for the underlying ESP-IDF implementation of BLE which is implemented on top of Bluedroid which ideally implements the BLE specification.  As such we have a whole multitude of layers, levels and abstractions.

BLE C++ -> ESP-IDF -> Bluedroid -> BLE Specification

In thinking about your question, we would ideally work bottom up.  The question would thus become ... "Is it a valid thing to ask that a BLE Central (client) can be connected to more than server at the same time?".   If yes, then we should be able to listen for new advertising packets while, at the same time, have a connection to a partner.    However it all starts with asking at the BLE specification level ... "Is this allowed?"

chegewara, 2017-12-11:
Short answer, yes, it is possible. Easiest way to check it is to run Sample example (client or server, whatever you need) and add scan function at the same time.