nkolban/esp32-snippets #259: UART example do not compile.

pbecchi, 2017-12-11:
Trying to compile UART example I get this error:

BLE_uart.ino: 107:24: error: 'class BLECharacteristic' has no member named 'isReadyForData
   if (pTxCharacteristic->isReadyForData())

What can I do?

nkolban, 2017-12-11:
Howdy @pbecchi 

This logic was written by Mr @Sepp62 ... paging him for a consult.  We are discussing the story in #209 which is where this logic was introduced.  Reading/reviewing the code that was added, I'm afraid I'm not yet getting it.  That doesn't mean it is wrong/bad/broken ... just that I don't yet understand it.  See the latest posts in #209.  We'll get clarification soon.

chegewara, 2017-12-11:
Strange, it is working for me with BLE_notify example (with issues complaining about c++ exceptions of course).

pbecchi, 2017-12-11:
Oh ...sorry...compiled is using the wrong library ....on esp32-arduino there is a duplicate of BLEcharacteristic.h.

nkolban, 2018-01-04:
Closing for just now.  Re-open as needed.

closed 2018-01-04 by nkolban