nkolban/esp32-snippets #260: BLE protected caratheristics

Gior80, 2017-12-11:
hi, 
i 'm  using with sucess your API to control some relays over esp32 ble and arduino,  it is working so great...
I would now need to protect some of those caratheristics in some way:

example:
- BTrelay open/close caratheristics would be open by anyone
- Name of device caratheristics:  could be changed only if a pin is entered 

could you suggest me best approach (is possible)  to do that?

thanks 

giorgio

nkolban, 2017-12-12:
Howdy Giorgio,
I have good news and bad news.  The good news is that what you are asking for is do-able and is on our list of things to do.  See: #69

As you see, we have been hoping for this since September (it is now nearly 3 months later).

The bad news, we are still not much closer.  Our problem isn't a desire to achieve the task ... its the inability to pin down a test framework where we say "Here is a protected BLE resource ... see we can't access it from ESP32 ... now ... what do we have to do to access it?".

If you have notions of how to create a test environment for BLE security ... we are all ears.

Gior80, 2017-12-12:
Just an idea , non technical and probably just it ...

Resource need to be pin protected . Playing with your class I was creating this: 

-a crypted aes pin stored pin  in nvs
- a bool variable set by an event to on a writable caratteristic (pin)
- on each things we want to protect we simply check bool value true (pin is correct?) and jump method if it's not 

In this way we would simply stop / jump method if pin is not correct . In my case this would work as if the my ios app is not passing the right pin related method are disabled . In Same way (if true /else) I can choose to notify app that pin is not right

Please consider it as a simple idea, schema  as i really haven't go deeply in technical and I'm not an experienced programmer

chegewara, 2017-12-20:
@Gior80 Have you seen last update about BLE security? It introduces protecting characteristics by adding connection authentication.

closed 2018-11-13 by chegewara