nkolban/esp32-snippets #261: BLECharacteristicCallbacks not called on write

rweller59, 2017-12-11:
ESP32 has ble server defined. 1 characteristic with callback. Using nRF Connect as client for testing.

ESP updates characteristic value and sends notification which is received and processed correctly at nRF. nRF reads characteristic value and ESP32 characteristic callback is triggered. nRF writes to value to characteristic and ESP32 callback is NOT triggered.

Characteristic is defined as:
	dataCharacteristic = pService->createCharacteristic(
		BLEUUID(DATA_CHARACTERISTIC_UUID),
		BLECharacteristic::PROPERTY_READ   |
		BLECharacteristic::PROPERTY_WRITE  |
		BLECharacteristic::PROPERTY_NOTIFY |
		BLECharacteristic::PROPERTY_INDICATE
	);
	dataCharacteristic->addDescriptor(new BLE2902());
	dataCharacteristic->setCallbacks(new bleCharacteristicCallback());

chegewara, 2017-12-11:
Can you paste code you are using, because with BLE_write example all works good:
```
D][BLEUtils.cpp:1645] dumpGattServerEvent(): GATT ServerEvent: ESP_GATTS_WRITE_EVT

[D][BLEUtils.cpp:1830] dumpGattServerEvent(): [conn_id: 0, trans_id: 2, bda: 78:1f:db:c0:c7:f3, handle: 0x2a, offset: 0, need_rsp: 1, is_prep: 0, len: 1]

[D][BLEUtils.cpp:1832] dumpGattServerEvent(): [Data: 68]

[D][BLEServer.cpp:175] handleGATTServerEvent(): >> handleGATTServerEvent: ESP_GATTS_WRITE_EVT

[D][BLECharacteristic.cpp:644] setValue(): >> setValue: length=1, data=68, characteristic UUID=beb5483e-36e1-4688-b7f5-ea07361b26a8

[D][BLECharacteristic.cpp:651] setValue(): << setValue

[D][BLECharacteristic.cpp:281] handleGATTServerEvent():  - Response to write event: New value: handle: 2a, uuid: beb5483e-36e1-4688-b7f5-ea07361b26a8

[D][BLECharacteristic.cpp:284] handleGATTServerEvent():  - Data: length: 1, data: 68

*********

New value: h

*********

[D][BLEServer.cpp:293] handleGATTServerEvent(): << handleGATTServerEvent

[D][BLEDevice.cpp:83] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 3] ... ESP_GATTS_RESPONSE_EVT

[D][BLEUtils.cpp:1645] dumpGattServerEvent(): GATT ServerEvent: ESP_GATTS_RESPONSE_EVT

```

rweller59, 2017-12-11:
had to cut and paste since its buried in a bunch of code. Kind of hard to see but the first characteristic is commented out (things don't seem to work when I add another characteristic). With nRF, I'm just writing a single character

// Custom UUIDs: 
#define SERVICE_UUID 				"bbfd0423-0869-4a22-b734-ddac4cc3ffab"
#define CONFIG_CHARACTERISTIC_UUID 	"ac65ebd8-72ec-4fd2-a964-b12dda6956e1"
#define DATA_CHARACTERISTIC_UUID	"97d085cb-84af-4f8e-80c3-2ca99089a4fa"
// Characteristics and descriptors

BLECharacteristic *configCharacteristic;
BLECharacteristic *dataCharacteristic;

// process BLE connections
class bleServerCallback: public BLEServerCallbacks {
	void onConnect(BLEServer* pServer) {
		ESP_LOGI(LOG, "Bluetooth client connected");
		ble_connected = true;
		}
	void onDisconnect(BLEServer* pServer) {
		ESP_LOGI(LOG, "Bluetooth client disconnected");
		ble_connected = false;
		pServer->getAdvertising()->start();
		}
};

// process external characteristic calls
class bleCharacteristicCallback: public BLECharacteristicCallbacks {
	void onRead(BLECharacteristic* pCharacteristic) {
		ESP_LOGI(LOG, "client read characteristic");
		}
	void onWrite(BLEServer* pServer) {
		ESP_LOGI(LOG, "client wrote characteristic");
		}
};

// create BLE server, services and characteristics
void init(void) {
	BLEDevice::init("ESP32");
	BLEServer *pServer = BLEDevice::createServer();
	pServer->setCallbacks(new bleServerCallback());
	BLEService *pService = pServer->createService(BLEUUID(SERVICE_UUID),8);
/*
	// characteristic for configuration with callback defined
	configCharacteristic = pService->createCharacteristic(
		BLEUUID(CONFIG_CHARACTERISTIC_UUID),
		BLECharacteristic::PROPERTY_READ   |
		BLECharacteristic::PROPERTY_WRITE  |
		BLECharacteristic::PROPERTY_NOTIFY |
		BLECharacteristic::PROPERTY_INDICATE
	);
	configCharacteristic->addDescriptor(new BLE2902());
	configCharacteristic->setCallbacks(new bleCharacteristicCallback());
*/

	dataCharacteristic = pService->createCharacteristic(
		BLEUUID(DATA_CHARACTERISTIC_UUID),
		BLECharacteristic::PROPERTY_READ   |
		BLECharacteristic::PROPERTY_WRITE  |
		BLECharacteristic::PROPERTY_NOTIFY |
		BLECharacteristic::PROPERTY_INDICATE
	);
	dataCharacteristic->addDescriptor(new BLE2902());
	dataCharacteristic->setCallbacks(new bleCharacteristicCallback());

	pService->start();
	pServer->getAdvertising()->start();
}

chegewara, 2017-12-11:
Just to be sure, it is about arduino ide?

chegewara, 2017-12-11:
Its long shot, but try to change this line by seting up more handlers:
`BLEService pService = pServer->createService(BLEUUID(SERVICE_UUID),8);`
Let say 20, just for debug purpose.

rweller59, 2017-12-11:
no, using espidf, eclipse c++. changed handlers to 20. Didn't get the callback.
I did notice that the value shown on nRF changed - don't know if that means it read it back or that it updated locally

chegewara, 2017-12-11:
You can re-read it and then you will find out if it is changed in esp32 or just in nRF, but im sure its changed in esp32.

rweller59, 2017-12-11:
yep, just tried that. the value in the esp32 did get changed

chegewara, 2017-12-11:
Then looks like its something with logging level. You have set `ESP_LOGI` (info), maybe change it to debug or just for try to error log `ESP_LOGE`

rweller59, 2017-12-11:
You don't mean the log statement in the callback do you? Added another log stmt just to make sure there isn't a hidden typo but its not getting called. Or did you mean something else?

chegewara, 2017-12-11:
Sorry. Yes, log in callback:

```
class bleCharacteristicCallback: public BLECharacteristicCallbacks {
void onRead(BLECharacteristic* pCharacteristic) {
ESP_LOGI(LOG, "client read characteristic");   <---- this
}
void onWrite(BLEServer* pServer) {
ESP_LOGI(LOG, "client wrote characteristic");  <----this
}
};
```

rweller59, 2017-12-11:
tried that, no difference. trying to figure out how to turn on the debug messages for cpp_utils. looks like there's a check to see if its the correct handle before triggering the onWrite callback in BLECharacteristic.cpp

chegewara, 2017-12-11:
Can you confirm that, i didnt see this before:

```
class bleCharacteristicCallback: public BLECharacteristicCallbacks {
void onRead(BLECharacteristic* pCharacteristic) {
ESP_LOGI(LOG, "client read characteristic");  
}
void onWrite(BLEServer* pServer) {     <------ THIS PARAMETER IS WRONG, SHOULD BE (BLECharacteristic* pCharacteristic)
ESP_LOGI(LOG, "client wrote characteristic");  
}
};
```

rweller59, 2017-12-11:
I've looked at that a hundred times - thanks for the eyeballs. And yes, fixing that fixed the issue and I've now got the callback working.

I noticed your input on the issue of multiple characteristics with one service. Do you know if that's been addressed and code updated? Since its not working for me.  If not, I'll go to that issue# and add something

chegewara, 2017-12-11:
It was addressed and should have working. If its not, then please give us feedback in proper issue

closed 2017-12-11 by rweller59