nkolban/esp32-snippets #263: Validate that we don't exceed handles ...

nkolban, 2017-12-12:
When we create a BLE Service, we must define the maximum number of handles associated with the service.  The default is 15.  As we add characteristics to the service and then descriptors to the characteristics (of that service), we start to consume the available handles.  Since we KNOW how many handles we have allocated to the service and we know how many characteristics and descriptors we have previously added, we should be able to "trap" adding too many characteristics and descriptors to any given service and alert the programmer to increase the number of handles they should allocate.

BobaTrek, 2017-12-14:
I have run into the max number of handles limit as well, and have been unable to increase it.

Observations:

If I keep my handle requirement to under 15, everything works fine.

If I go over 15 handles (but leave the default setting to 15), then as expected, everything still works, except all the BLECharacteristics and BLEDescriptors I try to add past 15 do not get added or appear in nRF Connect.

If I try to increase the Max numHandles to more than 15, then the BLECharacteristics and BLEDescriptors now add successfully in the code, but in nRF Connect, my BLEDevice still shows up, but the BLEService and all of it's BLECharacteristics no longer shows up.

The weirdest part is that if I keep my numHandles requirement to less than 15, and call BLEServer->createService() with a BLEUUID instead of a uuid, then nRF Connect no longer sees my BLEService as well.

I have tried the following ways to increase the numHandles to more than 15:

	//mpBLEService = mpBLEServer->createService( "91bad492-b950-4226-aa2b-4ede9fa42f59" );  // Sees Services, but I need more handles
	//mpBLEService = mpBLEServer->createService( BLEUUID("91bad492-b950-4226-aa2b-4ede9fa42f59") );  // Fails to see Services
	//mpBLEService = mpBLEServer->createService( BLEUUID("91bad492-b950-4226-aa2b-4ede9fa42f59"), 10 );  // Fails to see Services
	//mpBLEService = mpBLEServer->createService( BLEUUID("91bad492-b950-4226-aa2b-4ede9fa42f59"), 15 );  // Fails to see Services

I also increased the default numHandles in BLEServer.h and BLEService.h so that I could continue to call the createService passing a uuid instead of a BLEUUID, but even just increasing the default numHandles in the constructor and createService calls makes nRF Connect no longer see the BLEService.

Side note about the default numHandles:

In the code the max number of BLEService handles is 10 in one place, and 15 in another.

In BLEServer we have a default of 15:
	BLEService*     createService(BLEUUID uuid, uint32_t numHandles=15);
[https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/BLEServer.h#L56](url)

In BLEService we have a default of 10:
	BLEService(const char* uuid, uint32_t numHandles=10);
	BLEService(BLEUUID uuid, uint32_t numHandles=10);
[https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/BLEService.h#L55](url)

chegewara, 2017-12-14:
If you are working with esp-idf environment then you can read this to fix issue with creating service:
```
//mpBLEService = mpBLEServer->createService( "91bad492-b950-4226-aa2b-4ede9fa42f59" );  // Sees Services, but I need more handles
//mpBLEService = mpBLEServer->createService( BLEUUID("91bad492-b950-4226-aa2b-4ede9fa42f59") );  // Fails to see Services
//mpBLEService = mpBLEServer->createService( BLEUUID("91bad492-b950-4226-aa2b-4ede9fa42f59"), 10 );  // Fails to see Services
//mpBLEService = mpBLEServer->createService( BLEUUID("91bad492-b950-4226-aa2b-4ede9fa42f59"), 15 );  // Fails to see Services
```
https://github.com/nkolban/esp32-snippets/issues/264#issuecomment-350953515

BobaTrek, 2017-12-14:
Yup - same problem - Thanks!

For now changing the default numHandles in BLEService.h from 10 to 30, and in BLEServer.h from 15 to 30 allows me to bypass the BLEUUID issue.

Since the default numHandles is in two places, I would like to suggest using a #define instead.

Thanks for your incredibly quick response!

chegewara, 2017-12-14:
As i mentioned, if you are working with esp-idf you can change settings in menuconfig and create service with handle number:
`mpBLEService = mpBLEServer->createService( BLEUUID("91bad492-b950-4226-aa2b-4ede9fa42f59"), 15 );`

nkolban, 2017-12-15:
Changed the logic to make BLEService a private constructor ... this means that we can't perform a new BLEService() outside of the BLE classes themselves.  This makes sense since the only way we should create a service is by asking the server to do it for us.  I also removed the default number of handles from the BLEService constructor ... so now there is nothing to get out of sync.