nkolban/esp32-snippets #264: No services exposed when testing with SampleWrite.cpp

nkolban, 2017-12-12:
One of the samples supplied with the BLE C++ classes seems to be illustrating a problem.  The issue appears to be with "SampleWrite.cpp".

```
BLEService *pService = pServer->createService(BLEUUID(SERVICE_UUID));
```

results in no services being exposed but apparently:

```
BLEService *pService = pServer->createService(SERVICE_UUID);
```

works.

chegewara, 2017-12-12:
Im not sure yet, but this may be another weird bug hard to track. Ive found that this workaround will work, but why dunno. Strange is that all lines are required, if one is missing then service is not discovered in nRF connect.

```
BLEUUID uuid = BLEUUID(SERVICE_UUID);
ESP_LOGD(LOG_TAG, "uuid %s", uuid.toString().c_str());
BLEService *pService = pServer->createService(uuid);
ESP_LOGD(LOG_TAG, "uuid %s", pService->getUUID().toString().c_str());
```

nkolban, 2017-12-12:
I'm using the out of the box SampleWrite on a WROVER module and all appears to work correctly.  Can you pastebin a copy of your SDKCONFIG?

chegewara, 2017-12-12:
https://pastebin.com/w88NWqL4

nkolban, 2017-12-12:
I placed a "bump" on issue https://github.com/espressif/esp-idf/issues/857 and took the liberty of emailing mr @TianHao-Espressif to ask if there is anything we can do to assist.  The Espressif 857 issue has been open since July but paused because it was believed that it would be "resolved" as we got closer to 3.0.  Now that 3.0 is just about upon us, I think we need to get a little bit more focused as we can not really say that the problem is *early* 3.0 code.

TianHao-Espressif, 2017-12-12:
I'm sorry for the problem. We will check and debug ASAP.

TianHao-Espressif, 2017-12-13:
@nkolban , could you provide me a sample code run simply to reproduce this problem. And what's your toolchain version?

nkolban, 2017-12-13:
Howdy,
The sample code can be found here:

https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/tests/BLETests/SampleWrite.cpp

This is a C++ application that leverages some C++ classes that only invoke the ESP-IDF BLE exposed APIs.

The documentation for this library can be found here:

https://github.com/nkolban/esp32-snippets/blob/master/Documentation/BLE%20C%2B%2B%20Guide.pdf

The complete source for the library can be found here:

https://github.com/nkolban/esp32-snippets/tree/master/cpp_utils

The ESP-IDF being used is the latest Github Master branch.

The tool chain reports as:

```
gcc version 5.2.0 (crosstool-NG crosstool-ng-1.22.0-75-gbaf03c2)
```

If needed, I can demonstrate on my machine, or work with you to install in your test environment (assuming you aren't able to recreate yourself)/

fly135, 2018-01-12:
I'm running your samplewrite code with SDK 3.0 and cannot see any service advertised using nRF Connect.  I do see in the log that there are no services advertised.

"BLEAdvertising: - no services advertised"

But looking at the code I also see that there is no call to BLEAdvertising::addServiceUUID.  After adding a call before starting advertising I now don't get that error, but still no services on the phone.

One question.... if the ESP32 is a BLE server, then why on nRF Connect does it show up in the "client tab".

chegewara, 2018-01-12:
@fly135 How long is esp32 name in BLEDevice::init()? If you want to advertise 128 bit service UUID then name cant be longer than 5 characters.

fly135, 2018-01-12:
@chegewara I just recompiled to change from "MYDEVICE" to "ESP32".  I'm still seeing nothing in the client tab.  The name ESP32 does show on the device in nRF.  I thought the advertising was for services.  Thanks for looking into this.

chegewara, 2018-01-12:
@fly135 With advertising you create data PDU, its a lot informations that are send with advertising packet. You can use it to advertise name, connection timings, 16, 32 or 128 bit services uuid and a lot more. But you have only 31 bytes to match it all. In this library we have preconfigured part of advertising data and it already has timings data, its 5 bytes, when you add 128 bit service uuid its another 16 bytes. We have also included flags which is another 2 bytes and TX power (another 2 bytes). 
So we have 2+2+5+16=25, 31-25=6.
We can use 6 bytes, 1 for opcode+5 bytes. Thats why we can use only 5 characters long name if we want to advertise 128 bit uuid.
Its maybe a little bit complicated or confused, but its enough to remember that 1 rule: short names (i forgot it myself lately)

chegewara, 2018-01-12:
@fly135 I now remember where is issue you can have just now, try to run menuconfig and change `Compiler options->Optimization Level->Release`. We already posted this issue but still its not fixed. Sorry for inconvenience

fly135, 2018-01-12:
@chegewara I've been using release mode.  I previously saw the post regarding that.

chegewara, 2018-01-12:
@fly135 I dont know what to say. I just build my new ubuntu environment and all works just fine with SampleWrite example. I can offer you my sdkconfig file just in case.

fly135, 2018-01-12:
@chegewara I would be happy to try the sdkconfig.  This is driving me nuts.  This is the IDF_VER...

ESP-IDF v3.0-rc1-dirty

BTW, do you know that's defined.  I searched what seemed to be all the files in IDF_PATH and counldn't find it.  Also I was wondering why the BLE server shows in the client tab of nRF instead of the server tab.

chegewara, 2018-01-12:
https://gist.github.com/chegewara/e33194b5276c346fd0c310fa99487feb

fly135, 2018-01-12:
@chegewara I must be going crazy because now the device is showing up in the server tab.  Even when I went back to a previously compiled app.  Thanks for posting the sdkconfig.  I had mine configured for 2MB flash and didn't know.  Also I didn't realize I could increase the UART speed for download.

Here's a pic from my nRF. Thank you for your help!  I think the 5 character device name was a factor.
  
[image]
[image]

vibnwis, 2018-03-01:
I am experiencing the same issue with onWrite. When writing, onWrite is never called. I have set to Release mode. However, on iPhone with nRFConnect, if I invoke 3 times write, and disconnect and reconnect, the onWrite function was entered 3 times. And not responding to write again.

nkolban, 2018-03-02:
I for one had been expecting a formal release of ESP-IDF v3.0 by now ... and had been waiting for that.  For this issue, I'd like to wait until ESP-IDF v3.0 formally ships.

closed 2018-11-15 by chegewara