nkolban/esp32-snippets #267: var size to high restart

boonkerz, 2017-12-12:
i need for an epaper display (spi) an buffer:
uint8_t _buffer[30000];

but it results in an reset?

chegewara, 2017-12-12:
Most likely its because you dont have enough free heap to allocate space for 30000 bytes. Without code or at least memory debug info its hard to say. Try to use this to debug:
http://esp-idf.readthedocs.io/en/latest/api-reference/system/mem_alloc.html#_CPPv225heap_caps_print_heap_info8uint32_t

boonkerz, 2017-12-12:
https://github.com/boonkerz/weatherstation/blob/master/components/gxepd/include/GxEPD.h#L33

i'm new to c++ and esp32

nkolban, 2017-12-12:
Howdy Mr @boonkerz 
While we always want to assist all users, please realize that the issues being responded to in this repository are meant to be issues found while using the code that is distributed by this repository.  General questions on C, C++ or ESP32 programming that don't relate to the projects in this repository are likely going to be better answered elsewhere.  On the Internet that are many forums that dedicate themselves to answering C and C++ questions from new beginners to advanced users.  For the ESP32 (in general) the very best place to ask questions on generic ESP32 is the ESP32 forum found here:

https://esp32.com/index.php

The value of using such a forum is that you are far more likely to find pre-existing answers or receive assistance from a much greater audience than is found here.  This repository is usually only patrolled by folks who have interest in only the code/logic/functions found exclusively in this repository.

chegewara, 2017-12-12:
Like i said, its caused by lack of heap and too large buffer you are trying to allocate. If you will change this line https://github.com/boonkerz/weatherstation/blob/master/components/gxepd/include/GxEPD.h#L33 just for testing purpose to `uint8_t _buffer[100];` code will not reset.

boonkerz, 2017-12-12:
yea i know but the code is a copy from 
https://github.com/ZinggJM/GxEPD/blob/master/GxGDEW075T8/GxGDEW075T8.h#L109
and this library works with platformio and the example

but i will switch to forum

closed 2017-12-12 by boonkerz