nkolban/esp32-snippets #270: PCF8575 Port Expander

chinswain, 2017-12-12:
Are there any libraries working on the esp32 for the Ti PCF8575?

nkolban, 2017-12-12:
How does the PCF8575 differ from the PCF8574?  We appear to have a class for the PCF8574 and maybe that class will work just fine with the PCF8575?

chinswain, 2017-12-12:
They look fairly similar, t he PCF8575 is the 16 Bit version (PCF8574 being 8). 

http://www.ti.com/lit/ds/symlink/pcf8575.pdf
http://www.ti.com/lit/ds/symlink/pcf8574.pdf

A library was written for Arduino that supports both: https://github.com/skywodd/pcf8574_arduino_library

Also one in the Arduino playground: http://playground.arduino.cc/Main/PCF8574Class

With a port to the ESP8266: https://github.com/WereCatf/PCF8574_ESP

I have a mcp23017 in the tool box too so I can fall back to that.

nkolban, 2017-12-12:
About the best I can suggest is that I add some PCF8575s to my EBay shopping list, get some instances and then test that they work with the 8574 class ... and if they don't make changes.

nkolban, 2017-12-13:
Ordered some PC8575s from EBay ... expect them to arrive in 2-3 weeks.

nkolban, 2018-01-25:
Wow .... PC8575s arrived as of this date.   Scheduling examination and testing by / through weekend at latest.

nkolban, 2018-01-27:
Well ... that was a bit embarrassing.  With my new PCF8575s I pulled down the datasheet and compared it to the PCF8574s that I had tested before.  I quickly saw the distinction between them and made the code changes in the library.  We now have a PCF8575 class.  What I consider to be embarrassing is that I could have made these changes at any time just by reading the datasheet as opposed to waiting months for delivery of physical units.  My confidence level is high that the changes I have made will work (meaning that they probably won't).   Since I'm busy these days ... I'm going to ask that you test the new code.  Again, if something goes wrong THEN I'll be happy to test ... but I'm feeling cocky that it will work.

The PCF8574 library would not have worked.  Reads and writes to a PCF8574 must be 16 bits and not 8 bits as for the PCF8574.  The new PCF8575 library basically doubled the bit sizes .. other than that the code is identical.  Back to you for a quick test on your end.