nkolban/esp32-snippets #272: i2c driver not installed when using the mcp23017 snippet

chinswain, 2017-12-12:
I've just tried using the modified library on an esp32 but I'm getting an error stating the i2c driver is not installed?
```
E (3888) i2c: /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/driver/./i2c.c:1100 (i2c_master_cmd_begin):i2c driver not installed
```

chegewara, 2017-12-12:
Before you can start to use mcp23017 you need to init i2c first. I think you need to use Wire.begin() in setup() but im not sure, you have to check

chinswain, 2017-12-12:
I'm using this to test:

```#include <Wire.h>
#include "Adafruit_MCP23017.h"

Adafruit_MCP23017 mcp;
  
void setup() {  
  Wire.begin(21, 22)
  mcp.begin(); 
  mcp.pinMode(0, INPUT);
  mcp.pullUp(0, HIGH);
  pinMode(5, OUTPUT); 
}

void loop() {
  // The LED will 'echo' the button
  digitalWrite(5, mcp.digitalRead(0));
}
```

chegewara, 2017-12-12:
Honestly, you are using Adafruit library, why you are opening issue in this repository?

chinswain, 2017-12-12:
Because I'm using the version from this repository created by Neil (Page 233 of his book) which is based on the Adafruit library. This issue is most likely not related to their original code.

https://github.com/nkolban/esp32-snippets/tree/master/hardware/expanders/mcp23017

chegewara, 2017-12-12:
I believe its port to use mcp23017 with esp-idf, but i have this chip so i can investigate your issue. Just give me some time.

chinswain, 2017-12-12:
Great, Neil mentioned on the Espressif forum that it has not been tested yet so may be an issue.

chegewara, 2017-12-12:
ive been using it with esp-idf, just dont remember if i did some changes in code. but as for now maybe you will try original library from adafruit installed with library manager, should have work and is better to use with arduino-ide

chegewara, 2017-12-12:
i have this chip, but i dont have breadboard atm, but like i said, you can try to use library from adafruit with examples they provide. 

Without chip connected to esp32 and with your code or with example from adafruit it works for me, im getting those errors, which mean i2c works, just device doesnt respond
```
[W][esp32-hal-i2c.c:235] i2cWrite(): Ack Error! Addr: 20
[W][esp32-hal-i2c.c:334] i2cRead(): Ack Error! Addr: 20
```

nkolban, 2017-12-13:
Howdy,
This code:

https://github.com/nkolban/esp32-snippets/blob/master/hardware/expanders/mcp23017/Adafruit_MCP23017.cpp

is meant to be run no-where near the Arduino libraries.

Here is the story.  In the beginning there was Arduino and folks wrote great libraries.  Then came the ESP-IDF which is a programming environment for the ESP32 which has nothing to do with Arduino.  To satisfy a needs, I lifted the Arduino MCP23017 classes and "reworked" them to run on the ESP32 ESP-IDF.  I wanted to maintain knowledge that these were not my libraries but a port so I left the name as "Adafruit_MCP23017".   Now we have an Arduino environment on top of the ESP32 ... 

What you should be using is the original MCP23017 classes found here for Arduino:

https://github.com/adafruit/Adafruit-MCP23017-Arduino-Library

To cut to the chase ... if you are working in Arduino (real Arduino or Arduino on ESP32) ... then use https://github.com/adafruit/Adafruit-MCP23017-Arduino-Library

If you are working in ESP-IDF with no sign of Arduino, you should use:

https://github.com/nkolban/esp32-snippets/blob/master/hardware/expanders/mcp23017/Adafruit_MCP23017.cpp

ro269, 2018-01-29:
Hi, I'm getting the same error when programming in the esp-idf environment : i2c driver not installed

I'm using the following: https://github.com/telanoc/esp32_generic_i2c_rw  ; which makes use of Kolban's MCP23017 snippet.

Please help.

Thanks.