nkolban/esp32-snippets #275: RSSI update while Scanning

Prasenjit67, 2017-12-13:
Hi,
I was looking for ways  to track the RSSI strength of already discovered devices while scan is on. 
The current implementation of BLEScan is limited  in registering the RSSI strength when a device is detected for the 1st time and subsequent GAP events are ignored . The crude way I have achieved so is by modifying BLEScan:: handleGAPEvent  case ESP_GAP_SEARCH_INQ_RES_EVT: like so 

```
// Examine our list of previously scanned addresses and, if we found this one already,
// ignore it.
					BLEAddress advertisedAddress(param->scan_rst.bda);
					bool found = false;

					for (int i=0; i<m_scanResults.getCount(); i++) {
						if (m_scanResults.getDevice(i).getAddress().equals(advertisedAddress)) {
							found = true;
							m_scanResults.getDevice(i).setRSSI(param->scan_rst.rssi);
							if (m_pAdvertisedDeviceCallbacks !=nullptr) {
								m_pAdvertisedDeviceCallbacks->onResult(m_scanResults.getDevice(i));
							}
							break;
						}
					}
```
is there a better way to achieve what I'm trying ?
pls note , I'm avoiding connecting to the device.

Prasenjit67, 2017-12-13:
updated code tried

`				BLEAddress advertisedAddress(param->scan_rst.bda);
					bool found = false;
					for (int i=0; i<m_scanResults.getCount(); i++) {
						if (m_scanResults.getDevice(i).getAddress().equals(advertisedAddress)) {
							found = true;
							m_scanResults.m_vectorAdvertisedDevices.at(i).setRSSI(param->scan_rst.rssi);
							
							break;
						}
					}`

Prasenjit67, 2017-12-13:
Sorry Closed it unintentionally .

nkolban, 2017-12-14:
Would a new flag that allows one to receive duplicated callbacks for advertisement records previously seen be an acceptable solution?

Prasenjit67, 2017-12-14:
Yes , that should work fine. Thanks.

nkolban, 2017-12-14:
The code changes have been made.  We now have an optional extra parameter on 

`BLEScan#setAdvertisedDeviceCallbacks()`

The signature is now:

`BLEScan#setAdvertisedDeviceCallbacks(callbackFunction, bool wantDuplicates=false)`

When we register the callback function to be invoked when a device is found in a scan, we can supply a flag which indicates whether or not we also want to receive duplicates.  The default is false.

During testing, I seem to be having some problems ...  With advertising, we can receive an advert every few tens of milliseconds.  When we say we want callbacks in duplicates, we seem to get "confused" somewhere along the line.  So while the code for this issue is now in place, it may have resulted in a new race condition of some other malady.  What I'd like you to do is test it in your environment and report back with your findings.

Prasenjit67, 2017-12-16:
Thank you nokilban.
Tried  with 3 BLE devices and saw few WDT resets when the period between adverts reduced to tens of ms .  Used the example scan app  with call back printing the advertised beacon on Serial ( too long processing time in callback).  Your apprehensions are correct . 

Was wondering whether getting access to (a copy) the list of currently discovered devices (m_scanResults.m_vectorAdvertisedDevices)  is an option instead of  callback on each advert?

pls suggest if should I close this and start a separate one.

nkolban, 2017-12-16:
Howdy.  So far no need for a second issue but many thanks for noticing that I like to keep distinct stories separate from each other.

The algorithm currently in the BLE C++ libraries is as follows:

1. Start scanning
2. Process received advertisements

For the story as a whole, the most common usage is to scan "looking" for something.  This means that when we find an advert, we ask "is this what I'm looking for" ... if yes, stop scanning and do something with it ... and if no, carry on scanning.   Because there can be LOTS of adverts and they occur from each device very quickly, we want the cost of determining "is this what we are looking for" to be as cheap as possible ... and so the optimization is that if we see an advert and we have PREVIOUSLY seen an advert, then this implies that we have already discounted that device from being of interest to us.  The scanResults vector serves two purposes ....

1. It records a previously seen advert so that we can skip it as quickly as possible if it is not what we want
2. There is a mode of operation where we perform a scan for a period of time (say 10 seconds) and then after 10 seconds present a complete list (the list of seen devices)

What you may want to consider is a different design in your story.

Let us assume you want to know the RSSI signal strength in range.  Then run a scan for 2 seconds which will give you the list of all devices (one time) including their signal strengths ... and then repeat that ...

Basically, ask yourself the question ...

Do I need to measure the signal strength of device X 100 times a second ... or do I need to know the signal strength of device X only once every 2 seconds?

Just a thought.