nkolban/esp32-snippets #276: Case sensitive HTTP headers

HanVertegaal, 2017-12-13:
Hi,

When I send a POST request with an http header 'content-length:0', the HttpServer fails (doesn't respond until the connection is closed). If I send the same request with 'Content-Length:0' everything works fine.

RFC 7230 states: "Each header field consists of a case-insensitive field name followed by a colon (":"), optional leading whitespace, the field value, and optional trailing whitespace."

I'm not sure if the find() on the map can be made case-insensitive, otherwise the headers should probably all be made upper- or lower-case.

nkolban, 2017-12-14:
Yup ... thats exactly what RFC 7230 says ... but sadly its not what the code does.  Taking this as a defect for resolution .. MANY thanks for raising this.

nkolban, 2017-12-14:
I have made some code fixes ... but haven't had time to test them yet.  Can I ask you to test the new code and see if that cures up the problem?

HanVertegaal, 2017-12-14:
Tested with the same client as before, and now it works fine. 
Thank you very much for all your efforts.

closed 2017-12-14 by HanVertegaal