nkolban/esp32-snippets #279: wifi scan fails to return access points after disconnect

rweller59, 2017-12-13:
Using ESPIDF with C++. ESPIDF, cpp_utils and toolchain all update a day or so ago. I've opened an issue on the ESP forum regarding this but since I'm using cpp_utils thought I'd mention it here as well. If strictly an ESPIDF issue, let me know and I'll close this.

Process:  scan for ap's, connect, send data every few seconds using paho mqtt. Everything is fine for 2-5 minutes then I get StaDisconnected event. When this occurs, I restart the process but the wifi scan returns 0 results. Not getting any errors logged other than wifi.cpp line 454 (ESP_LOGE(LOG_TAG, "Failed to allocate memory");) which is just the result of esp_wifi_scan_get_ap_num returning 0.

After re-boot, everything is fine again.

nkolban, 2017-12-14:
If I was attempt to recreate, the story would be:

```
while (forever) {
   scan for access points;
   connect to access point;
   form a TCP connection;
   send some data;
   end the TCP connection;
   disconnect from the access point;
}
```

Is that about the shape of it?

If you have a debug trace please post that as a PasteBin too.

rweller59, 2017-12-14:
more like this...
while (!ap connected) {
   scan for access points;
   connect to access point; }
form a TCP connection;
while(ap connected)
   send some data;

wifiEventHandler triggers StaDisconnected which starts the process from the top.

Shutdown system before I saw your response. Will attach trace tomorrow.

rweller59, 2017-12-14:
I forgot to delete json object in a function and after a time it propagated itself to this wifi issue. Having fixed that, I can now re-connect successfully after losing a wifi connection.

closed 2017-12-14 by rweller59

nkolban, 2017-12-14:
Many thanks for taking the time to post back.  I've lost count of the number of times I changed code in place A that broke logic in place B and spent hours and hours cursing Place B ... so I am completely on the same page as you.  I'm just pleased you are able to make progress.  You are welcome to create as many issues on the code we host on this repository as you wish and we appreciate you.