nkolban/esp32-snippets #280: WiFi auth methods, dhcp and events

Stormhand, 2017-12-14:
Right now the startAP() method is not very useful. I changed some code on myself to use it. For example:
`startAP(const std::string& ssid, const std::string& passwd, wifi_auth_mode_t auth=WIFI_AUTH_OPEN);`
`apConfig.ap.authmode        = auth;`
then:
`wifi->startAP("SSID", "PASS", WIFI_AUTH_WPA2_PSK);`

Second issue: there is no tcpip_adapter_dhcps_start(TCPIP_ADAPTER_IF_AP) so the class as AP does nothing. It should use some additional method to configure the dhcp for AP.

Last: **SYSTEM_EVENT_AP_STACONNECTED** and **SYSTEM_EVENT_AP_STADISCONNECTED** are not linked to **apStaDisconnected()** or **apStaConnected()** in **WiFiEventHandler::eventHandler**

nkolban, 2017-12-15:
Thank you sir ... I am deeply in your debt.  Not only did you point out some awful problems with the code but your recommendations were spot on.  I believe I have address the three primary issues but will of course leave it to your examination to determine if they are satisfactory.  It is disappointing to realize that these problems have been here for so long and gone un-noticed.  It makes me question the value of the class in the first place.  However, un-detered ... I'll keep on plugging.  If there is anything further I can do on this issue, don't hesitate to let me know.

Stormhand, 2017-12-15:
Thank you, Neil. Its working now. In the future you may add some setDhcpInfo/setApInfo to configure the network for the Access point.

Angel

nkolban, 2017-12-15:
Thank you for that ... I'm inept ... feel free to yell from roof tops about my ineptness.  Writing resume for a Walmart greeter now :-)

Your friend ... Neil

Stormhand, 2017-12-15:
Thank you Neil :)

I've proposed another addition for the WiFiEventHandler class. If you have time please look at it.

Angel

closed 2018-01-04 by Stormhand