nkolban/esp32-snippets #285: BLEDescriptor and BLERemoteDescriptor upgrade request

chegewara, 2017-12-15:
Thanks to last findings we realized there is some inconsistency between documentation on bluetooth.org and real data that is required to send if we want to turn on/off notifications and/or indications in descriptor 2902. That leads us to situation when we need sometimes to send 16 bit value. It would be nice to have overloaded function `writeValue(uint16_t)`.

nkolban, 2017-12-15:
Howdy ... can you elaborate.  Can you explain in more details the inconsistency that you mentioned?

chegewara, 2017-12-16:
Inconsistency:
- at first i just tought so there is value uint8 in documentation
- now i know there is uint16, but i dont see any information about bytes order.

closed 2018-01-05 by chegewara